MarlinFirmware/Marlin #4677: Feature request: add ability to use G38.2 command (CNC)

Bob-the-Kuhn, 2016-08-21:
Some CNC programs are now automatically adding the G38.2 commands to the gcode files.  It would be nice to have this implemented in Marlin.

Roxy-3D, 2016-08-21:
Do you want it to do the probe?  Or do you want all the output the command generates?  What exactly are you trying to accomplish?

thinkyhead, 2016-08-22:
RepRap GCode doesn't use the full GCode specification, but is more "inspired by" GCode, so we won't be implementing any codes with decimal points. The feature request is fine, but should be restated as a request for "a command to automatically establish the tool height from the build surface using the nozzle (or tool) as a conductor or switch."

As it happens, you can use a probe for homing with `G28` and that works for any situation where you have already established the distance from the probe to the nozzle. There is also an option to use the nozzle itself as a probe, and some machines do pass current through the nozzle to use it as the switch. Take a look at the Prusa i3 MK2 for example.

Marlin does not currently support CNC features or in-place tool-changing, so that would need to be established first to make G38.2 (or equivalent) viable.

Bob-the-Kuhn, 2016-08-23:
I like how it was restated except it sounds like it's just for the Z axis.  I'd like this to be available for the X, Y and Z axis.  

Maybe something like  "**command(s) to automatically establish the tool location from the build origin along an axis using the nozzle (or tool) as a conductor or switch**."

   Bob

thinkyhead, 2016-08-23:
> available for the X, Y and Z axis

You're talking about something which pertains very well to CNC, but not so much to 3D printing, at least not at the RepRap level. That said, we have a thread going already on the subject of implementing [more general CNC style support](https://github.com/MarlinFirmware/Marlin/issues/3208) within Marlin. I think you would do well to jump on that topic and look at what has already been discussed.

Bob-the-Kuhn, 2016-08-23:
I see you've already added this request to the **more general CNC style support** thread.  Thanks

That thread seems to be almost exclusively aimed at laser engraving.  I don't mind bothering them.

I'm puzzled about your G28 comment.  As best I can tell G28 will do a two stage move to the origin unless Z_SAFE_HOMING is enabled and the Z probe signal activates.  Is there something similar for the X & Y axis?

   Bob

Bob-the-Kuhn, 2016-09-12:
Attached are the modified files implementing G38.2 and G38.3 commands on Marlin 1.02

  Bob
[Marlin_G38.zip](https://github.com/MarlinFirmware/Marlin/files/468480/Marlin_G38.zip)

thinkyhead, 2016-09-13:
I'd personally rather you used the tools for making PRs provided by Github rather than asking me to do the work for you. My time is very limited right now, and it's not cheap.

Bob-the-Kuhn, 2016-09-18:
I've created a pull request for this feature request.

thinkyhead, 2016-09-18:
Your pull request didn't go to the [right place](https://github.com/MarlinFirmware/Marlin/pulls). Keep trying.

Bob-the-Kuhn, 2016-09-19:
Pull request 4845 was just created to implement the G38.2 and G38.3 commands.

https://github.com/MarlinFirmware/Marlin/pull/4845

closed 2016-09-27 by thinkyhead