MarlinFirmware/Marlin #4681: Error: T max - Thermocouples. help pls(

vyrd, 2016-08-22:
Hello!
I have RAMPS 1.4
Thermocouples
[image]
 (bought here - [https://reprapworld.com/?products_details&products_id/280/cPath/1591_1600_1613](url) )
Fee for connecting Ramps and thermocouples (bought here - [http://ru.aliexpress.com/item/3D-Printer-AD597-K-type-Thermocouple-Temperature-Control-Interface-Board-Ultimaker-Ramps-AD597-Free-Shipping/32367282285.html?spm=2114.13010608.0.50.aN3qpx](url) )

Connected properly checked several times. The firmware has changed #define TEMP_SENSOR_0 -1

and pins.h

#ifndef TEMP_SENSOR_0
   #define TEMP_0_PIN 3
#endif

But when you turn on an error T max.

thinkyhead, 2016-08-22:
What version of Marlin are you trying to use?

vyrd, 2016-08-22:
yes, i find!) The latest version 1.1.0 RC7. It should not edit pins.h! A file with the name of your motherboard (For RAMPS 1.4 it's - pins_RAMPS_14.h) 

It's work! 

Thx!

closed 2016-08-23 by thinkyhead