MarlinFirmware/Marlin #4682: effector doesnt raise between probe during autolevel

tyssoon, 2016-08-22:
hi everyone !

im struggling as the big noob i am with marlin for a whole week now. I have finnally manage to have a decent homing and the g29 command is recognized.

when i pull the button of the omron switch with my finger oh the Z probe i have this on m119:

16:07:40.914 : x_max: open
16:07:40.914 : y_max: open
16:07:40.930 : z_min: TRIGGERED
16:07:40.930 : z_max: open
16:07:40.930 : z_probe: open

the fact that the zmin and zprobe arent the same might be a problem but i dont understand why they're arent.

and when i ask a g29, after a g28, the effector get down. the omron hit the print surface, stop, move on x/y and get down again. And of course he crash on it.

do you guys have an idea of where i messed up ?

here are the config files

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/430378/Marlin.zip)

thanks !

Roxy-3D, 2016-08-22:
I answered over at RepRap.org.     But really, this issue would be better handled here.  I said:

OK. Turn on:

#define DEBUG_LEVELING_FEATURE

And give a M111 S 32767 to the system right after power up. 

It will give us a very detailed summary of what the printer is doing at each point in time. 

Please post the log here.

tyssoon, 2016-08-22:
hey roxy ! :)

`17:37:14.307 : OpenGL version:4.5.0 NVIDIA 364.72
17:37:14.309 : OpenGL extensions:GL_AMD_multi_draw_indirect GL_AMD_seamless_cubemap_per_texture GL_AMD_vertex_shader_viewport_index GL_AMD_vertex_shader_layer GL_ARB_arrays_of_arrays GL_ARB_base_instance GL_ARB_bindless_texture GL_ARB_blend_func_extended GL_ARB_buffer_storage GL_ARB_clear_buffer_object GL_ARB_clear_texture GL_ARB_clip_control GL_ARB_color_buffer_float GL_ARB_compatibility GL_ARB_compressed_texture_pixel_storage GL_ARB_conservative_depth GL_ARB_compute_shader GL_ARB_compute_variable_group_size GL_ARB_conditional_render_inverted GL_ARB_copy_buffer GL_ARB_copy_image GL_ARB_cull_distance GL_ARB_debug_output GL_ARB_depth_buffer_float GL_ARB_depth_clamp GL_ARB_depth_texture GL_ARB_derivative_control GL_ARB_direct_state_access GL_ARB_draw_buffers GL_ARB_draw_buffers_blend GL_ARB_draw_indirect GL_ARB_draw_elements_base_vertex GL_ARB_draw_instanced GL_ARB_enhanced_layouts GL_ARB_ES2_compatibility GL_ARB_ES3_compatibility GL_ARB_ES3_1_compatibility GL_ARB_ES3_2_compatibility GL_ARB_explicit_attrib_location GL_ARB_explicit_uniform_location GL_ARB_fragment_coord_conventions GL_ARB_fragment_layer_viewport GL_ARB_fragment_program GL_ARB_fragment_program_shadow GL_ARB_fragment_shader GL_ARB_fragment_shader_interlock GL_ARB_framebuffer_no_attachments GL_ARB_framebuffer_object GL_ARB_framebuffer_sRGB GL_ARB_geometry_shader4 GL_ARB_get_program_binary GL_ARB_get_texture_sub_image GL_ARB_gpu_shader5 GL_ARB_gpu_shader_fp64 GL_ARB_gpu_shader_int64 GL_ARB_half_float_pixel GL_ARB_half_float_vertex GL_ARB_imaging GL_ARB_indirect_parameters GL_ARB_instanced_arrays GL_ARB_internalformat_query GL_ARB_internalformat_query2 GL_ARB_invalidate_subdata GL_ARB_map_buffer_alignment GL_ARB_map_buffer_range GL_ARB_multi_bind GL_ARB_multi_draw_indirect GL_ARB_multisample GL_ARB_multitexture GL_ARB_occlusion_query GL_ARB_occlusion_query2 GL_ARB_parallel_shader_compile GL_ARB_pipeline_statistics_query GL_ARB_pixel_buffer_object GL_ARB_point_parameters GL_ARB_point_sprite GL_ARB_post_depth_coverage GL_ARB_program_interface_query GL_ARB_provoking_vertex GL_ARB_query_buffer_object GL_ARB_robust_buffer_access_behavior GL_ARB_robustness GL_ARB_sample_locations GL_ARB_sample_shading GL_ARB_sampler_objects GL_ARB_seamless_cube_map GL_ARB_seamless_cubemap_per_texture GL_ARB_separate_shader_objects GL_ARB_shader_atomic_counter_ops GL_ARB_shader_atomic_counters GL_ARB_shader_ballot GL_ARB_shader_bit_encoding GL_ARB_shader_clock GL_ARB_shader_draw_parameters GL_ARB_shader_group_vote GL_ARB_shader_image_load_store GL_ARB_shader_image_size GL_ARB_shader_objects GL_ARB_shader_precision GL_ARB_shader_storage_buffer_object GL_ARB_shader_subroutine GL_ARB_shader_texture_image_samples GL_ARB_shader_texture_lod GL_ARB_shading_language_100 GL_ARB_shader_viewport_layer_array GL_ARB_shading_language_420pack GL_ARB_shading_language_include GL_ARB_shading_language_packing GL_ARB_shadow GL_ARB_sparse_buffer GL_ARB_sparse_texture GL_ARB_sparse_texture2 GL_ARB_sparse_texture_clamp GL_ARB_stencil_texturing GL_ARB_sync GL_ARB_tessellation_shader GL_ARB_texture_barrier GL_ARB_texture_border_clamp GL_ARB_texture_buffer_object GL_ARB_texture_buffer_object_rgb32 GL_ARB_texture_buffer_range GL_ARB_texture_compression GL_ARB_texture_compression_bptc GL_ARB_texture_compression_rgtc GL_ARB_texture_cube_map GL_ARB_texture_cube_map_array GL_ARB_texture_env_add GL_ARB_texture_env_combine GL_ARB_texture_env_crossbar GL_ARB_texture_env_dot3 GL_ARB_texture_filter_minmax GL_ARB_texture_float GL_ARB_texture_gather GL_ARB_texture_mirror_clamp_to_edge GL_ARB_texture_mirrored_repeat GL_ARB_texture_multisample GL_ARB_texture_non_power_of_two GL_ARB_texture_query_levels GL_ARB_texture_query_lod GL_ARB_texture_rectangle GL_ARB_texture_rg GL_ARB_texture_rgb10_a2ui GL_ARB_texture_stencil8 GL_ARB_texture_storage GL_ARB_texture_storage_multisample GL_ARB_texture_swizzle GL_ARB_texture_view GL_ARB_timer_query GL_ARB_transform_feedback2 GL_ARB_transform_feedback3 GL_ARB_transform_feedback_instanced GL_ARB_transform_feedback_overflow_query GL_ARB_transpose_matrix GL_ARB_uniform_buffer_object GL_ARB_vertex_array_bgra GL_ARB_vertex_array_object GL_ARB_vertex_attrib_64bit GL_ARB_vertex_attrib_binding GL_ARB_vertex_buffer_object GL_ARB_vertex_program GL_ARB_vertex_shader GL_ARB_vertex_type_10f_11f_11f_rev GL_ARB_vertex_type_2_10_10_10_rev GL_ARB_viewport_array GL_ARB_window_pos GL_ATI_draw_buffers GL_ATI_texture_float GL_ATI_texture_mirror_once GL_S3_s3tc GL_EXT_texture_env_add GL_EXT_abgr GL_EXT_bgra GL_EXT_bindable_uniform GL_EXT_blend_color GL_EXT_blend_equation_separate GL_EXT_blend_func_separate GL_EXT_blend_minmax GL_EXT_blend_subtract GL_EXT_compiled_vertex_array GL_EXT_Cg_shader GL_EXT_depth_bounds_test GL_EXT_direct_state_access GL_EXT_draw_buffers2 GL_EXT_draw_instanced GL_EXT_draw_range_elements GL_EXT_fog_coord GL_EXT_framebuffer_blit GL_EXT_framebuffer_multisample GL_EXTX_framebuffer_mixed_formats GL_EXT_framebuffer_multisample_blit_scaled GL_EXT_framebuffer_object GL_EXT_framebuffer_sRGB GL_EXT_geometry_shader4 GL_EXT_gpu_program_parameters GL_EXT_gpu_shader4 GL_EXT_multi_draw_arrays GL_EXT_packed_depth_stencil GL_EXT_packed_float GL_EXT_packed_pixels GL_EXT_pixel_buffer_object GL_EXT_point_parameters GL_EXT_polygon_offset_clamp GL_EXT_post_depth_coverage GL_EXT_provoking_vertex GL_EXT_raster_multisample GL_EXT_rescale_normal GL_EXT_secondary_color GL_EXT_separate_shader_objects GL_EXT_separate_specula [... cut ...]
17:38:51.531 : echo:Z-Probe Offset (mm):
17:38:51.531 : echo:  M851 Z-13.40
17:38:54.627 : FIRMWARE_NAME:Marlin 1.1.0-RCBugFix (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Deltannok RC72mmbump EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
17:38:54.628 : X:0.00 Y:0.00 Z:0.00 E:0.00 Count X: 81017 Y:81017 Z:81017
17:38:54.628 : echo:DEBUG:INFO,ERRORS
17:38:54.628 : echo:Active Extruder: 0
17:38:54.707 : echo:DEBUG:INFO,ERRORS
17:38:54.707 : echo:Active Extruder: 0
`

Roxy-3D, 2016-08-22:
Yuch!   Can you re-post with the line breaks as they show up in PronterFace?

tyssoon, 2016-08-22:
yep of course:

[m111.txt](https://github.com/MarlinFirmware/Marlin/files/430632/m111.txt)

tyssoon, 2016-08-22:
is the txt file good ?

Roxy-3D, 2016-08-22:
Yes.  The file is good.    Can you reset the machine, do a G28 and then lower the Z-axis to the middle of the print envelope?     I want to see a M119 with the printer positioned there.   And then do a M401 and another M119     I think you have end stops (in particular, the Z-Probe) showing as triggered when they are not.

tyssoon, 2016-08-23:
@thinkyhead 
hi ! i have done what you said, i realy thought it could work but it didnt.
i tried Z_PROBE_TRAVEL_HEIGHT 20 and the effector still go down during g29 without raising between probes.
i believed in you ! :D

@Roxy-3D

i tried this
> //#define Z_MIN_PROBE_ENDSTOP
> #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
and that didnt change the no raising.

 z probe is plugged on the zmin endstop yes.

m119 in the middle of the print enveloppe:

Reporting endstop status
x_max: open
y_max: open
z_min: open
z_max: open

m401 had no effect for an mysterious reason

[m111 2.txt](https://github.com/MarlinFirmware/Marlin/files/431941/m111.2.txt)

i hope i have done the manipulation right.

thinkyhead, 2016-08-23:
By the way, welcome to Github!

tyssoon, 2016-08-23:
@thinkyhead 

Hey thank you ! :)

thinkyhead, 2016-08-23:
Your configuration looks ok to me. The last log you posted has nothing of use in it. You need to run `G29` and log the entire procedure, then post the output.

tyssoon, 2016-08-23:
oh ok sorry, i had switch it back to see the difference.

so with Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
i let my finger on the switch during the whole g29 to let him finish it.

[m111 3.txt](https://github.com/MarlinFirmware/Marlin/files/432102/m111.3.txt)

thinkyhead, 2016-08-23:
And, again, not useful. The output we're looking for isn't present. You must do two things, as Roxanne described above. First, enable this option in your configuration:
```cpp
#define DEBUG_LEVELING_FEATURE
```
...and after you have re-flashed the firmware, execute the command `M111 S255` before you do `G28` or `G29`. Also, please clear the log before you start, and only include the output from one session of homing and/or bed leveling.

tyssoon, 2016-08-23:
no no, im sorry, i really dont know what i am doing. I just understood that repetier kept everything from the beginning, i think i have cleaned it now, even if it's still quite long

#define DEBUG_LEVELING_FEATURE was uncommented, i dont understand why it doesnt work.

[m111 4.txt](https://github.com/MarlinFirmware/Marlin/files/432193/m111.4.txt)

thanks for your patience, really. i hope the txt is ok.

Roxy-3D, 2016-08-23:
@tyssoon    I've been looking at the latest m111-4.txt log.   I need to see the output of this code block in the G29:
```cpp

        if (verbose_level) {
          SERIAL_PROTOCOLPGM("Eqn coefficients: a: ");
          SERIAL_PROTOCOL_F(plane_equation_coefficients[0], 8);
          SERIAL_PROTOCOLPGM(" b: ");
          SERIAL_PROTOCOL_F(plane_equation_coefficients[1], 8);
          SERIAL_PROTOCOLPGM(" d: ");
          SERIAL_PROTOCOL_F(plane_equation_coefficients[2], 8);
          SERIAL_EOL;
          if (verbose_level > 2) {
            SERIAL_PROTOCOLPGM("Mean of sampled points: ");
            SERIAL_PROTOCOL_F(mean, 8);
            SERIAL_EOL;
          }
        }
```
Can you clear the log, do the following:   
- M111 S255       
- G28   
- G29 P4 V4

The V4 on the G29 will print those values for me.

tyssoon, 2016-08-23:
@Roxy-3D tell me if it's okay:

[m111 5.txt](https://github.com/MarlinFirmware/Marlin/files/432854/m111.5.txt)

Roxy-3D, 2016-08-23:
@tyssoon   I made a mistake.   I forgot this is on a Delta printer.   The coefficients do not get calculated using qr_solve() on a Delta.    But the log is good.   I keep looking at it.

tyssoon, 2016-08-23:
the beast ! :)

https://goo.gl/photos/9f4q7piJpAfHMRK76

maybe someone have a similar delta build with a marlin well configured where i would just need to change the rods lengths, etc. ?

thinkyhead, 2016-08-24:
The log shows that your Z never goes below ~453mm in height. That points to a possible mis-configuration. Is this known to be correct? :
```cpp
#define MANUAL_Z_HOME_POS 684 // Distance between the nozzle to printbed after homing
```
This specifies the distance between the **nozzle tip and the bed** when homed.

And, if that is correct, what does the output of `M119` show as you press each endstop and the probe? Are your endstops triggering normally in the way that they should?

tyssoon, 2016-08-24:
Hi @thinkyhead 

Z never goes below 453 cause i stay with my finger on the probe switch. I dont want it to crash on the print bed. otherwise the G29 goes to it "normally", it just doesnt raise between probes.

with the four switches pressed:

09:48:19.505 : Reporting endstop status
09:48:19.505 : x_max: TRIGGERED
09:48:19.505 : y_max: TRIGGERED
09:48:19.505 : z_min: TRIGGERED
09:48:19.505 : z_max: TRIGGERED

and the four unpressed;

09:49:33.863 : Reporting endstop status
09:49:33.863 : x_max: open
09:49:33.863 : y_max: open
09:49:33.863 : z_min: open
09:49:33.863 : z_max: open

tyssoon, 2016-08-24:
wow, it works now and i dont understand why, i have changed nothing. after u told me about the height i let the effector going down to the bed with my finger on reset waiting for the crash, and he didnt crashed. So in the code i suppose that if an height isnt reach by the probe et doesnt raise between probes.

im happy but also puzzled.

anyway thanks guys, you made it work ! even if i dont understand how ! :D :+1: 

thaaaaaaaaaaaanks !!!

closed 2016-08-24 by tyssoon

thinkyhead, 2016-08-24:
I'm glad to hear it was resolved. Sometimes all it takes is a re-flash and reset of EEPROM to clear up anomalies.