MarlinFirmware/Marlin #4684: Mesh bed leveling with manual modification

dumarjo, 2016-08-22:
Hi,

I try the latest RC7 with my COREXY and I need to do some tweak manualy to the mapping. 

I use the 3x3 point to have a rough correction. When I modify a point (or many), what command whould I use to get the new value take into acount. ?

I try:
G29 S3 X1 Y1 Z1.4
M500

G28

--- 
Test the new value. The new value looks like not taken into account. Do I need to reset the board every time I change the value ?

Regards

Jonathan

thinkyhead, 2016-08-22:
The value is stored right away and used from then on. You can confirm this with `M503`.

dumarjo, 2016-08-22:
Hi,

Well I will restest this tonight. Is it possible to debug this easily ? I try M111 S4 but, nothing usefull is comming out from serial. 

Regards

Jonathan

Roxy-3D, 2016-08-22:
I'm not sure this makes sense to do.   But if you want to use Mesh Leveling, the UBL System is built on a high resolution Mesh and has very nice tools in place to store and edit the Mesh.   If you aren't having any other issues with RC-7, you might want to bring up the UBL branch and see if it benefits you.   

(The CoreXY is a Cartesian printer...  I don't think anybody with a CoreXY has tried the UBL branch yet.  But it should work.)

dumarjo, 2016-08-22:
Hi,
@Roxy-3D I know and I follow your branch with attention. The problem is that I can't try it since I don't have the LCD and the button. If I can test/use it without the button, I will do it. 

Do you think this can be done quickly ?

Regards

Jonathan

thinkyhead, 2016-08-23:
`M111 S255` will give you the ultimate ultimate logging.

dumarjo, 2016-08-23:
Hi,

I have something working. I have implemented an backlash compensation on Z axis in the Planner and I compensated before the z compensation for the mesh leveling. Now I can print something !

I use the 3 x 3 matrix. I should probably try the 5 x 5 to have a better result. 

Thanx for your hellp @thinkyhead and @Roxy-3D 

Regards

Jonathan

Roxy-3D, 2016-08-23:
> @Roxy-3D I know and I follow your branch with attention. The problem is that I can't try it since I don't have the LCD and the button. If I can test/use it without the button, I will do it.

What kind of electronics do you have?   Because you can buy a 20x4 LCD Panel for $10 on eBay.  It should just plug into most boards.   I just got this one:

http://www.ebay.com/itm/251242416836?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

> Do you think this can be done quickly ?

I am going to make it so the UBL System works without a LCD Panel.  But that won't happen before I get the mesh_buffer_line() routine re-written to speed things up.   And immediately after I get that going, I'm going to start the work to bring up Delta printers.      

So, it could be done quickly, but it isn't going to play out that way.    Seriously...  The LCD Panel adds a lot of value to the system because of the way the Mesh Editor works.   It is worth having an LCD Panel!

closed 2017-11-01 by thinkyhead