MarlinFirmware/Marlin #4686: Printed objects are ~5% shorter after updating to RC7

ChristopherHoffman, 2016-08-22:
I recently updated from the stable 1.0.2 to the 1.1.0-RC7, and noted that objects are about 5% shorter (in the z-axis) than what they were before updating. I made sure to transfer over the configuration settings from my old version, and verified that the DEFAULT_AXIS_STEPS_PER_UNIT are the same in the configuration. 

An object printed pre-upgrade measured 155.21mm (which is correct for that 3D model), but the same gcode printed with the same filament after upgrade measures 146.46 mm. A 20mm calibration cube measures 19.2mm.

X and Y axes are not affected, only the Z axis is off.

The printer has Auto Bed Leveling enabled, and I noticed that the nozzle offset was completely off what is was when it was set from HOME, but there is already an issue open for that. I had to baby-step the nozzle down, I don't know if that could have affected the layer heights and caused the object's height discrepancy.

Roxy-3D, 2016-08-22:
My guess is you had to Baby Step down .8mm    This .8mm number keeps showing up.

ChristopherHoffman, 2016-08-22:
I think it was more than that, ~1.4 mm for the cube. I ended up reducing my z-offset by that to compensate, and needed to babystep ~0.3 mm for the second 155mm-tall object.

thinkyhead, 2016-08-23:
Try reducing your Z acceleration slightly and see if it helps. The acceleration code was fixed since 1.0.2-1, so your previous acceleration setting may now be too high.

ChristopherHoffman, 2016-08-24:
I will give that a try later today!

renzaaa, 2016-08-24:
To this version of Marlin RC7 the all axis steps is changed. Marlin RC3 the X/Y steps was 160.30 and Z was 795, but now X is 160.45, Y is 160.17 and Z is 800.  
Why this happens?

thinkyhead, 2016-08-24:
Your axis steps are wholly determined by the formula dividing steps-per-revolution by mm-per-revolution. If you find that you need to use values that are off from this, it implies lost steps or loose pulleys.

renzaaa, 2016-08-24:
Acceleration and jerk values is same, as it was for RC3 version.
Not lost steps and pulleys not loose.
Tomorrow I'll try to test different printer with Marlin RC7.

thinkyhead, 2016-08-24:
Not `RC7` but `RCBugFix` if you please.

closed 2017-10-31 by thinkyhead