MarlinFirmware/Marlin #4687: Feature Request: Nozzle Wipe sequence

adamfilip, 2016-08-22:
Im not sure if this is already possible. But i dont recall it as an option.

I would like it if if marlin had the ability to run a Nozzle wipe sequence which was just a user defined set of Gcode, at a set interval of time or layers..

This isnt something ive been able in simplify 3d.

Being able to tell my machine to move the X gantry all the the way over to the right side and do a back and forth wipe a few times and then continue on printing.

brainscan, 2016-08-22:
This has just been added along with park, settings and info are in configuration.h G12 and G27

Blue-Marlin, 2016-08-22:
How about g-codes for storing a position and returning to the stored point? In between you could do what you want.

Roxy-3D, 2016-08-22:
If we did something like this, my preference would be it is a single GCode and its behaviour is controlled with parameters.   

G13 S - Save current position
G13 R - Return to last saved position

closed 2016-08-23 by thinkyhead