MarlinFirmware/Marlin #4689: RAMBo 1.3 - RepRapDiscount Full Graphic Smart Controller - Blank Screen

dvseliteus, 2016-08-22:
Running [Marlin-RCBugFix](https://github.com/MarlinFirmware/Marlin/tree/RCBugFix)
[RAMBo 1.3](https://ultimachine.com/products/rambo-1-3)
[Geeetech Display Adapter for RAMBo Board](http://smile.amazon.com/dp/B00HUIQU8G)
[RepRapDiscount Full Graphic Smart Controller](http://smile.amazon.com/dp/B0151H6XWK)
Arduino version 1.6.9
Config.h: http://pastebin.com/yTq2Dn2Q
[image]
When I power-up I am getting a blank screen?

@thinkyhead it's me again...

brainscan, 2016-08-23:
Try uncommenting line 977

[dsc03185]<https://cloud.githubusercontent.com/assets/20667028/17874532/189bc08c-6881-11e6-9f51-4ab7cfbde3b0.JPG>

thinkyhead, 2016-08-23:
You don't need to redundantly enable `DOGLCD` because that is done automatically. Instead, I suggest you first check the wiring, and then if that doesn't resolve it uncomment these lines in `ultralcd_st7920_u8glib_rrd.h`…
```cpp
//#define ST7920_DELAY_1 DELAY_0_NOP
//#define ST7920_DELAY_2 DELAY_0_NOP
//#define ST7920_DELAY_3 DELAY_0_NOP
```
…and try different delay values. First try some of the delay values that are used below these lines for other boards. If you find some set of delays that works for you, great. If no amount of tweaking these helps, then we can look further.

Blue-Marlin, 2016-08-23:
Try to turn the contrast potentiometer on the display.

dvseliteus, 2016-08-23:
@brainscan thanks for the suggestion. I tried it with and without DOGLCD enabled and had the same result.

@Blue-Marlin thanks for the suggestion. I wished it were that easy. Turning the potentiometer on the display made the screen backlight dimmer and brighter but did not show and text or data.

@thinkyhead Thanks for all the details!
For the wiring I have the adapter installed and I tried switching the serial connectors. When I switch them the screen back-light does not come on. Do this mean that I have it connected correctly since I see a back-light even though I see no text?
I tried uncommenting the manual delay and tried various delay options and still no text displays. 
Lastly I installed Arduino 1.6.8 but still had the same result.

I'm not sure what it could be?

bgort, 2016-08-24:
@dvseliteus Try compiling with the Rambo board definition/plugin if you're using the Arduino IDE, or with 'board = reprap_rambo' if you're using platformio.  

Arduino Rambo board def:  http://reprap.org/wiki/Rambo_firmware#Arduino_1.6.4.2B_Board_Manager_Plugin

I ran into the same problem with a VIKI2; turns out that some of the Rambo's extended pins aren't supported by the standard Arduino or PlatformIO Mega 2560 board profiles; as in, the pins aren't mapped to the ports/registers on the 2560.  Not surprisingly, using unmapped pins doesn't work.

Using the board definition immediately cleared up the problem for me.

Hope that's all it is; took me days to sort that out.

Roxy-3D, 2016-08-24:
> I ran into the same problem with a VIKI2; turns out that some of the Rambo's extended pins aren't supported by the standard Arduino or PlatformIO Mega 2560 board profiles; as in, the extended pins aren't mapped to the ports/registers on the 2560. 

@bgort  Is the pins file in the Marlin directory for Rambo incomplete?  Can you make a recommendation for what has to be done to make sure you don't have a problem in the future?

bgort, 2016-08-24:
It's not that it's incomplete, I don't think.

It's that the extended pins (like 70, 71, etc.) aren't mapped by the 'default' Arduino/PlatformIO base libraries (pins_arduino.h) because they aren't usable on the commercially-available Arduino [reference?] boards, so those pins aren't usable in the SPI or other libraries.  The way I sorted out my issue was by putting a scope on the VIKI2 LCD CS line (pin 70) and realizing that it was noisy and thus not pulled high or low, which suggested it hadn't been set up as an output, and therefore couldn't be used as a SPI CS line.  That led me to realize - after talking with Ultimachine, who mentioned their board/pin definitions - that there's likely something in Marlin or U8glib that relies on the standard Arduino pin mappings, rather than the mappings in fastio.h (which seems to cover everything).

Using the Rambo board definition or specifying 'reprap_rambo' in platformio.ini both effectively include [pins_arduino.h.txt](https://github.com/MarlinFirmware/Marlin/files/435586/pins_arduino.h.txt) in place of the standard mappings (another, lesser, version of pins_arduino.h, I suspect, though I haven't gone looking), which only maps the pins actually used on the Arduino boards.

I'm not sure what the best way to go about letting others know of this issue is; perhaps just a comment in pins_RAMBO.h to use the Rambo board plugin/definition for Arduino IDE or 'reprap_rambo' in platformio is enough?  A few days ago I added the Rambo to RCBugFix's platformio.ini (https://github.com/MarlinFirmware/Marlin/pull/4652) and don't mind writing the comment/explanation in pins_RAMBO.h.

Roxy-3D, 2016-08-24:
> It's not that it's incomplete, I don't think.

You didn't say this, but I think what I'm hearing is they selected the wrong board.  Is the problem happening because they selected Mega2560 and not Rambo ?

> A few days ago I added the Rambo to RCBugFix's platformio.ini (#4652) and don't mind writing the comment/explanation in pins_RAMBO.h.

@bgort   Please do!!!

bgort, 2016-08-24:
> You didn't say this, but I think what I'm hearing is they selected the wrong board. Is the problem happening because they selected Mega2560 and not Rambo ?

Ultimately, yes, though it seems a good number of people don't know that the Mega 2560 board type doesn't map all of the pins on the Rambo, that there's a Rambo board definition that does map them, or even that any of this matters.  That includes multiple tech. support people both where I bought the Viki2 (I initially thought I had a bad Viki2) and where I bought the Rambo.  I certainly didn't know and spun my wheels for a while .. though it does make perfect sense now, of course... ;)

> @bgort Please do!!!

Will do.

dvseliteus, 2016-08-24:
@bgort thanks for the help. I have successfully installed the Rambo board definition/plugin for Arduino.
[image]
Unfortunately I am still getting a blank screen.
Here is a link to my Marlin folder if you want to have a look: http://www.mediafire.com/download/93228sblc6qzic5/Marlin-RCBugFix-Rambo.zip
Is it possible my wiring in wrong? I thought it was pretty straight forward?
[image]
I do have a viki lcd on hand but am not sure how to wire that up and had issues getting the menu to work with my previous setup.

thinkyhead, 2016-08-24:
Wiring up a Viki is always a challenge. I had to do so with an MKS board and – well, take a look at pins_MKS_13.h and you'll see how fun it was.

thinkyhead, 2016-08-24:
So, lessee… Did you try Marlin 1.0.2-1 to see if the display works with that version?

dvseliteus, 2016-08-24:
Haha I had a feeling that would be the case with the Viki. I wonder why none of my hardware combos want to work together. I am new at this so I always point the finger at my inexperience.

@thinkyhead let me give Marlin 1.0.2-1 a go and see what happens.

dvseliteus, 2016-08-24:
Uploaded Marlin 1.0.2-1 and here is what I get: https://youtu.be/zSO_7VdTTQc
Here's my quick Marlin 1.0.2-1 config.h file: http://pastebin.com/x7dWPYQk
Weird that when I switch the connectors I get a blinking and beep noise?
This doesn't happen with the Marlin-RCBugFix

thinkyhead, 2016-08-24:
Well, that is hideous and strange behavior. I note that the little breakout board that is used to plug in graphical displays is actually different from the little breakout board that is used to plug in character displays. Is it possible you're using one in place of the other?

esenapaj, 2016-08-24:
@dvseliteus 
Maybe if you connect the cable by reversal, problem is solved.
[image]

Roxy-3D, 2016-08-24:
> Maybe if you connect the cable by reversal, problem is solved.

Is this a picture of your hardware and it had to be modified?  Or did you find some place on the internet where somebody needed to do this?   But either way:

@dvseliteus  See if you can find pictures showing the connectors on both boards (in YouTube videos or where ever) where the board is working.  And see if you can tell which way the connector is keyed on both boards.   Obviously...  You boards should be keyed the same way!

dvseliteus, 2016-08-24:
@esenapaj YYYYYEEESSSSSSSSSSSSSSSS!
That did the trick. Thanks all for your help!
[image]

esenapaj, 2016-08-24:
@Roxy-3D 
>Is this a picture of your hardware and it had to be modified?

Yes it is. This is my RRD Full Graphic LCD.

dvseliteus, 2016-08-24:
Ya no issue with the firmware on this one. This issue was due to the hardware. Cutting the back of the header and inverting the connectors like esenapaj suggested is what made everything work in this case.

@thinkyhead Marlin-RCBugFix is uploaded and works extremely well. Looks like the only thing I can;t figure out is how to get the autobed leveling probe to trigger? When I send the M119 command it does not trigger when the red line goes on?
http://pastebin.com/4jvygZU7

Roxy-3D, 2016-08-25:
@esenapaj   Were the cables made wrong?  Because I have a Full Graphic LCD that has the connectors identical to the picture you posted.    I have not installed or used the Full Graphic LCD panel yet.  But it looks like I may need to cut the connectors also if I run into trouble. 

The thing I'm having trouble with is this can't be a requirement to use the board, right?   Something else is going on here????

dvseliteus, 2016-08-25:
@Roxy-3D I'm thinking this may be the manufacturer of the Full Graphic Display with the red PCB?
I noticed all of the displays during my research had white pcb's. Just a thought.

dvseliteus, 2016-08-25:
@thinkyhead 
I had to disable //#define ENDSTOPPULLUPS
and
// Enable Z Probe Repeatability test to see how accurate your probe is
#define Z_MIN_PROBE_REPEATABILITY_TEST
Not the probe triggers just fine

Roxy-3D, 2016-08-25:
> I'm thinking this may be the manufacturer of the Full Graphic Display with the red PCB?
I noticed all of the displays during my research had white pcb's. Just a thought.

@dvseliteus 
During your research, did see any pictures of the white pcb's having the connector flipped from what the red board has?

dvseliteus, 2016-08-25:
@Roxy-3D Nope I only saw instances where people were swapping connector Ext1 and Ext2 locations.

esenapaj, 2016-08-25:
>Were the cables made wrong? 

No it isn't (wasn't).
This problem is almost caused when you connect between LCD and "adapter by another maker (seller)". 
Example:

cut-out portion located top
http://www.ebay.com/itm/Full-GraphicSmart-12864-128-64-LCD-Display-controller-adapter-for-RAMPS-1-4-/200975999785

cut-out portion located bottom
http://www.ebay.com/itm/12864-LCD-Display-Smart-Controller-With-Adapter-For-Reprap-RAMPS-1-4-3D-Printer/331627519644

cut-out portion located bottom,  and exp1 and exp2  are leftside right against above two
http://www.ebay.com/itm/LCD-12864-Smart-Display-Controller-Module-Adapter-for-RAMPS-1-4-3D-Printer-/282070616858

If you bought your Full Graphic LCD as set item, probably problem isn't caused.

closed 2016-08-29 by dvseliteus