MarlinFirmware/Marlin #4690: Motor turns off during ooze prevention in slic3r

athulsnair, 2016-08-23:
Hi,

I've recently upgraded extruder in my printer to dual extruder and i'm printing my first test file now
http://www.thingiverse.com/thing:21773
FW = marlin(it's a slightly old version,  I don't know how to configure current version for printrBoard rev D and extruder board)
https://drive.google.com/open?id=0B5_IWI68ZqZTS0hEamd1RUtrTU5TYlAySlU1M1VvR00xTEdj
slcer = slic3r 1.2.9
host = Octoprint

I noticed some shifting while printing.

So I spent some time in front of the machine. I've enabled the ooze prevention in slic3r so that inactive extruder cools down to 5 degree. After each layer both extruders go outside the ooze prevention wall and wait there till it reaches the correct temperature. I took around 10 seconds for that. During this waiting period stepper motors turn off after a few seconds. and it looses the position (mainly in X, might be due to the extra weight of dual extruders) . To verify this I manually tried to turn the motors during the waiting period. They were powered up initially, but after that that they turn OFF and I can rotate them manually. 

What should I do to avoid it?? 

Thanks﻿
Show less
1

thinkyhead, 2016-08-23:
Do your settings look like this…?
```cpp
#define DISABLE_X false
#define DISABLE_Y false
#define DISABLE_Z false
#define DISABLE_E false // For all extruders

#define DEFAULT_STEPPER_DEACTIVE_TIME 120
#define DISABLE_INACTIVE_X true
#define DISABLE_INACTIVE_Y true
#define DISABLE_INACTIVE_Z true  // set to false if the nozzle will fall down on your printed part when print has finished.
#define DISABLE_INACTIVE_E true
```
With these settings the steppers should only disable after 2 minutes of inactivity. You can use the `M18` (aka `M84`) command to change the timeout value. For example, `M18 S300` will set the timeout period to 5 minutes.

athulsnair, 2016-08-25:
In mine it looks like this,
`//By default pololu step drivers require an active high signal. However, some high power drivers require an active low signal as step.
#define INVERT_X_STEP_PIN false
#define INVERT_Y_STEP_PIN false
#define INVERT_Z_STEP_PIN false
#define INVERT_E_STEP_PIN false

//default stepper release if idle
#define DEFAULT_STEPPER_DEACTIVE_TIME 15
#ifdef XLBOT
#define DEFAULT_Z_STEPPER_DEACTIVE_TIME 5
#endif
`
SO it means X, Y and E motors gonna disable if motor is inactive for more than 15 seconds and Z after 5 seconds??

Changing it to 120 or 300 will solve it right??

#define DEFAULT_STEPPER_DEACTIVE_TIME 120

closed 2017-10-31 by thinkyhead