MarlinFirmware/Marlin #4692: Z_SAFE_HOMING outside of bed

tampas, 2016-08-23:
Hi, i'm trying to figure it out, how to config a z-probe only used for homing, to home on a safe zone.

It works when i send a Home all Axis command, but it fails when i try only to home the Z_Axis.
Sometimes it was outside of the bed and causes some potentially harmful accidents.

How can i correct this feature?

Thanks in advance

thinkyhead, 2016-08-24:
We'll have to look closer at that. I'm not sure that `SAFE_HOMING` is actually intended to work for `G28 Z`, but only for `G28` homing all axes. Do you see the same problem if you home just 2 axes, such as `G28 X Z` or `G28 Y Z`?

tampas, 2016-08-24:
Hi,
I'm not sure also that SAFE_HOMING is intended to work for G28 Z command.
From the description provided it seems to be a routine to home all axes only.
But this case can provide some trouble situations.

I will test the G28 X Z and the G28 Y Z command and provide more info about it.

thinkyhead, 2016-08-24:
@Roxy-3D It's just you and me around here lately. So, do you think `Z_SAFE_HOMING` should always apply when homing Z, or only when using `G28` on all axes, as originally specified? The thing is, if we extended it in that manner, then if you haven't already homed X or Y then those would need to be homed first before you could home Z. So then we'd have to choose whether to unexpectedly home X or Y or both first, or to throw up an error (which might be unseen with no console open) when you tried to home Z without homing X and Y first. It seems to me that this only sows more confusion, but what are your thoughts?

Roxy-3D, 2016-08-24:
I probably lack the perspective to make my opinion valuable.  But they way I see it, homing is done so seldom, is there a benefit to doing it fast?   I think I would argue we should always try to protect the machine from doing damage to itself. 

Looking at it from a slightly different perspective, both Delta's (in general) and Cartesian machines with probes offset from the nozzle need to know where they are in the (X,Y) before downward Z movement is safe.   I know people are going to disagree, but I think it makes sense to insist (X,Y) are known prior to downward movement of Z.

thinkyhead, 2016-08-24:
I tend to agree. But the problem is we can't guarantee that (a) the user will realize either why the Z is refusing to home before homing XY in the case where we simply prevent it, or (b) the user will be annoyed that the XY homes (unexpectedly) in response to `G28 Z`. So one way or another, if we go down this path there will have to be compromise.

Roxy-3D, 2016-08-24:
> So one way or another, if we go down this path there will have to be compromise.

There may be multiple paths forward.   One 'compromise' is we do nothing in RC-7 & RC-8.  We keep it just the way it is.   But we could make the change over in the devel_ubl branch and that give us time to see if people agree with the direction.     If we did that, it might make sense to make SAFE_HOMING the default, and you have to enable UNSAFE_AND_RECKLESS_HOMING (just kidding!)  if you want the other behavior.

thinkyhead, 2016-08-24:
I hear you. Frankly, the use of `G28 Z` is somewhat niche. 99% of the time we just do a plain `G28`.

Roxy-3D, 2016-08-24:
> I hear you. Frankly, the use of G28 Z is somewhat niche. 99% of the time we just do a plain G28.

Agreed.  Except, doesn't PronterFace do a G28 X or G28 Y or G28 Z when you click the home button?  I thought it did.   

Anyway...   G28 was kind of an early example of what happened to G29.   As stuff got added, pretty soon we ended up with #ifdef's nested 37 levels deep.   It would be nice to clear out some of that clutter!     If we rewrote G28 we could clean it up a lot.

thinkyhead, 2016-08-24:
Funny you should say. I was just thinking last night, heck, let's just make separate versions of the whole `G28` and `G29` functions for each type, and that will keep us from having our eyes crossed trying to see the implications quite so often.

Roxy-3D, 2016-08-24:
> ~~and G29 functions for each type, and that will keep us from having our eyes crossed trying to see the implications quite so often.

Or...   There is only ONE version of G29 in UBL !  :)    (But I will have a few #ifdef DELTA 's to change where the nozzle and probe are allowed to travel.)    So if you are going to rewrite anything, G28 is more needed than G29 !

tampas, 2016-08-29:
Hi, 
first of all, my apologies for taking so long to give you the promised feedback.
The problem persists when i do the G28 X Z and G28 Y Z.
The machine doesn't move in XY plan ensuring that the Z Probe is in a safe place inside of the bed area.

does not exist any easy solution for this? that i can force by firmware that when Z is argument of G28 command, that the firmware ensures that we always make a Z_SAFE_Home routine?

my problem is that for some unexperienced users, this safety fault could be a huge problem.
Some host softwares (as simplify3D in my case) grants a Home Z button, and i cannot predict that any user will not trigger it.

Best regards,
T_Ermida

thinkyhead, 2016-08-30:
Someone will have to implement it in their copious spare time.

tampas, 2016-08-30:
Ok, thank you for your feedback on this matter.
For now, i'm thinking on a turn around solution for this.
I'm trying to limit the machine moves, reducing is X_MAX_POS and Y_MAX_POS, to ensure that Z_Probe is always above the bed.

As i say, is only a temporary solution, because it makes me loose some print area, and the machine will never touch its X_Max_ENDSTOP and Y_MAX_ENDSTOP. This fact could cause another security issue.

Do you think that this solution could work?
Thank you

thinkyhead, 2016-08-30:
The solution isn't that hard. Instead of only doing the safe home based on `home_all_axis` it should instead just check that X and Y have already been homed.

lrpirlet, 2016-09-01:
This is not a new discussion... You'll need to home X and Y before homing Z
whenever power was cut to the motor of X and/or Y axis...

RogerRepRap, 2016-09-02:
If you are doing a print with items left on bed from previous run you may not be able to home X and Y before Z without crashing into something. A G92 X0Y0 followed by a G28 Z works fine in this scenario. May be just bad habits picked up from running a CNC mill with multiple setups.

thinkyhead, 2016-09-02:
Actually G92 doesn't cause the XY to become marked as "known" or "homed" and so if Z safe homing is enabled, it will still behave the same way even after a G92.

thinkyhead, 2016-09-02:
> You'll need to home X and Y before homing Z whenever power was cut to the motor of X and/or Y axis

True. The positions become "unknown" at that point. The machine has no way to know that X and Y weren't moved arbitrarily.

However, in the case where X and Y steppers have not been disabled, it should be ok to do z safe homing. As mentioned, the fix isn't too involved. But it's not something I'll be able to implement for at least a few days.

Blue-Marlin, 2016-09-02:
With the behaviour as it is today you can home z wher evever you want - when homed alone. This is important when there is a part on the bed. Homing z always at a pre-specified point will destroy this flexibility and will prevent from homing z in too many cases. At lesat there must be a parameter to restore the used behaviour.

htaxil, 2017-03-29:
hi all,

I'm actualy hacking a Cube-X 3D printer, who have magnetic switches (REED) for endstop, including for the z axe. for this z axe, one magnet is housed under the bed at the coordinates X140, Y150 and the endstop is located on the effector. I must therefore place the effector above the magnet before leveling Z, Including when I only do the Z homing (G28 Z).

Is there a solution ?

Regards,
Hubert

htaxil, 2017-04-07:
up

thinkyhead, 2017-04-12:
I believe this is the behavior of 1.1.0 `RCBugFix` currently. If you try to home Z with a "probe" — or using `Z_SAFE_HOMING` and if **_XY are already homed_**, then `G28 Z` should move to the safe homing point (as defined by `Z_SAFE_HOMING_X_POINT` and `Z_SAFE_HOMING_X_POINT`) before homing Z.

thinkyhead, 2018-05-24:
@cryocide — Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem isn't resolved yet, then please open a new issue.

ManuelMcLure, 2018-05-24:
Also, do you have `STEALTHCHOP` enabled? There is some homing weirdness that has been seen if `STEALTHCHOP` is disabled. Look at #10605.

cryocide, 2018-05-24:
@thinkyhead Will do, and thanks! 

@ManuelMcLure Actually, yes I have it enabled, but the hybrid threshold is (was) set to the same rate as my homing_feedrate.  I think it was going into spreadcycle during homing.  I'll raise the hybrid threshold by 1 to make sure homing happens in stealthchop mode and will test under 1.1.8 in addition to trying out bugfix 1.1.x and 2.0.x.

closed 2018-12-10 by thinkyhead