MarlinFirmware/Marlin #4694: Dual_X_Carriage proper behavior is broken

AndrewDiehl, 2016-08-23:
M605 S0 is default. 
Homed All axis. 

T0 is at 0
T1 is at 400

Send following commands:
--------------------------------
T0
G1 X10 F1000
---------------------------

The T0 head (left side) moves to X35 instead of X10, 
Subsequently can not move the T0 head to anything less than X35 without a homing command. 

X35 is the the T1 minimum position, and should not be affecting T0

Additionally, 
Issuing a T1 command immediately sends the T1 head to the same position as the T0 head, causing a collision. Simply issuing T1 should not cause any movement.

thinkyhead, 2016-08-24:
Unfortunately we haven't had any testers with a dual-x setup. If you can help us to work on this it would be very much appreciated.

AndrewDiehl, 2016-08-24:
I will do whatever I can to help.

AndrewDiehl, 2016-08-29:
Is there anything I can do to help right now, or just wait until its patched and try again?

boelle, 2016-09-03:
@thinkyhead anything he can do ?

thinkyhead, 2016-09-10:
He can call in the geek troops to help me! I have very limited time to work on the main Marlin project than I would like these days, and I have some other issues to tackle today.

boelle, 2016-09-10:
@AndrewDiehl :-D so if you know some geek people that can help out let us know

AndrewDiehl, 2016-09-14:
My people are all Electrical/Mechanical but no software :(

If any of your tasks involve CAD or fabrication, I'd be happy to help lighten the load.

psyvision, 2016-10-28:
I'm watching this now as I have the T0/T1 crash issue.  I'll take a look at the code later today

boelle, 2016-11-02:
@AndrewDiehl i assume this "problem" is still present?

AndrewDiehl, 2016-11-02:
I would assume so, as nobody has asked me to test a fix.

boelle, 2016-11-02:
@AndrewDiehl are you able to work with @thinkyhead to figure it out?

you test and he tries to fix?

AndrewDiehl, 2016-11-02:
I'll be away from my machines until Friday, but generally, yes.

boelle, 2016-11-02:
oki doke... then i think @thinkyhead will poke you when he is ready :-)

thinkyhead, 2016-11-02:
@AndrewDiehl What time zone are you in? Perhaps we can do an online hangout or chat session where I can try patching some things and you could give feedback. I'd like to make sure all Dual X behavior is proper, including`T0`/`T1`, parking the head, doing duplication mode, etc., and I know if I try to work it out on my own it'll be hit-or-miss.

AndrewDiehl, 2016-11-02:
Eastern time USA.

thinkyhead, 2016-11-03:
@AndrewDiehl I'm on Central Time currently. Do you have some availability Friday night, or over the weekend to hack on this?

boelle, 2016-11-05:
@AndrewDiehl will you have time this weekend?

thinkyhead, 2016-11-06:
@AndrewDiehl If I had a Dual-X machine on hand to test with this and other Dual X issues could be solved in a day. Without one I'm terribly afraid of making it worse.

AndrewDiehl, 2016-11-06:
Sorry for the delay here,

I'm trying to make time for this on Sunday.

My newborn is being extra fussy the last few days so its been harder to
find time than expected.

~Andrew

AndrewDiehl, 2016-11-06:
@thinkyhead 

 I can do google hangouts using landruthegreat@gmail.com or you can call 610-295-7867. For voice I may have to call back using wi-fi since my house has poor signal.

thinkyhead, 2016-11-07:
@AndrewDiehl When you have a chance, please post a log of `G28` so I can see what state the machine is in after you've homed. Also post your configuration, or the relevant settings.

I'm starting to make some tweaks based on your description of the issue, but I'm being cagey because I'm not entirely sure what all the tool-change behaviors should be for every Dual X mode.

Let me see if I have the general idea anyway…

- `DXC_FULL_CONTROL_MODE`: Simply set the current position to the position of the newly-selected nozzle, without any movement.
- `DXC_AUTO_PARK_MODE`: Park the active carriage and (unless `S1` -stay- is given) move the selected carriage into its vacated position.
- `DXC_DUPLICATION_MODE`: I presume there should be no movement on tool-change in this mode.

Tool-change in `DXC_DUPLICATION_MODE` is a curious concept. With extruder 0 active, its movements are duplicated in extruder 1. But if extruder 1 is selected, should it then become the one that is moved, while extruder 0 follows it? Or should the duplication flag (`extruder_duplication_enabled`) just be disabled until extruder 0 is selected again?

thinkyhead, 2016-11-07:
@AndrewDiehl Also test this branch, which has some tweaks that may help (or not):
https://github.com/thinkyhead/Marlin/tree/rc_dualx_homing

thinkyhead, 2016-11-07:
@psyvision @pbrgle1 @gsart84 Joining this thread…

gsart84, 2016-11-07:
Thanks. Still got some troubles. Now zaxis moves up withoud stopping

thinkyhead, 2016-11-07:
@gsart84. When does Z move up without stopping?

gsart84, 2016-11-07:
After toolchange...
I do g28,
Then move t0 to x140
Then a toolchange. 
First the x2 crashes in the endstop
Then z lifts up with no stopping

psyvision, 2016-11-07:
Can someone remind me what g-code will enable logging and i'll do a log of G28 if it is still needed? @thinkyhead 

@gsart84 was the tool crashing with RC7 or the branch that @thinkyhead posted above?

gsart84, 2016-11-07:
It crashes with the branch from thinkyhead.
In rc7 after g28, when i do a movement (i.e. g1z1)
All axes do some strage slow movements to x0 y0 z10
To move the head or table i have to:
G28
M605 s0
M605 s1
And then i can move withoud pronlem

gsart84, 2016-11-07:
M111 S247 is the code for logging.
 And here the video of my problem with rc7
https://youtu.be/Y8UuKQfCeCE

psyvision, 2016-11-07:
Send: M111 S247
Recv: echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION
Recv: ok
Send: M105
Recv: echo:M105
Recv: ok T:19.5 /0.0 B:16.8 /0.0 T0:19.5 /0.0 T1:17.0 /0.0 @:0 B@:0 @0:0 @1:0
Send: M105
Recv: echo:M105
Recv: ok T:19.4 /0.0 B:16.7 /0.0 T0:19.4 /0.0 T1:16.8 /0.0 @:0 B@:0 @0:0 @1:0
Send: G28
Recv: echo:G28
Recv: echo:Active Extruder: 0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:Active Extruder: 0
Recv: X:-14.00 Y:-50.00 Z:1.50 E:0.00 Count X: -1120 Y:-9000 Z:4000
Recv: ok
Send: M105
Recv: echo:M105

I have mesh bed levelling enabled and have an x-min of -14 and an y-min of -50

Tool change with S1:

Send: T0 S1
Recv: echo:T0 S1
Recv: echo:Active Extruder: 0

Tool change without S1:

Send: T1
Recv: echo:T1
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing

Crashes T1 into T0.  I'm not sure how that helps though.  Is there some additional info I can get somewhere?

Blue-Marlin, 2016-11-07:
Likely it would be a bit more informative if you'd activated `DEBUG_LEVELING_FEATURE` in your config.

psyvision, 2016-11-07:
Cheers, i'll reflash with that on! Sorry I'm new to debugging Marlin :/

psyvision, 2016-11-07:
Okay I've put a gist together with the log in it:

https://gist.github.com/psyvision/e0dcae5087cf4c262b0f62d032469a84

G28 was okay.  T1 S1 caused no movement, T1 caused no movement. T0 caused T0 to crash into T1.

gsart84, 2016-11-07:
@psyvision
In yout configuration adv i see you have x2-max-pos x-max-pos

So if i understand right, if my printer would be configured like yours,  t1 would crash into t0.

psyvision, 2016-11-07:
It's just short hand for saying the max X2 position is the same as used for X max without putting the same number twice.  I had seen this done for a few different IDEX printers so went with it for initial configuration.

I can get T0 to hit T1 and T1 to hit T0.

thinkyhead, 2016-11-08:
Note: On a dual setup your first carriage cannot move as far as your second carriage, so `X_MAX_POS` should be smaller than `X2_MAX_POS`. Also, the second carriage can't move as far left as the first one, so `X2_MIN_POS` should be larger than `X_MIN_POS`.

thinkyhead, 2016-11-08:
> I can get T0 to hit T1 and T1 to hit T0.

That's the great thing about `DXC_FULL_CONTROL_MODE`. Total freedom.

thinkyhead, 2016-11-08:
> G28 was okay. T1 S1 caused no movement, T1 caused no movement. T0 caused T0 to crash into T1.

The test branch I linked is far from ready. Thanks for giving it a test. I will look at the log more closely. I may add more logging and ask for more testing.

AndrewDiehl, 2016-11-08:
Wow! Lots to go over here.

Learning about the T0 vs T0 S1 is totally new news to me. Is this in the code comments anywhere? Its not in the marlin.main documentation nor under the Dual X Carriage define as far as I can tell. This should be in both places. 

The "Unpark when using Z_SAFE_HOMING with Dual X #5175" branch fixes the park/unpark issues while in DXC_FULL_CONTROL_MODE.  Changing from T0 to T1 no longer crashes the extruder heads into one another. 

However, T0 is still being made to obey the X2_MIN_POS. 

After homing, a move command makes T0 move to the X2_MIN_POS if the command doesn't move it past that value. 

ie: my X2_MIN_POS is 35

I home my machine. 
Select T0
Issue a G1 X1 F10000

T0 immediately moves to X35. 

If I home and issue a G1 X100 command, T0 moves to X100. 

----------------------------------------------------------------------------

The Autopark mode in 5175 is much scarier than before. 

Dropbox link to my configs https://www.dropbox.com/sh/mg5z4r6i2fag7p9/AADZOaF2SrCTgiu-sVn2_Yzta?dl=0

Issuing a T1 command after homing and Setting M605 S1 now sends the Z axis down to Z0 and Back instead of up to the z hop distance. Afterwords, Selecting T0 again forces both heads past their end stops and moves the Z axis to Z0 and back

psyvision, 2016-11-08:
@thinkyhead I had set my X2 min/max to those values because I was observing the behaviour that @AndrewDiehl has described above, when switching to T0 it would move to X2_MIN and could never go further left (i.e. back into it's dock/park position).  Putting it at 0 and X2_MAX = X_MAX (as BCN3D have done with the IDEX Sigma firmware, although heavily modified) allowed me to get the carriages in the appropriate places.

boelle, 2016-11-12:
@AndrewDiehl @thinkyhead what outut did you get from the testing session?

thinkyhead, 2016-11-21:
@boelle We ended up getting drunk and falling asleep before we could get anywhere. But soon…

closed 2016-12-06 by thinkyhead

wallacezhang, 2017-03-27:
@psyvision @AndrewDiehl @thinkyhead have you solved the T0/T1 crash issue? I'm suffering this issue now, don't know where I'm wrong, would you like to help? thanks a lot.

gsart84, 2017-03-27:
RC 8 / RC bugfix work perfect.
You have to adjust configuration.adv the right way. For a example you can download my configuratin in my github...

wallacezhang, 2017-03-28:
@gsart84 I downloaded your RC8 firmware and tested in my printer, the T1 carsh into T0 still. any thought?

gsart84, 2017-03-28:
@wallacezhang 
ok, hm, i do not know where the problem. my printer is working perfect with the configuration you use!

gsart84, 2017-03-28:
is it a custom printer? witch board? i use a rumba board, have 2 endstops and have a x offset of 40mm from nozzle to heatbed...

wallacezhang, 2017-03-29:
@gsart84 my printer is Prusa i3 structure, I also use Rumba board. here is my issue when uploading your firmware:
1. when T0 finish printing, it'll go home normally, but T1 moves to print **very slowly**, moreover, T1 doesn't start to print when it go to the place of model.
2. when switch to T0 again, T1 can't go home normally, I have to cut power, so as to avoiding T0/T1 crash.
do you think it's the issue of firmware or slicing?

gsart84, 2017-03-29:
Do you use dxc auto park mode?
With auto park mode, there is no problem with simplify3d and cura. Do not use a tool change skript.
Your problem seems to me like wrong settings with your offsets. I know this problem with slow movements. The Only one slow movement that remains on my printer, is moving y after homing x and y after printt is finished. Cant solve this issue.

gsart84, 2017-03-29:
Attention if you use my firmware, check the invertet stepper motors, i hope you adjust this firmware correctly for your movements

wallacezhang, 2017-03-29:
@gsart84 I use your firmware directly and have such issue after uploading it. 
wrong settings with offsets? which one do you mean?

gsart84, 2017-03-29:
Sorry, i forgot to say;
My printer is a complete rework of a hellobeeprusa. Like pinting volume of 280x280,
Bltouch bed level sensor and so on.
You can not download any firmware and just flash. You have to configure it. Example: you flash a firmware, move the x in x+ direction and it goes wrong. Then you have to adjust the settings so it moves in the right direction. You can not flash my firmware without configure it like your printer. My firmware is only a example for your printer. 
Sorry my english skills are to bad to explain you what i mean exactly...

wallacezhang, 2017-03-29:
yeah, I have checked your firmware before updating. 
I also have a BLTouch on my printer, and the axis direction are also same as yours, so it doesn't need to change firmware.

wallacezhang, 2017-03-29:
here are the errors Repetier Host gives me when printing from computer directly:

[image]

[image]

wallacezhang, 2017-03-30:
ok, thanks. @thinkyhead could you find any clue?

wallacezhang, 2017-03-30:
@gsart84 as I see from records, you also had the issue of slowly moving before, how did you fix it?

gsart84, 2017-03-30:
The biggest problem with slow movement after g28 is away now. This was caused using m605 s0. It was fixed by thinkyhead with rc8. Only the movements after print is finished after g28 x0 y0 is not solved

gsart84, 2017-03-30:
@wallacezhang try this:
Start printer
-g28
-M605 S1
-M605 S0
And then try to move the printer
In this way i solved the problem with slow movements

wallacezhang, 2017-03-30:
it seems @thinkyhead is too busy, could you send your workable firmware to my email: wallace.zhang@yahoo.com, I'm afraid I have downloaded wrong firmware, otherwise, it should be also workable in my printer. right?

gsart84, 2017-03-30:
@wallacezhang the firmware you downloaded from my github is the newest i'm using. I uploaded it a few days before. Have you tryed download the newest rcbugfix and configure it like my firmware?

[41 more replies]