MarlinFirmware/Marlin #4695: Dual_X_Carriage not respecting soft Z max value

AndrewDiehl, 2016-08-23:
Homing machine to Z max value (z=300)

Issue a M605 S1 to put in autopark mode

When changing from T0 to T1 and back, the Z lift exceeds the Zmax value and crashes the end stop into the top of the machine.

thinkyhead, 2016-08-24:
Probably this use-case was simply never considered. We'll have to fix it.

boelle, 2016-09-10:
@thinkyhead will this one be fixed in next rcbugfix?

thinkyhead, 2016-09-10:
The "next RCBugFix" was 10 minutes ago as I added a few cleanup commits. I have a few other fish to fry before I get to work on this specific issue.

Anyway, once I find some time to work on it, with testing help from @AndrewDiehl I'm sure we'll have this tackled in no time.

boelle, 2016-11-02:
@thinkyhead think there will be time for this old one?

thinkyhead, 2016-11-03:
This should now be fixed in `RCBugFix` with the merger of the above PR. Please confirm @AndrewDiehl when you have a chance.

AndrewDiehl, 2016-11-06:
@thinkyhead 

Finally got everything updated and tested. Same issue persists. 

I am around most of the day. I can do google hangouts using landruthegreat@gmail.com or you can call 610-295-7867. For voice I may have to call back using wi-fi since my house has poor signal.

thinkyhead, 2016-11-07:
@AndrewDiehl Oops. I applied the fix for `tool_change` but not for `G28`. I'm including the patch for this in #5175.

thinkyhead, 2016-11-19:
@AndrewDiehl I'll try and reach you over the weekend some time. It will be good to get these Dual-X issues finally cleared and foist RC8 onto the world.

thinkyhead, 2016-11-21:
@AndrewDiehl Concerning the duplication mode behavior. Does my PR make it worse? Or is it just as bad without it?

AndrewDiehl, 2016-11-23:
Current behavior on duplication mode:

M605 S2 //nothing happens
T0 behaves as it would under the M605 S0 mode
Call T1, behaves like it would under M605 S0 mode
Call T0 again, heads are linked together, but not at the set distance apart.

AndrewDiehl, 2016-11-23:
Nevermind, I forgot to home after calling the M605 S2. Seems to be working correctly.

boelle, 2016-11-23:
@AndrewDiehl so this one and https://github.com/MarlinFirmware/Marlin/issues/4694 is solved?

landodragon141, 2016-11-27:
@thinkyhead Is this one resolved?

thinkyhead, 2016-11-27:
This is resolved in a PR that hasn't been merged yet.

thinkyhead, 2016-11-27:
Actually, yes. The main issue in this is resolved. Some of the other things later added are not.

thinkyhead, 2016-11-27:
Actually, yes. The main issue in this is resolved. Some of the other things later added are not. The notes on PR #5175 indicate what I **_tried_** to resolve. The Duplication Mode issue may not actually exist, _**except**_ in that PR — i.e., something I changed while trying to fix one thing might have broken it. (You can see why I'm hesitant to touch this code when you look at it. It's not very well organized or documented. In fact, _documenting_ it liberally with comments would be a useful first step.)

closed 2016-12-06 by thinkyhead