MarlinFirmware/Marlin #4696: analogRead or extend Temperature::isr?

thinkyhead, 2016-08-24:
I need to add support for reading (yet another) analog sensor to Marlin. I notice analogRead is avoided, and I presume this is because it eats up too much time if you call it directly. So, is the ADC reading code in `Temperature::isr` the proper way to do it, and should I just add my new sensor there?

Also, the pin I want to use is defined (in the code of a Marlin fork) simply as "A3" – presumably from `fastio.h`. If I want to implement this for more than one board, how do I translate A3 to the specific pin number like we use in the pins files?

Roxy-3D, 2016-08-24:
analogRead() looks like this:
```cppp

int analogRead(uint8_t pin)
{
	uint8_t low, high;
#if defined(__AVR_ATmega1280__) || defined(__AVR_ATmega2560__)
	if (pin >= 54) pin -= 54; // allow for channel or pin numbers
#elif defined(__AVR_ATmega32U4__)
	if (pin >= 18) pin -= 18; // allow for channel or pin numbers
#elif defined(__AVR_ATmega1284P__) || defined(__AVR_ATmega644P__)
	if (pin >= 24) pin -= 24; // allow for channel or pin numbers
#else
	if (pin >= 14) pin -= 14; // allow for channel or pin numbers
#endif
#if defined(__AVR_ATmega32U4__)
	pin = analogPinToChannel(pin);
	ADCSRB = (ADCSRB & ~(1 << MUX5)) | (((pin >> 3) & 0x01) << MUX5);
#elif defined(ADCSRB) && defined(MUX5)
	// the MUX5 bit of ADCSRB selects whether we're reading from channels
	// 0 to 7 (MUX5 low) or 8 to 15 (MUX5 high).
	ADCSRB = (ADCSRB & ~(1 << MUX5)) | (((pin >> 3) & 0x01) << MUX5);
#endif
	// set the analog reference (high two bits of ADMUX) and select the
	// channel (low 4 bits).  this also sets ADLAR (left-adjust result)
	// to 0 (the default).
#if defined(ADMUX)
	ADMUX = (analog_reference << 6) | (pin & 0x07);
#endif
	// without a delay, we seem to read from the wrong channel
	//delay(1);
#if defined(ADCSRA) && defined(ADCL)
	// start the conversion
	sbi(ADCSRA, ADSC);
	// ADSC is cleared when the conversion finishes
	while (bit_is_set(ADCSRA, ADSC));
	// we have to read ADCL first; doing so locks both ADCL
	// and ADCH until ADCH is read.  reading ADCL second would
	// cause the results of each conversion to be discarded,
	// as ADCL and ADCH would be locked when it completed.
	low  = ADCL;
	high = ADCH;
#else
	// we dont have an ADC, return 0
	low  = 0;
	high = 0;
#endif
	// combine the two bytes
	return (high << 8) | low;
}
```
A to D conversions always take a bunch of time.   (as you know!)   The heart of the routine is here:
```cpp
	// start the conversion
	sbi(ADCSRA, ADSC);
	// ADSC is cleared when the conversion finishes
	while (bit_is_set(ADCSRA, ADSC));
```
If you can kick of the start of the conversion with the sbi() and do other useful work for a while....
When you do a:
```cpp
	while (bit_is_set(ADCSRA, ADSC));
```
you won't have wasted a bunch of processing cycles!

Roxy-3D, 2016-08-24:
>  If I want to implement this for more than one board, how do I translate A3 to the specific pin number like we use in the pins files?

The analog pin numbers are bigger than the digital I/O pin numbers.   If you want, I can put a wire on A3 and get you the digital pin # (using M43 in the devel_ubl branch).   But for the Analog Library stuff, I suspect you want to use the A3 symbol.   (I have switches on Aux-3 analog lines I'm using for debugging UBL.  The digital I/O pin numbers are 64 and 65 but they are connected to A11 and A12)

thinkyhead, 2016-08-24:
Reading the actual source code is a good way to get insights! I would do more of that, but I'm so busy it's crazy. So it looks like the `Temperature::isr` is where I would add support for another sensor, since it already does the kick-off now, read later strategy.

Roxy-3D, 2016-08-24:
> So it looks like the Temperature::isr is where I would add support for another sensor, since it already does the kick-off now, read later strategy.

I think I would try that.   If you do that...  Your temperature reading for the sensor will always be one reading behind.  But temperature changes so slowly, I doubt it matters.   (You would at most points in time have a conversion in process.  But instead of waiting for it to complete so you have the latest and most correct value, you would be using the 'last' value that got converted.  )

At: https://www.arduino.cc/en/Reference/AnalogRead  it says that the conversion process takes .1 ms to read an analog input.  That doesn't seem that bad.  It may not be worth all the complexity to kick off a conversion and then come back later to read it.

thinkyhead, 2016-08-24:
> Your temperature reading for the sensor

It just needs to read a resistor integrated into a tool-head to identify the tool, so that won't be a factor.

madhunm, 2017-01-16:
Is there a possibility to add code that changes the analog reference value from AVCC to the internal 1.1 reference when TEMP_SENSOR 147 is selected?

With the current AVCC reference, the current resolution is ~9 degrees. This defeats the purpose of using a high resolution sensor like a PT100.

With the reference set to the internal 1.1V reference, the resolution will become ~1.5 degrees.

From the 2560 datasheet, setting ADMUX.REFS1 = 1 and ADMUX.REFS0 = 0 will set the reference to the internal 1.1V reference

thinkyhead, 2017-02-26:
@madhunm Do we have a separate feature request for that change? I'm closing this issue as it is resolved related to my project.

Would we need to have a new `thermistortable_147.h` if we alter the reference voltage?

closed 2017-02-26 by thinkyhead