MarlinFirmware/Marlin #4697: G29 values save to eeprom on AUTO_BED_LEVELING_FEATURE

oxivanisher, 2016-08-24:
Is it correct, that if `AUTO_BED_LEVELING_FEATURE` is used, the values can not be stored to EEPROM and the bed leveling has to be run before every print?

If so, the problems with printers which are bed leveling with a nozzle probe are:
* The bed gets dirtied up just before the print (the nozzle needs to be hot to get a clean Z reading)
* Starting a clean print is more difficult, because the nozzle needs to be primed somehow
* It takes more time (not really a problem)

Roxy-3D, 2016-08-24:
Which branch are you asking about?   If you are using the devel_ubl branch, this issue is anticipated.   In that case the last High Resolution Mesh in use is loaded automatically, and if you specify a G29 P0 T it will do a 3-Point leveling with it.

I'm working on speeding up the mesh_buffer_line() routine at the moment.   But coming up very soon will be the Grid Leveling for the High Resolution Mesh.   Until then, you can use the 3-Point Leveling if you have a problem with dirt getting between the bed and the glass.

thinkyhead, 2016-08-24:
> the values can not be stored to EEPROM 

That is correct. It's a feature request that has not yet been implemented in the main project, but has been in Roxy's Universal Bed Leveling and some other forks (such as Josef Prusa's and perhaps in MarlinKimbra). We need to come up with GCodes to go along with the feature, also. Probably we'll use the same codes as Repetier Firmware.

closed 2017-10-31 by thinkyhead