MarlinFirmware/Marlin #4698: RC7 - MINTEMP not triggered

AxMod3DPrint, 2016-08-24:
Hi,

I've just put RC7 on one of my Migbot Prusa i3s. While swapping out the layer fan I managed to break the thermistor lead, however MINTEMP wasn't triggered. I can't see any particular updates since it was downloaded on the 13th, I've just got around to configuring it now.

All thermal protections are set.

Configs:
http://pastebin.com/JKnEyqvr - configuration.h
http://pastebin.com/vva6eLCQ - configuration_adv.h

Machine runs a MKS Base 1.2, but is set to RAMPS 1.4 EFB as it's quicker to configure the extruder fan using pin 7 that way (E1 on the MKS). 

I can't see any issues with the config and apart from that the printer is working fine.

thinkyhead, 2016-08-26:
We received a request not to throw the error until you set the temperature to non-zero. See if you get an error when you set the temperature.

esenapaj, 2016-10-14:
I've perceived that MINTEMP Error isn't threw when any thermistor isn't connected, since RC7. RC6 throw error.
Is this intended behavior?

Blue-Marlin, 2016-10-14:
Yes. That was one of the changes made necessary by the support for high temperature thermistors.
https://github.com/MarlinFirmware/Marlin/issues/4698#issuecomment-242618560

esenapaj, 2016-10-14:
high temp... ah, maybe this. https://github.com/MarlinFirmware/Marlin/pull/4244 (Dyze High Temp Thermistor Support)
I see, thanks for the teaching.

Blue-Marlin, 2016-10-14:
https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/temperature.cpp#L1783-L1792`
and the folowing bloks for the other heaters.
__&& target_temperature[0] > 0.0f__

Blue-Marlin, 2016-10-15:
Very ugly code.
Maybe something like:
```
int temp_direction[EXTRUDERS] = {
#if HEATER_0_RAW_LO_TEMP > HEATER_0_RAW_HI_TEMP
  -1,
#else
1,
#endif

#if HAS_TEMP_1 && HOTENDS > 1
  #if HEATER_1_RAW_LO_TEMP > HEATER_1_RAW_HI_TEMP
    -1,
  #else
    1,
  #endif
#endif
...
}

for (int8_t i=0; i< HOTENDS; i++) {
      if (temp_direction[i]*current_temperature_raw[0] >= temp_direction[i]*maxttemp_raw[0]) max_temp_error(0);
        if (temp_direction[i]*minttemp_raw[0] >= temp_direction[i]*current_temperature_raw[0] && !is_preheating(0) && target_temperature[0] > 0.0f) {
...
}
```
could be less ugly.

thinkyhead, 2016-10-22:
@Blue-Marlin I like your solution to the ugly code. I might have to go that way!

closed 2017-10-31 by thinkyhead