MarlinFirmware/Marlin #4699: Character Encoding Issues in UBL Dev Branch

psavva, 2016-08-25:
@Roxy-3D Please have a look though all the commits done on UBL, It seems you have a character encoding issue, when it comes to some special characters.

Mariln_main.cpp
Example:
 *    ÃƒÂ¢Ã¢â€šÂ¬Ã‚Â¢ temperature
 *    ÃƒÂ¢Ã¢â€šÂ¬Ã‚Â¢ planner
 *    ÃƒÂ¢Ã¢â€šÂ¬Ã‚Â¢ watchdog
 *    ÃƒÂ¢Ã¢â€šÂ¬Ã‚Â¢ stepper
 *    ÃƒÂ¢Ã¢â€šÂ¬Ã‚Â¢ photo pin
 *    ÃƒÂ¢Ã¢â€šÂ¬Ã‚Â¢ servos
 *    ÃƒÂ¢Ã¢â€šÂ¬Ã‚Â¢ LCD controller
 *    ÃƒÂ¢Ã¢â€šÂ¬Ã‚Â¢ Digipot I2C
 *    ÃƒÂ¢Ã¢â€šÂ¬Ã‚Â¢ Z probe sled
 *    ÃƒÂ¢Ã¢â€šÂ¬Ã‚Â¢ status LEDs
 *    ÃƒÆ’Ã‚Â¢ÃƒÂ¢Ã¢â‚¬Å¡Ã‚Â¬Ãƒâ€šÃ‚Â¢ temperature
 *    ÃƒÆ’Ã‚Â¢ÃƒÂ¢Ã¢â‚¬Å¡Ã‚Â¬Ãƒâ€šÃ‚Â¢ planner
 *    ÃƒÆ’Ã‚Â¢ÃƒÂ¢Ã¢â‚¬Å¡Ã‚Â¬Ãƒâ€šÃ‚Â¢ watchdog
 *    ÃƒÆ’Ã‚Â¢ÃƒÂ¢Ã¢â‚¬Å¡Ã‚Â¬Ãƒâ€šÃ‚Â¢ stepper
 *    ÃƒÆ’Ã‚Â¢ÃƒÂ¢Ã¢â‚¬Å¡Ã‚Â¬Ãƒâ€šÃ‚Â¢ photo pin
 *    ÃƒÆ’Ã‚Â¢ÃƒÂ¢Ã¢â‚¬Å¡Ã‚Â¬Ãƒâ€šÃ‚Â¢ servos
 *    ÃƒÆ’Ã‚Â¢ÃƒÂ¢Ã¢â‚¬Å¡Ã‚Â¬Ãƒâ€šÃ‚Â¢ LCD controller
 *    ÃƒÆ’Ã‚Â¢ÃƒÂ¢Ã¢â‚¬Å¡Ã‚Â¬Ãƒâ€šÃ‚Â¢ Digipot I2C
 *    ÃƒÆ’Ã‚Â¢ÃƒÂ¢Ã¢â‚¬Å¡Ã‚Â¬Ãƒâ€šÃ‚Â¢ Z probe sled
 *    ÃƒÆ’Ã‚Â¢ÃƒÂ¢Ã¢â‚¬Å¡Ã‚Â¬Ãƒâ€šÃ‚Â¢ status LEDs

First time it was changed:
 *    • temperature
 *    • planner
 *    • watchdog
 *    • stepper
 *    • photo pin
 *    • servos
 *    • LCD controller
 *    • Digipot I2C
 *    • Z probe sled
 *    • status LEDs
 *    â€¢ temperature
 *    â€¢ planner
 *    â€¢ watchdog
 *    â€¢ stepper
 *    â€¢ photo pin
 *    â€¢ servos
 *    â€¢ LCD controller
 *    â€¢ Digipot I2C
 *    â€¢ Z probe sled
 *    â€¢ status LEDs

Was in revision: 23af5a826c75e4ffd19611dd7683828a66e14341

I some this in some other places as well, check the configuration file, after the 300
edc99da2a68a64f9ddfd256f048861236b73e0cc

// 13 is 100k Hisens 3950  1% up to 300°C for hotend "Simple ONE " & "Hotend "All In ONE"
// 13 is 100k Hisens 3950  1% up to 300Â°C for hotend "Simple ONE " & "Hotend "All In ONE"

Roxy-3D, 2016-08-25:
Is that being caused by my text editor?   Once we get this fixed, we don't want it to happen again!   
To keep it from happening again, should we not be using non-ascii characters?

psavva, 2016-08-25:
Have a look at this.
http://stackoverflow.com/questions/5166652/how-to-view-utf-8-characters-in-vim-or-gvim

Consider that special and non ASCII characters are supported and should be displayed correctly.

Even in the language files, we use UTF8 and even Greek characters :)

Roxy-3D, 2016-08-25:
> Consider that special and non ASCII characters are supported and should be displayed correctly.
> Even in the language files, we use UTF8 and even Greek characters :)

OK...    I hear you!   I'll read the article and see if I can get Vim to handle the special characters.

Roxy-3D, 2016-08-25:
@psavva    I have Vim defaulting to use UTF-8.   That seems to allow me to pull in files and write them back out without problems.    I have a few more things to clean up in the UBL branch, but over the next few days those encoding errors should go away.    If you see any encoding errors a week from now, please let me know.  I'll get them fixed.

thinkyhead, 2016-08-26:
Perhaps you can configure your editor to simply respect the existing encoding of the file and tell it to make sure to preserve it on save…?

Roxy-3D, 2016-08-26:
> Perhaps you can configure your editor to simply respect the existing encoding of the file and tell it to make sure to preserve it on save…?

I thought I was doing good to get Vim to treat the file as UTF-8.

psavva, 2017-02-19:
I guess i can close this issue @Roxy3D ?

Roxy-3D, 2017-02-19:
I don't know?  Does something have to be edited?

closed 2017-02-20 by psavva