MarlinFirmware/Marlin #4700: sd init fail simple solution for slow sd cards

akaJes, 2016-08-25:
just insert after
```
      if (sd_status) {
        card.initsd();

```
https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/ultralcd.cpp#L2546
`if (!card.cardOK) card.initsd();`
double card initialization resolve it and you no need restart after each card insertion

thinkyhead, 2016-08-26:
Does adding a small delay after the card has been inserted also work?
```diff
      if (sd_status) {
+       delay(100);
        card.initsd();
```

akaJes, 2016-08-26:
i tried to place a delay inside initsd everywhere and it didn't results
timeout extention also didnt, because watchdog reset cpu and wait more than 4 minutes no reason
also tried set SPI_SD_INIT_RATE 6 and no results

tried your code and it does about 50% success
my code works about 80-90%
maybe there is some troubles with deinitializadion SPI or init timeout

thinkyhead, 2016-08-26:
@akaJes Do you find that the problem goes away if you do a **low-level format** on the SD card?

akaJes, 2016-08-26:
@thinkyhead sd card is too old, maybe 2 class or 4 - i dont know, because it marked as Kingston SD-C02G
and its realy slow
i tried only ordinary formatting with Gparted fat32 lbs

akaJes, 2016-08-29:
all tries was unsuccessfull
i checked my card and it is First class!! with 1.2 MB/s read/write speed
and my code works only when less than ten files on the card
i decided to buy new card and don't fight with marlin

in finale - do not use slow cards :)

thinkyhead, 2016-08-30:
@akaJes You could probably still use the card once in a while. Just send `M21` after you insert it, and then wait a few seconds before trying to read it.

thinkyhead, 2016-08-30:
There's always this ugly solution, too…
```cpp
if (sd_status) {
  slow_card_timeout = millis() + 5000;
  do {
    card.initsd();
    thermalManager.manage_heater();
  } while (!card.cardOk && PENDING(millis(), slow_card_timeout));
```

dagnall53, 2017-05-11:
I have just seen same issue with a new controller set up I have just programmed with latest version ( CONFIGURATION_H_VERSION 010100). 
I have three SD cards that all work in two other Marlin printers. (slightly older software) but with the new setup, two cards refuse to init when inserted. I changed the graphic panel etc, but same effects remained. 
However: by removing the SD card detect feature (for me, find the Pin define in the ramps .h file and comment out), the "CHANGE SD card" and "Init SD card" menu items are installed and using these, all the cards are read and initiate (on command) ok.

I tried the add delay(100): This was not a full answer,  the cards were initiated (with SD detect pin switched back on) sometimes, but it was not a complete solution.

I also tried the "ugly solution" ..
( nb typo needs correction "!card.cardOK" ..  and you need to add the definbition "unsigned long slow_card_timeout; "  after "void lcd_update() ")
This also works for all 3 cards, although it is not completely reliable, - sometimes I have had to remove the SD card and reinstall, but it works most of the time with all 3 test SD cards, but is not a complete answer. 

For me, I am leaving the SD detect pin undefined, so as to add the manual menu items, and am leaving in the "ugly" code as it may help.
Cheers
Dagnall

closed 2018-02-06 by thinkyhead