MarlinFirmware/Marlin #4701: [BUG] Freeze during Z axis homing with RC7

theodroid, 2016-08-25:
Hi,
After updated Marlin to RC7, a bug appeared when homing the Z axis.
When I send a G28 command, the X and Y axis homes without problems, and next, the build plate starts to go up. But if the bed is too far away the extruder, the Z axis stops to move and the machine freezes (and nothing happens if I press the knob of the LCD). So I have to turn it off and on again.
However, if the bed is enough close to the extruder, the Z probe detects it quickly so the 3D printer doesn’t freeze.

My 3D printer is an Ulticampy 3, similar to an Ultimaker but with different electronics. I’m using a SainSmart 2-in-1 board, with a Full Graphic LCD screen. My Z-probe is a 12mm capacitive sensor. Also I have a Raspberry Pi 3 with OctoPrint to run prints. The problem happens no matter the way I home the axes: if I select Auto home from the LCD, if I send a G28 command from my computer, or even at the beginning of a print.

So when I try to home axes: first, the bed goes down 1mm, the X and Y axes homes, the extruder goes to the center of the machine, next, the bed go down 10mm, and finally the bed goes up. It’s raised for about 26mm (so 15mm upper than before homing) but the machine freezes at this moment.

So for the moment my solution is to raise up the bed close enough to the extruder, turn the printer off and on again (to reset its positon) and next run a print… but that’s annoying…
I tried to install the RCBugFix version and it doesn’t change anything. Previously I was using RC6 and this bug didn’t exist.

I attach my configuration.h and configuration_adv.h files and a video to show the problem.

Regards
[Video and config files.zip](https://github.com/MarlinFirmware/Marlin/files/437085/Video.and.config.files.zip)

thinkyhead, 2016-08-26:
Peculiar. Do you see the same issue with the current `RCBugFix` code?

thinkyhead, 2016-08-26:
Some commentary on your configuration:
```cpp
#define X_PROBE_OFFSET_FROM_EXTRUDER 38  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 38  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER 0   // Z offset: -below +above  [the nozzle]
```
I do believe that your probe is 38mm diagonally mounted from your nozzle, but I don't quite trust that your probe triggers only when the nozzle is touching the bed. More likely it triggers at some point when the nozzle is above the bed, yes? If your probe triggers when the nozzle is 10mm above the bed, then your `Z_PROBE_OFFSET_FROM_EXTRUDER` would be set to -10.

```cpp
#define Z_PROBE_DEPLOY_HEIGHT   1 // Z position for the probe to deploy/stow
#define Z_PROBE_TRAVEL_HEIGHT  10 // Z position for travel between points, default is 5
```
If, as per my example, the probe triggers when the nozzle is 10mm above the bed — making your `Z_PROBE_OFFSET_FROM_EXTRUDER` -10 — then your `Z_PROBE_DEPLOY_HEIGHT` would need to be 10 **plus** the additional height that you want to raise it. The same applies to `Z_PROBE_TRAVEL_HEIGHT`. So, for example, you might set these to 15 for a probe that triggers when the nozzle is 10mm above the bed, so the probe will be 5mm above where it might trigger.

theodroid, 2016-08-27:
My probe detects the bed when the nozzle is 3mm above it.
So I have corrected my configuration.h file, now I have `Z_PROBE_OFFSET_FROM_EXTRUDER` set to -3
and `Z_PROBE_DEPLOY_HEIGHT` and `Z_PROBE_TRAVEL_HEIGHT` set to 8.

However, the bug is always here… When I send G28, the X and Y axes homes correctly, the extruder
goes to the center of the bed, the bed goes down 8 mm, and goes up for about 27 mm. It seems to be
always at the same distance that the machine freezes.

I tried the latest RCBugFix and the machine freezes too.

thinkyhead, 2016-08-28:
At this point it will help to see some logging of the home procedure. Enable `DEBUG_LEVELING_FEATURE` and re-flash the firmware. Before doing `G28` enter `M111 S255` and then clear the log. After `G28` post the log here so we can have a look.

theodroid, 2016-08-28:
I downloaded and flashed the latest RCBugFix, with `DEBUG_LEVELING_FEATURE` enabled.
Next, I sent `M111 S255` and `G28`. This is the log from OctoPrint:

```
Send: M111 S255
Recv: echo:DEBUG:ECHO,INFO,ERRORS,DRYRUN,COMMUNICATION,LEVELING
Recv: ok
[...]
Recv: echo:M105
[...]
Send: G28
Recv: echo:G28
Recv: >>> gcode_G28
Recv: current_position=(0.00, 0.00, 0.00) : setup_for_endstop_or_probe_move
Recv: > endstops.enable(true)
Recv: Raise Z (before homing) to 1.00
Recv: >>> do_blocking_move_to(0.00, 0.00, 1.00)
Recv: current_position=(0.00, 0.00, 1.00) : sync_plan_position
Recv: >>> do_blocking_move_to(-319.50, -330.00, 1.00)
Recv: >>> homeaxis(X)
Recv: current_position=(0.00, 0.00, 1.00) : sync_plan_position
Recv: > 1st Home-319.50
Recv: current_position=(0.00, 0.00, 1.00) : sync_plan_position
Recv: current_position=(0.00, 0.00, 1.00) : sync_plan_position
Recv: > 2nd Home-10.00
Recv: >>> set_axis_is_at_home(X)
Recv: For X axis:
Recv:  home_offset = 0.00
Recv:  position_shift = 0.00
Recv:  soft_endstop_min = 0.00
Recv:  soft_endstop_max = 213.00
Recv: > home_offset[X] = 0.00
Recv: current_position=(0.00, 0.00, 1.00) :
Recv: <<< set_axis_is_at_home(X)
Recv: current_position=(0.00, 0.00, 1.00) : sync_plan_position
Recv: current_position=(0.00, 0.00, 1.00) : > AFTER set_axis_is_at_home
Recv: <<< homeaxis(X)
Recv: current_position=(0.00, 0.00, 1.00) : > homeX
Recv: >>> homeaxis(Y)
Recv: current_position=(0.00, 0.00, 1.00) : sync_plan_position
Recv: > 1st Home-330.00
Recv: current_position=(0.00, 0.00, 1.00) : sync_plan_position
Recv: current_position=(0.00, 0.00, 1.00) : sync_plan_position
Recv: echo:busy: processing
Recv: > 2nd Home-10.00
Recv: >>> set_axis_is_at_home(Y)
Recv: For Y axis:
Recv:  home_offset = 0.00
Recv:  position_shift = 0.00
Recv:  soft_endstop_min = 0.00
Recv:  soft_endstop_max = 220.00
Recv: > home_offset[Y] = 0.00
Recv: current_position=(0.00, 0.00, 1.00) :
Recv: <<< set_axis_is_at_home(Y)
Recv: current_position=(0.00, 0.00, 1.00) : sync_plan_position
Recv: current_position=(0.00, 0.00, 1.00) : > AFTER set_axis_is_at_home
Recv: <<< homeaxis(Y)
Recv: current_position=(0.00, 0.00, 1.00) : > homeY
Recv: > Z_SAFE_HOMING >>>
Recv: current_position=(0.00, 0.00, 1.00) : sync_plan_position
Recv: current_position=(0.00, 0.00, 1.00) : > Z_SAFE_HOMING > home_all_axis
Recv: destination=(68.00, 72.00, 1.00) : > Z_SAFE_HOMING > home_all_axis
Recv: >>> do_blocking_move_to(68.00, 72.00, 1.00)
Recv: >>> homeaxis(Z)
Recv: > current_position=(68.00, 72.00, 1.00) : set_probe_deployed
Recv: deploy: 1
Recv: do_probe_raise(8.00)
Recv: >>> do_blocking_move_to(68.00, 72.00, 8.00)
Recv: >>> do_blocking_move_to(68.00, 72.00, 8.00)
Recv: current_position=(68.00, 72.00, 0.00) : sync_plan_position
```

thinkyhead, 2016-08-28:
Hmm. That last line concerns me. It gives the appearance that the last raise isn't being transmitted to the current position. I'll examine the code later on and see if it is just that.

thinkyhead, 2016-08-28:
Actually the log seems to be incomplete. It ends right after deploying the probe, without any z homing detail. Try again?

theodroid, 2016-08-28:
Ok. This is the new one:

```
Send: M111 S255
Recv: echo:DEBUG:ECHO,INFO,ERRORS,DRYRUN,COMMUNICATION,LEVELING
Recv: ok
[...]
Recv: echo:M105
[...]
Recv: echo:M105
[...]
Recv: echo:M105
[...]
Send: G28
Recv: echo:G28
Recv: >>> gcode_G28
Recv: current_position=(0.00, 0.00, 0.00) : setup_for_endstop_or_probe_move
Recv: > endstops.enable(true)
Recv: Raise Z (before homing) to 1.00
Recv: >>> do_blocking_move_to(0.00, 0.00, 1.00)
Recv: current_position=(0.00, 0.00, 1.00) : sync_plan_position
Recv: >>> do_blocking_move_to(-319.50, -330.00, 1.00)
Recv: >>> homeaxis(X)
Recv: current_position=(0.00, 0.00, 1.00) : sync_plan_position
Recv: echo:busy: processing
Recv: > 1st Home-319.50
Recv: current_position=(0.00, 0.00, 1.00) : sync_plan_position
Recv: current_position=(0.00, 0.00, 1.00) : sync_plan_position
Recv: > 2nd Home-10.00
Recv: >>> set_axis_is_at_home(X)
Recv: For X axis:
Recv:  home_offset = 0.00
Recv:  position_shift = 0.00
Recv:  soft_endstop_min = 0.00
Recv:  soft_endstop_max = 213.00
Recv: > home_offset[X] = 0.00
Recv: current_position=(0.00, 0.00, 1.00) :
Recv: <<< set_axis_is_at_home(X)
Recv: current_position=(0.00, 0.00, 1.00) : sync_plan_position
Recv: current_position=(0.00, 0.00, 1.00) : > AFTER set_axis_is_at_home
Recv: <<< homeaxis(X)
Recv: current_position=(0.00, 0.00, 1.00) : > homeX
Recv: >>> homeaxis(Y)
Recv: current_position=(0.00, 0.00, 1.00) : sync_plan_position
Recv: > 1st Home-330.00
Recv: current_position=(0.00, 0.00, 1.00) : sync_plan_position
Recv: current_position=(0.00, 0.00, 1.00) : sync_plan_position
Recv: > 2nd Home-10.00
Recv: >>> set_axis_is_at_home(Y)
Recv: For Y axis:
Recv:  home_offset = 0.00
Recv:  position_shift = 0.00
Recv:  soft_endstop_min = 0.00
Recv:  soft_endstop_max = 220.00
Recv: > home_offset[Y] = 0.00
Recv: current_position=(0.00, 0.00, 1.00) :
Recv: <<< set_axis_is_at_home(Y)
Recv: current_position=(0.00, 0.00, 1.00) : sync_plan_position
Recv: current_position=(0.00, 0.00, 1.00) : > AFTER set_axis_is_at_home
Recv: <<< homeaxis(Y)
Recv: current_position=(0.00, 0.00, 1.00) : > homeY
Recv: > Z_SAFE_HOMING >>>
Recv: current_position=(0.00, 0.00, 1.00) : sync_plan_position
Recv: current_position=(0.00, 0.00, 1.00) : > Z_SAFE_HOMING > home_all_axis
Recv: destination=(68.00, 72.00, 1.00) : > Z_SAFE_HOMING > home_all_axis
Recv: >>> do_blocking_move_to(68.00, 72.00, 1.00)
Recv: >>> homeaxis(Z)
Recv: > current_position=(68.00, 72.00, 1.00) : set_probe_deployed
Recv: deploy: 1
Recv: do_probe_raise(8.00)
Recv: >>> do_blocking_move_to(68.00, 72.00, 8.00)
Recv: echo:busy: processing
Recv: >>> do_blocking_move_to(68.00, 72.00, 8.00)
Recv: current_position=(68.00, 72.00, 0.00) : sync_plan_position
```
After the last line, the machine sends nothing anymore because she is freezed, just like when you send `M112` command.

thinkyhead, 2016-08-29:
@theodroid I figured out why the probe move was too short and just patched it. Give the current `RCBugFix` another try and see if I fixed that part of the issue.

theodroid, 2016-08-29:
I downloaded and flashed the latest RCBugFix but unfortunately the issue is always here...
I also tried to enable `PROBE_DOUBLE_TOUCH` but it didn't change anything...

thinkyhead, 2016-08-30:
Ok, so of course the reason your bed goes down 1mm is because of…
```cpp
#define Z_HOMING_HEIGHT 1
```
…so that's normal.

The next thing to consider is the probe deployment. For that the Z axis wants to move to what is considered to be 8mm above the bed (not 10mm). So it moves the bed down another 8mm and at that point enables the probe. Again, all normal.

Next comes the actual probe. For that the bed needs to move up for as long as it takes to reach and trigger the probe. That seems to start out alright, but at some point you get the freeze. I think I might have some ideas….

I notice that your Z axis has 6400 steps per millimeter, and your Z homing speed is 350 mm/m or just over 5.8mm/s. In order to sustain this feedrate the stepper functions need to perform 37,333 steps per second on the Z axis alone. On a 16MHz processor, this means the stepper ISR only has about 442 cycles before it starts to overlap with itself and block all other activity. As soon as your display needs to update, plus the temperature needs to be read, plus the main thread needs to do its work, basically there's no time left and I suspect the CPU just can't keep up.

So I have two suggestions. One is to lower your Z homing feedrate. The other, which may be even more helpful, is to reconfigure your Z stepper driver with half as many microsteps. Or maybe even 1/4 as many. So, if you're using a DRV8825 at 64x microstepping, change it to 32 or even 16 by removing the appropriate jumpers. Before going that far, of course try lowering the Z homing feedrate to 180 to see if it stops the freeze from occurring. If it does, then reducing the microsteps will allow you to go back to 350 and get a better Z homing speed.

theodroid, 2016-08-30:
Yes, it works!!!!
I changed the microstepping from 1/32 to 1/16, and divided by 2 the Z steps per mm, and now the Z axis homes correctly!
Thanks a lot!

thinkyhead, 2016-08-30:
> stepper functions need to perform 37,333 steps per second on the Z axis alone

@Blue-Marlin Now that we know we can use `TCNT0` inside the stepper ISR, it occurs to me that we could use that to throttle the speed down when we notice the stepper ISR is in danger of overlapping on itself. Pretty simple idea…

```cpp
static uint32_t last_isr_end_time = (uint32_t)(-1000);
. . .
// at the start of the ISR
uint32_t isr_leeway = TCNT0 - last_isr_end_time;
if (isr_leeway < 100) reduce_speed = true; else if (isr_leeway > 500) reduce_speed = false;
. . .
// at the end of the ISR
isr_last_end_time = TCNT0;
```

Blue-Marlin, 2016-08-30:
@thinkyhead 
Wasn't 
```
OCR1A = (OCR1A < (TCNT1 + 16)) ? (TCNT1 + 16) : OCR1A;
```
for this case? Is 16 not enough?

thinkyhead, 2016-08-30:
It's hard to say, really, as to the deepest cause of freezing in the case of this issue. The code you point out seems like it might have a real issue anytime that `TCNT1` overflows. And I note that `TCNT1` is never being reset to zero, whereas this code seems to assume it must be.

thinkyhead, 2016-08-30:
Hmm… It's been like this since at least 1.0.2-1 but apparently this commented line might be needed…
```cpp
void st_wake_up() {
  //  TCNT1 = 0;
  ENABLE_STEPPER_DRIVER_INTERRUPT();
}
```

Blue-Marlin, 2016-08-30:
From the ATmega2560 data sheet

> 17.9.2 Clear Timer on Compare Match (CTC) Mode
In Clear Timer on Compare or CTC mode (WGMn3:0 = 4 or 12), the OCRnA or ICRn Register are used to manipulate
the counter resolution. In CTC mode the counter is cleared to zero when the counter value (TCNTn) matches
either the OCRnA (WGMn3:0 = 4) or the ICRn (WGMn3:0 = 12). The OCRnA or ICRn define the top value for the
Table 17-2. Waveform Generation Mode Bit Description(1)
Mode WGMn3
WGMn2
(CTCn)
WGMn1
(PWMn1)
WGMn0
(PWMn0)
Timer/Counter
Mode of Operation TOP
Update of
OCRnx at
TOVn Flag
Set on
00 0 0 0 Normal 0xFFFF Immediate MAX
1 0 0 0 1PWM, Phase Correct, 8-bit 0x00FF TOP BOTTOM
2 0 0 1 0PWM, Phase Correct, 9-bit 0x01FF TOP BOTTOM
3 0 0 1 1PWM, Phase Correct, 10-bit 0x03FF TOP BOTTOM
4 0 1 0 0 CTC OCRnA Immediate MAX
5 0 1 0 1 Fast PWM, 8-bit 0x00FF BOTTOM TOP
6 0 1 1 0 Fast PWM, 9-bit 0x01FF BOTTOM TOP
7 0 1 1 1 Fast PWM, 10-bit 0x03FF BOTTOM TOP
81 0 0 0 PWM, Phase and Frequency
Correct ICRn BOTTOM BOTTOM
91 0 0 1 PWM,Phase and Frequency
Correct OCRnA BOTTOM BOTTOM
10 1 0 1 0 PWM, Phase Correct ICRn TOP BOTTOM
11 1 0 1 1 PWM, Phase Correct OCRnA TOP BOTTOM
12 1 1 0 0 CTC ICRn Immediate MAX
13 1 1 0 1 (Reserved) – – –
14 1 1 1 0 Fast PWM ICRn BOTTOM TOP
15 1 1 1 1 Fast PWM OCRnA BOTTOM TOP
ATmega640/V-1280/V-1281/V-2560/V-2561/V [DATASHEET] 146
2549Q–AVR–02/2014
counter, hence also its resolution. This mode allows greater control of the compare match output frequency. It also
simplifies the operation of counting external events.
The timing diagram for the CTC mode is shown in Figure 17-6. The counter value (TCNTn) increases until a compare
match occurs with either OCRnA or ICRn, and then counter (TCNTn) is cleared.
Figure 17-6. CTC Mode, Timing Diagram
An interrupt can be generated at each time the counter value reaches the TOP value by either using the OCFnA or
ICFn Flag according to the register used to define the TOP value. If the interrupt is enabled, the interrupt handler
routine can be used for updating the TOP value. However, changing the TOP to a value close to BOTTOM when
the counter is running with none or a low prescaler value must be done with care since the CTC mode does not
have the double buffering feature. If the new value written to OCRnA or ICRn is lower than the current value of
TCNTn, the counter will miss the compare match. The counter will then have to count to its maximum value
(0xFFFF) and wrap around starting at 0x0000 before the compare match can occur. In many cases this feature is
not desirable. An alternative will then be to use the fast PWM mode using OCRnA for defining TOP (WGMn3:0 =
15) since the OCRnA then will be double buffered.
For generating a waveform output in CTC mode, the OCnA output can be set to toggle its logical level on each
compare match by setting the Compare Output mode bits to toggle mode (COMnA1:0 = 1). The OCnA value will
not be visible on the port pin unless the data direction for the pin is set to output (DDR_OCnA = 1). The waveform
generated will have a maximum frequency of fOCnA = fclk_I/O/2 when OCRnA is set to zero (0x0000). The waveform
frequency is defined by the following equation:
The N variable represents the prescaler factor (1, 8, 64, 256, or 1024).
As for the Normal mode of operation, the TOVn Flag is set in the same timer clock cycle that the counter counts
from MAX to 0x0000.

Our timer1 is in CTC mode. TCNT1 is reset, to 0, when it reaches OCR1A. If we write a too small value into OCR1A (< TCNT1 at that moment) we have to wait for the overflow (what can last a while).

Blue-Marlin, 2016-08-31:
> what can last a while

Ahhh, not toooooo long. 1kHz = 1ms. Not really nice when stepping - but not a hang.

closed 2016-08-31 by thinkyhead