MarlinFirmware/Marlin #4702: G28, G29 ok but bug when i try to print

tyssoon, 2016-08-25:
when i start a print with repetier directly from usb, and after g28/g29, the effector do another homing and fall, like a reset. here is the m111 log:

[m111 - Copie.txt](https://github.com/MarlinFirmware/Marlin/files/437457/m111.-.Copie.txt)

what am i doing wrong ?

thinkyhead, 2016-08-26:
First, what version of Marlin are you running?

Second, either the host software is sending this, or your slicer is inserting it into the beginning of the generated GCode:
```gcode
M190 S70
M117 FIN 7m 54s
M109 S210
G21
G90
M82
M107
G28 X0 Y0
```
I'm not sure what the effect of `G28 X0 Y0` is on a delta, but what actually seems to be happening is that the firmware is resetting. This usually indicates that something got stuck and the "watchdog timer" is running out and resetting the board. If you disable `USE_WATCHDOG` the machine may not reset, but it might still hang. It's worth testing it without `USE_WATCHDOG` just to see if you get some other kind of crash.

tyssoon, 2016-08-26:
hi @thinkyhead !

im on marlin rc7.

i think it might be because the extruder temperature isnt stable nough.

when i ask 210)C, it raise to +250)C and then slowly go back to 210°C. I might have badly tuned the temperatures on marlin. 

let me try the watchdog thing, i'll tell you the result.

Thanks !

tyssoon, 2016-08-26:
uncommenting watchdog seems to have worked ! thanks !

closed 2016-08-26 by tyssoon

thinkyhead, 2016-08-26:
Thanks for the report. I'm not sure why the watchdog is failing to get reset. There has to be a blockage of 4 seconds before the board will reset. Another thing you could try is to change `WDTO_4S` to `WDTO_8S` in `watchdog.cpp`, which doubles the amount of time before it will reset. It's recommended to keep the watchdog on as a safety feature, because a locked-up firmware could leave the heaters on.