MarlinFirmware/Marlin #4705: Homing with kinematics

thinkyhead, 2016-08-26:
I'm working on a SCARA implementation and I noticed that when homing we're being naïve about DELTA and SCARA. It just happens to work for DELTA because it always homes XYZ at the same time. But for SCARA and future kinematics it should be smarter.

From `G28` the delta procedure homes each of its axes and calls `set_axis_at_home(axis)` after each tower is homed.
- Good: The `set_axis_at_home` function sets the `axis_homed` and `axis_known_position` flags, which is fine. We can treat these flags as pertaining to the ABC towers on a DELTA or to the A (Theta) / B (Phi) / Z actuators on a SCARA.
- Bad: `set_axis_at_home` treats the axis parameter as pertaining to XYZ and sets the X, Y, or Z `current_position` to its configured home position. This needs to be skipped and handled separately in the `G28` function instead.

For `DELTA`, since you always home all axes you just wait till the end and only then set the cartesian position to home.

For `SCARA` you can home Z by itself, but you must always home A and B together.

So… this something I will be working on.

boelle, 2016-11-02:
@thinkyhead have you fixed this one? or is time still being in limited quantities

thinkyhead, 2016-11-03:
Done.

closed 2016-11-03 by thinkyhead