MarlinFirmware/Marlin #4708: Auto bed leveling ignores Z-probe offset

universam1, 2016-08-26:
Marlin now ignores M851 Z-probe offset completely with the latest RCBugFix!
Actually, all EEPROM offset settings are ignored by auto bed leveling, that means X and Y are ignored as well. I could help myself by hardcoding these values, but Z is impossible for me.

I tried RCBugFix since I had the same problem with RC7. On RC6 and before it worked.

I've cleared EEPROM so this is not the reason. 
No matter what I save in EEPROM for M851 or over display, it just will be ignored. Restarted several times.

Please fix since this renders my printer unusable!

Thank you

thinkyhead, 2016-08-26:
How are you seeing this manifested? What are your settings for:
```cpp
#define Z_PROBE_DEPLOY_HEIGHT 15 // Z position for the probe to deploy/stow
#define Z_PROBE_TRAVEL_HEIGHT  5 // Z position for travel between points
```
…and what is your probe Z offset?

Because the above settings must be larger than your probe offset.

Roxy-3D, 2016-08-26:
I have a side question on this topic.   Does anybody know the history of why Mesh Bed Leveling added its own z_offset variable for the Mesh?   

In the initial devel_ubl implementation, I tried to preserve the lowest levels of the Mesh Bed Leveling as much as possible.   But it seems like it would be cleaner to eliminate it and just make use of the zprobe_zoffset (that defaults to Z_PROBE_OFFSET_FROM_EXTRUDER).

brainscan, 2016-08-26:
Maybe because mesh doesn't require a probe so it would have been confusing.

Roxy-3D, 2016-08-26:
That what happens when you get too close to the forest.  You can't see the trees!

Thanks!

thinkyhead, 2016-08-26:
> When you have a few moments

I replied to that comment, but it ended up at https://github.com/MarlinFirmware/Marlin/pull/4670#issuecomment-242618312

epatel, 2016-08-26:
@Roxy-3D I added the `z_offset` because MBL was manual in the beginning, and when doing the measuring some times one maybe was probing too close and other times too loose, i.e. how one was "feeling" the paper during the measuring. Sure one could use some else Z offset, but it felt best to add an offset for MBL close to what one was doing, so to speak. I think an overall offset still can be good to have, as probing will give a good sampling of the shape, but the getting a "real" good overall level might be easier to tune with a configurable offset.

Roxy-3D, 2016-08-26:
Thanks!   With the extra information, I'm leaning towards leaving both z_offset and zprobe_zoffset in place.

epatel, 2016-08-26:
@Roxy-3D I see it a little like this. The most important thing is that probing should be consistent while probing, meaning it is important that the points are sampled relative to each other. The the probe offset can be different from probing the whole bed, i.e. the probe could be manually attached etc. In my own auto version of MBL (minimalistic probe setup) http://www.thingiverse.com/thing:1120142 I added a menu option to manually probe this offset after the bed had been probed with the auto probe sequence (and I had detached the probe).

Roxy-3D, 2016-08-26:
Yes.   Prior to me asking the question, I was confused because it seemed like we had duplicate functionality.  But I agree.  There are separate issue here.    And at the root of the confusion was what BrainScan said:   I was forgetting that Mesh Bed Leveling was setup to work without a probe (even though I was being careful to preserve that capability).

universam1, 2016-08-27:
Thanks @thinkyhead very much for the detailed description and the backgrounds!

I went ahead and set up my settings according your proposal, you can have a look at them altogether at https://github.com/universam1/Marlin/tree/RCBugFix
Anyway, it is definitely broken! M851 has actually no impact on the offset anymore!

After a lot of tryings I though realize that the eeprom settings for "Z-home offset" are used **instead** of the "Z-probe offset". With this I can at least get my printer up and running again, although this is not a solution.

I understand that there is consistancy of code a goal, but I assume this logic at least hast to be mentioned somewhere or even moved to configuration_adv.

Thanks for fixing!

thinkyhead, 2016-08-27:
EEPROM settings are always loaded at boot. M851 is one setting that is saved and restored. If you don't issue M500 after M851 then it isn't saved to EEPROM.

universam1, 2016-08-28:
Absolutely, I made sure that it is saved and reloaded. For sake of test I even restarted Marlin. 

The M851 I've pointed out to make clear that I'm talking about the z probe offset that I can reach also via the settings in display conveniently. This one is broken now. 
The one that is now used is the z home offset which is not reachable via display. 
Does this make sense?

thinkyhead, 2016-08-28:
Yes, one can set positive values for M851. This applies usually when the nozzle is the probe and needs to press down a little to engage a switch.

thinkyhead, 2016-08-28:
@universam Did I already request a log? Enable `DEBUG_LEVELING_FEATURE` then do `M111 S255` before `G28` and `G29`. Post the log output here so we can see what's happening.

universam1, 2016-08-28:
@thinkyhead thanks for heads up on this issue! BTW. I've tried with negative values, no change.

OK here is the output with debug enabled. For info, M851 is set to -0.9:

`Recv:  position_shift = 0.00
Recv:  soft_endstop_min = -27.00
Recv:  soft_endstop_max = 200.00
Recv: > home_offset[X] = 0.00
Recv: current_position=(-27.00, 0.00, 187.00) :
Recv: <<< set_axis_is_at_home(X)
Recv: current_position=(-27.00, 0.00, 187.00) : sync_plan_position
Recv: current_position=(-27.00, 0.00, 187.00) : > AFTER set_axis_is_at_home
Recv: <<< homeaxis(X)
Recv: current_position=(-27.00, 0.00, 187.00) : > homeX
Recv: >>> homeaxis(Y)
Recv: current_position=(-27.00, 0.00, 187.00) : sync_plan_position
Recv: > 1st Home330.00
Recv: current_position=(-27.00, 0.00, 187.00) : sync_plan_position
Recv: current_position=(-27.00, 0.00, 187.00) : sync_plan_position
Recv: > 2nd Home10.00
Recv: >>> set_axis_is_at_home(Y)
Recv: For Y axis:
Recv:  home_offset = 0.00
Recv:  position_shift = 0.00
Recv:  soft_endstop_min = 0.00
Recv:  soft_endstop_max = 220.00
Recv: > home_offset[Y] = 0.00
Recv: current_position=(-27.00, 220.00, 187.00) :
Recv: <<< set_axis_is_at_home(Y)
Recv: current_position=(-27.00, 220.00, 187.00) : sync_plan_position
Recv: current_position=(-27.00, 220.00, 187.00) : > AFTER set_axis_is_at_home
Recv: <<< homeaxis(Y)
Recv: current_position=(-27.00, 220.00, 187.00) : > homeY
Recv: current_position=(-27.00, 220.00, 187.00) : sync_plan_position
Recv: current_position=(-27.00, 220.00, 187.00) : clean_up_after_endstop_or_probe_move
Recv: <<< gcode_G28
Recv: X:-27.00 Y:220.00 Z:187.00 E:0.00 Count X: -2160 Y:17600 Z:74800
Recv: ok
[...]
Recv: echo:M105
[...]
Send: G29
Recv: echo:G29
Recv: >>> gcode_G29
Recv: current_position=(-27.00, 220.00, 187.00) :
Recv: Probe: SERVO PROBE
Recv: Probe Offset X:24 Y:9 Z:-0.90(Right-Back & Below Nozzle)
Recv: G29 Auto Bed Leveling
Recv: corrected_position=(-27.00, 220.00, 187.00) : BEFORE matrix.set_to_identity
Recv: current_position=(-27.00, 220.00, 187.00) : BEFORE matrix.set_to_identity
Recv: uncorrected_position=(-27.00, 220.00, 187.00) : AFTER matrix.set_to_identity
Recv: current_position=(-27.00, 220.00, 187.00) : sync_plan_position
Recv: current_position=(-27.00, 220.00, 187.00) : setup_for_endstop_or_probe_move
Recv: current_position=(-27.00, 220.00, 187.00) : set_probe_deployed
Recv: deploy: 1
Recv: do_probe_raise(20.00)
Recv: >>> do_blocking_move_to(-27.00, 220.00, 187.00)
Recv: >>> probe_pt(15.00, 20.00, no stow)
Recv: current_position=(-27.00, 220.00, 187.00) :
Recv: do_probe_raise(15.00)
Recv: > do_blocking_move_to_xy(-9.00, 11.00)
Recv: >>> do_blocking_move_to(-9.00, 11.00, 187.00)
Recv: echo:busy: processing
Recv: > current_position=(-9.00, 11.00, 187.00) : set_probe_deployed
Recv: deploy: 1
Recv: current_position=(-9.00, 11.00, 187.00) : >>> run_z_probe
Recv: current_position=(-9.00, 11.00, 187.00) : >>> do_probe_move
Recv: >>> do_blocking_move_to(-9.00, 11.00, -197.00)
Recv: echo:busy: processing
Recv: echo:busy: processing
[... 204 lines cut ...]
Recv: deploy: 0
Recv: do_probe_raise(20.00)
Recv: >>> do_blocking_move_to(161.00, 165.00, 20.00)
Recv: >>> do_blocking_move_to(161.00, 165.00, 20.00)
Recv: current_position=(161.00, 165.00, 20.00) : clean_up_after_endstop_or_probe_move
Recv: current_position=(161.00, 165.00, 20.00) : > probing complete
Recv: Eqn coefficients: a: 0.00063235 b: 0.00113636 d: 6.02681636
Recv: uncorrected_position=(161.00, 165.00, 20.00) : >>> set_bed_level_equation_lsq
Recv: current_position=(161.00, 165.00, 20.00) : >>> set_bed_level_equation_lsq
Recv: corrected_position=(161.01, 165.02, 19.71) : <<< set_bed_level_equation_lsq
Recv: current_position=(161.01, 165.02, 19.71) : sync_plan_position
Recv: 
Recv: 
Recv: Bed Level Correction Matrix:
Recv: +1.000000 +0.000000 +0.000632
Recv: -0.000001 +0.999999 +0.001136
Recv: -0.000632 -0.001136 +0.999999
Recv: > BEFORE apply_rotation_xyz > stepper_z = 20.00 ... z_tmp  = 19.71
Recv: > AFTER apply_rotation_xyz > z_tmp  = 20.03
Recv: current_position=(161.01, 165.02, 19.74) : sync_plan_position
Recv: current_position=(161.01, 165.02, 19.74) : > corrected Z in G29
Recv: <<< gcode_G29
Recv: X:161.01 Y:165.02 Z:19.74 E:0.00 Count X: 12880 Y:13200 Z:8010
Recv: ok
`

thinkyhead, 2016-08-29:
Your log doesn't show the homing of the Z axis.

thinkyhead, 2016-08-29:
I note that the probe is triggered when your nozzle is between 5.99mm and 6.32mm above the bed. So your probe offset should be set somewhere in-between, maybe -6.16 or so.

universam1, 2016-08-29:
@thinkyhead sure, octoprint's buffer is kind of short... Alright see below.

Z_PROBE_OFFSET_FROM_EXTRUDER cant be set to -6.16 because the define expects integer - so set to -6

I've tried with z-probe set to -4 and 0 and it has no impact on the eventually height of Z.
So wouldn't you agree finally that this is a bug, not a support problem? :)

`Connecting to: /dev/ttyUSB0
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0xb3bf6d10, open=True>(port='/dev/ttyUSB0', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125
Recv: start
Send: N0 M110 N0*125
Recv: echo:Marlin 1.1.0-RCBugFix
Recv: echo: Last Updated: 2016-07-26 12:00 | Author: (Sam, Prusam v1.0)
Recv: Compiled: Aug 29 2016
Recv: echo: Free Memory: 2976  PlannerBufferBytes: 1232
Recv: echo:V24 stored settings retrieved (398 bytes)
Recv: echo:Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E465.00
Recv: echo:Maximum feedrates (mm/s):
Recv: echo:  M203 X110.00 Y110.00 Z15.00 E50.00
Recv: echo:Maximum Acceleration (mm/s2):
Recv: echo:  M201 X2000 Y2000 Z1000 E10000
Recv: echo:Accelerations: P=printing, R=retract and T=travel
Recv: echo:  M204 P1500.00 R5000.00 T1500.00
Recv: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
Recv: echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
Recv: echo:Home offset (mm)
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Material heatup parameters:
Recv: echo:  M145 S0 H180 B75 F0
Recv: echo:  M145 S1 H240 B110 F0
Recv: echo:PID settings:
Recv: echo:  M301 P16.10 I0.89 D72.40
Recv: echo:  M304 P274.43 I14.03 D1342.12
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D3.00
Recv: echo:  M200 D0
Recv: echo:Z-Probe Offset (mm):
Recv: echo:  M851 Z-4.00
Recv: echo:SD card ok
Recv: ok
Changing monitoring state from 'Connecting' to 'Operational'
Send: N0 M110 N0*125
Recv: echo:SD car�ok
Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
[...]
Send: M20
Recv: Begin file list
Recv: End file list
Recv: ok
Send: M80
Recv: ok
[...]
Send: M111 S255
Recv: echo:DEBUG:ECHO,INFO,ERRORS,DRYRUN,COMMUNICATION,LEVELING
Recv: ok
[...]
Recv: echo:M105
[...]
Send: G28
Recv: echo:G28
Recv: >>> gcode_G28
Recv: current_position=(0.00, 0.00, 0.00) : setup_for_endstop_or_probe_move
Recv: > endstops.enable(true)
Recv: >>> homeaxis(Z)
Recv: current_position=(0.00, 0.00, 0.00) : sync_plan_position
Recv: > 1st Home280.50
[... 307 lines cut ...]
Recv: deploy: 0
Recv: do_probe_raise(16.00)
Recv: >>> do_blocking_move_to(161.00, 165.00, 16.00)
Recv: echo:busy: processing
Recv: >>> do_blocking_move_to(161.00, 165.00, 16.00)
Recv: current_position=(161.00, 165.00, 16.00) : clean_up_after_endstop_or_probe_move
Recv: current_position=(161.00, 165.00, 16.00) : > probing complete
Recv: Eqn coefficients: a: 0.00120588 b: 0.00114177 d: 5.94421625
Recv: uncorrected_position=(161.00, 165.00, 16.00) : >>> set_bed_level_equation_lsq
Recv: current_position=(161.00, 165.00, 16.00) : >>> set_bed_level_equation_lsq
Recv: corrected_position=(161.02, 165.02, 15.62) : <<< set_bed_level_equation_lsq
Recv: current_position=(161.02, 165.02, 15.62) : sync_plan_position
Recv: 
Recv: 
Recv: Bed Level Correction Matrix:
Recv: +0.999999 +0.000000 +0.001206
Recv: -0.000001 +0.999999 +0.001142
Recv: -0.001206 -0.001142 +0.999999
Recv: > BEFORE apply_rotation_xyz > stepper_z = 16.00 ... z_tmp  = 15.62
Recv: > AFTER apply_rotation_xyz > z_tmp  = 16.04
Recv: current_position=(161.02, 165.02, 15.66) : sync_plan_position
Recv: current_position=(161.02, 165.02, 15.66) : > corrected Z in G29
Recv: <<< gcode_G29
Recv: X:161.02 Y:165.02 Z:15.66 E:0.00 Count X: 12880 Y:13200 Z:6416
Recv: ok`

universam1, 2016-08-29:
@thinkyhead can you please have a look at https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Marlin_main.cpp#L3660
or
https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Marlin_main.cpp#L3721

For Deltas it's all right, but for other printer it simply looks like there is __newer__  zoffset or zprobe_zoffset addition done **anywhere** ...

Please correct me if I'm wrong.

universam1, 2016-08-29:
@thinkyhead Looks like I found the root cause of this issue! :smile: 

In
https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Marlin_main.cpp#L1588

this `&& Z_HOME_DIR < 0` is negative, thus the `current_position[Z_AXIS] -= zprobe_zoffset;` is never executed!
I have no idea why this condition was defined, but in my case it is wrong.

Do you like a MR?

thinkyhead, 2016-08-30:
Why do you have `Z_HOME_DIR` set to 1 if you are homing to the bed and not to a max Z endstop?

closed 2016-08-30 by thinkyhead

universam1, 2016-08-30:
@thinkyhead No, I'm _not_ homing to bed, I'm homing to max Z endstop!
So my `Z_HOME_DIR`is set to 1.

Where is my fault?

thinkyhead, 2016-08-30:
How come, twice you have posted logs of G28 in which the Z homing never occurs?

universam1, 2016-08-30:
Honestly, I have only this single printer and did not change anything here...

I dont understand your question, lets use this slice from output, thats homing to Z max, doesnt it?

`Recv: >>> homeaxis(Z)
Recv: current_position=(0.00, 0.00, 0.00) : sync_plan_position
Recv: > 1st Home280.50
Recv: current_position=(0.00, 0.00, 0.00) : sync_plan_position
Recv: current_position=(0.00, 0.00, 0.00) : sync_plan_position
Recv: > 2nd Home4.00
Recv: >>> set_axis_is_at_home(Z)
Recv: For Z axis:
Recv:  home_offset = 0.00
Recv:  position_shift = 0.00
Recv:  soft_endstop_min = 0.00
Recv:  soft_endstop_max = 187.00
Recv: > home_offset[Z] = 0.00
Recv: current_position=(0.00, 0.00, 187.00) :
Recv: <<< set_axis_is_at_home(Z)
Recv: current_position=(0.00, 0.00, 187.00) : sync_plan_position
Recv: current_position=(0.00, 0.00, 187.00) : > AFTER set_axis_is_at_home
Recv: <<< homeaxis(Z)`

thinkyhead, 2016-08-30:
My error there. I was looking for it to occur after XY.

universam1, 2016-08-30:
NP, thanks for help!

I made a PR, if you like.

thinkyhead, 2016-08-30:
It's not over yet!

thinkyhead, 2016-08-30:
Now then, `set_axis_is_at_home` should set Z to the `Z_HOME_POS` after homing to max. Why should it add the probe offset?

universam1, 2016-08-30:
Good question, I was asking myself the same.
But I realized that Z homing_offset is added also at that point, which works (surprisingly)!
So since I'm not the inventor of that logic I understand that this offsets the current position. Later in G29 you can see that this is taken into account ....

thinkyhead, 2016-08-30:
Let me ask, if you simply do a G28 without this change in the code, and then lower the nozzle down to zero, does the nozzle touch the bed?

thinkyhead, 2016-08-30:
> Later in G29 you can see that this is taken into account

Then the error is in G29, which is not supposed to molest the Z position, only correct it based on the bed tilt.

universam1, 2016-08-30:
sorry cant tell now, not at home. Does it play a role?

thinkyhead, 2016-08-30:
If the `Z_HOME_POS` is set correctly, then the nozzle will be right up to the bed. If it's too high, then you would need to reduce the value of `Z_MAX_POS` (or `Z_MANUAL_HOME_POS`) to fix that.

universam1, 2016-08-30:
So why then there is the M851 if not to adjust the diff from probe to nozzle?

thinkyhead, 2016-08-30:
It's there for the case where you home Z using the probe.

universam1, 2016-08-30:
So the reprap docu says:

> Example: M851 Z-4
The example above will set the z-probe offset EEPROM setting to -4mm below the nozzle and enables the nozzle travel 4mm lower than the probe triggered position.

But really, why should you want to limit this nice and great functionality to someone who only does homing via probe?
I dont get it really.

Lets explain: I have several different bed plates, like carbon, glas, glas with tape, glas with capton. I change these plates and I loved Marlin with G29 since it was always so clever to _detect_ via probe the tilt but especially the real height. What a blessing! It worked long time nicely, without issues. So I wonder, why should this be removed now?

thinkyhead, 2016-08-30:
In fact, homing to the bed with the Z probe is the _**only**_ place that the `M851` (`zprobe_zoffset` / `Z_PROBE_OFFSET_FROM_EXTRUDER`) is used any more.

thinkyhead, 2016-08-30:
> The example above will set the z-probe offset EEPROM setting to -4mm below the nozzle and enables the nozzle travel 4mm lower than the probe triggered position.

Whoever wrote that was tripping. It's wrong.

universam1, 2016-08-30:
Please dont get me wrong, but why are you changing this? Marlin was so great ...

Okay I dont have a problem to use home_offset from now on, but then this should be exposed to display and not the M851!!

thinkyhead, 2016-08-30:
It's not a change. This is how it's always been.

universam1, 2016-08-30:
Come on, I was printing months with this, and it worked till RC6.

thinkyhead, 2016-08-30:
Do as I asked. Home to max. Lower to zero. If the nozzle isn't touching the bed, simply fix your `Z_MAX_POS`.

thinkyhead, 2016-08-30:
What is it you specifically want to do, other than have the correct Z position upon `G28`?

universam1, 2016-08-30:
I specifically want to fine tune the bed touch via eeprom since its complicated to flash my printer (octoprint connected in garage)
At least via eeprom, but best via Display setting (which is the M851)

I fully respect your great work and decisions! Would it be okay to make a feature request for that then?

thinkyhead, 2016-08-30:
> fine tune the bed touch via eeprom

I don't know what this means.

universam1, 2016-08-30:
I go to display settings --> "Z probe offset" and add or remove a little, press safe EPROM and load EPROM - if I'm not 100% happy with distance of the nozzle to the bed. This might be for instance when I remove the nozzle for maintenance etc.

thinkyhead, 2016-08-30:
The probe offset has never been used in `G29` to correct the z distance to the bed. It has only ever been used to adjust the current Z based on the bed tilt. It may be that I'm missing the fact entirely that it was doing so, but truly that section of the code was never documented, and if it was fine-tuning the Z position then that fact was lost on me.

thinkyhead, 2016-08-30:
> I go to display settings --> "Z probe offset" and add or remove a little, press safe EPROM and load EPROM - if I'm not 100% happy with distance of the nozzle to the bed.

That feature already exists, but it has only (again, as far as I know) applied to homing with the probe, and not to a fine-tuning applied in `G29`.

universam1, 2016-08-30:
I was coming to it by reading the wiki:

https://github.com/MarlinFirmware/Marlin/wiki/DNE-Bed-Auto-Leveling

> n) Fill in the values below. If the endstop mark is in front of the line running left-to-right, use positive values. If it is behind, use negative values. For the Z axis use the value from (k) and subtract 0.1mm. (Note: you can adjust z probe offset via M851)
Use M851 to adjust the length Z probe (M851 Z-6.80 if it was -7). Store in EEPROM with M500 and repeat the sequence G28, G29, M114 again. Is the gap bigger then you adjusted the Z probe offset in the opposite direction. So redo M851 Z-something, M500 and repeat G28, G29, M114 and measure (read Z from M114) z at 0.

thinkyhead, 2016-08-30:
See, I was going to go to bed early tonight, and now I have to dig through the old code and probably end up slapping my forehead a lot. Because, truly, if `G29` was re-writing Z based on the probe distance, I have been missing it for … years now.

thinkyhead, 2016-08-30:
ewwww… that wiki is so out of date with 1.1.0. You can't even use G92 that way any more.

universam1, 2016-08-30:
Sorry I really dont want to blame. Thanks for heads up on this!
I can tell, that with this little change everything works like before and ( I ) expected to be.

Thanks

thinkyhead, 2016-08-30:
Well, please understand, in rewriting and trying to unify the bed leveling I made it a maxim not to alter the Z position after G28, and to trust in the home positions set in configuration. Lately I've been feeling like a lot more things should be adjustable. But right now, my priority is in fixing bugs and getting this release out the door.

I don't mind re-examining `G29`, as I'm in the middle of this, as I am looking for ways to improve the Z position result once `G29` is done. It seems _almost obvious_ that every individual probe in `G29` affords an opportunity to determine a more accurate Z position (assuming the probe z offset is good). But then, realize that no point on the bed is definitive. Or, they all are. Which height is the "true" height, and which is just lower or higher than some reference point? Was the home position set correctly? What about the min endstop situation? Is that the "real" zero? And at the end, you have to decide whether the Z position is off, or whether the bed is just high or low at the final XY. It's not cut-and-dried how best to use the probe results to determine Z.

However, we do know that right now in `RCBugFix` the Z correction is wrong, and there are various reasons for that, including the addition of `Z_SAFE_HOMING` and other factors. That is why I put out this branch and have been asking for testing…

https://github.com/thinkyhead/Marlin/tree/rc_fix_leveling_maths

universam1, 2016-08-30:
Very interesting!

Let's take this case with the nozzle offset. Here I understand that the offset will apply to every probe point in the same manner and is not to be tilted at all. Therefore this offset can be Z - additive to every point directly.
Now if we take into account the true height, we should subtract this value again from max Z. Then the nozzle would basically move independently in height, but the distance from Z max to probe would be fix ...

thinkyhead, 2016-08-30:
First, take note that the probe offset doesn't matter in itself to measuring the tilt of the bed. The points are "thrown out" once the tilt angle of the bed is determined.

However, the probe offset is useful to know, at any point, what the distance to the nozzle is. And therefore you can consider any point that you've probed, or any point that you can interpolate from the matrix, as being a valid reference point to determine the height of the bed.

Let's take a concrete example. With a probe offset of -10 you probe the bed and get the following 4 points:
```
1. X:  0 Y:  0 Z:9.95
2. X:  0 Y:200 Z:9.90
3. X:200 Y:200 Z:9.85
4. X:200 Y:  0 Z:9.90
```
Of course we also have a home position from G28 to an endstop that simply applies over the entire bed, making the assumption that the bed is flat in relation to the XY axes. In terms of the distance from the probe to the nozzle, it is idealized, and therefore you can take it as 10.00 across the entire bed.

Now, we could take the home Z at any point, but generally it applies to wherever you calibrated your endstop. For a min endstop, usually you use a screw to adjust until G28 brings the nozzle down nice and close to the bed. And you usually check a few points across the bed to ensure it's as close to level as you can get it. But let's assume, for the moment, that you haven't done that in a while, and so it's fine at the home position — probably 0, 0 — or maybe 200, 0 for an X-max-endstop. So you'd like to have the probe process find out where the bed really is.

So we have these points above. They show that the bed is tilted diagonally. At the front-left the bed is lower, and at the back-right the bed is higher. So the bed leveling matrix will produce a tilted plane reflecting that.

Now, let's say you're using a docking sled probe, or maybe a self-stowing allen key. So at the end of the leveling procedure, but before applying the bed level matrix, you have the carriage move off to the corner and drop off or push the probe to put it away. And you leave Z there.

Now, you want to turn on the matrix. But as soon as you do, the Z position will no longer be correct. If it's near the high corner, it will have too high a value. It's actually closer to the bed at that point. So what we do is take the Z position and apply the inverse matrix to it. In the process we also adjust XY a tiny bit, because that's how planar leveling works. However, we haven't yet figured out our reference point, so we can't actually apply the Z adjustment based on the tilted plane.

Remember, we need a valid reference point to make the determination of how much the bed is raised or lowered at the current Z, and we haven't yet decided where the fulcrum of the plane lies.

So how do we proceed to get the "most correct" Z for the nozzle, where it is now? Well, let's take another look at those probed points.

Obviously, all across the bed the Z position of the nozzle is not 10.0. Everywhere we probed, the nozzle was a little lower than 10.0. We have to ask ourselves: Was the Z=0 that G28 gave us wrong? Or is the probe offset wrong? How can we determine which is true, and does it matter?

Then: What about the case where leveling is turned off? Shouldn't we make sure we have the best Z with `G28` alone first? It's less vital if you plan to always do G29, but if in the future it becomes possible to save the result of G29, then we're going to need a reliable G28 to give us a good Z to work from.

Going back to correcting Z…

If we assume that the probe offset is 100% correct, then clearly our Z=0 isn't good. Taking the 10.0 offset as gospel, the probe hit the bed to early. Whatever Z=0 we have from homing, it's clearly too low. So we simply have to adjust the current Z upward by some amount. How much?

Since every probed point was under 10.0 we could try adjusting by the average difference. That corrective makes some of the probed Zs over 10.0 and some under 10.0. So let's start there.
```
10 - (9.95+9.90+9.85+9.95) ÷ 4 = 0.0875
```
Adding that to our current Z —let's say it's "15.0"— makes it now 15.0875. So now our probed bed looks like this:
9.95 + 0.0875 = 10.0375
9.90 + 0.0875 =  9.9875
9.85 + 0.0875 =  9.9375
9.95 + 0.0875 = 10.0375
```
This correction to Z applies perfectly if the nozzle were in the _middle_ of the bed. Obviously, right? The mean of all Zs probed on a flat but tilted plane must be in the dead center.

So now we have our fulcrum. Now we just need to get the distance from the center of the nozzle's current XY, apply the bed tilt compensation for that XY distance, and presumably we will have the correct Z.

See how simple it all could be?

Blue-Marlin, 2016-08-30:
Maybe my calc_sheet can help to see what is going on.
Simplified to 2D.
When z_probe_offset and home_position is compensated out, we seem to need an other way to tune the nozzle hight.
Still calculating back and forth for different homing configurations.

[Matrix_base_2D.zip](https://github.com/MarlinFirmware/Marlin/files/444789/Matrix_base_2D.zip)

thinkyhead, 2016-08-30:
Your spreadsheet is comprehensive, but for the case of a perfectly flat plane I think we can optimize and simplify according to what I wrote above. It was as I was writing the above that I hit on the realization that we can in fact always use the center of the bed - or probe region - as the fulcrum for converting the XYZ position into the tilted space by taking the average probe height — again, presuming a perfectly flat bed and a very accurate probe. But I realize that since outliers will throw it off we have to use the **median** or **mode** instead of the **mean** average.