MarlinFirmware/Marlin #4711: confused by autoleveling setting

argus0209, 2016-08-27:
My autoleveling to trying to probe off my heat bed.

// Travel limits after homing
#define X_MAX_POS_DEFAULT 225
#define X_MIN_POS_DEFAULT 33
#define Y_MAX_POS_DEFAULT 225
#define Y_MIN_POS_DEFAULT 33
#define Z_MAX_POS_DEFAULT 225
#define Z_MIN_POS_DEFAULT 0

```
#define ENABLE_AUTO_BED_LEVELING // Delete the comment to enable (remove // at the start of the line)

#ifdef ENABLE_AUTO_BED_LEVELING

  // these are the positions on the bed to do the probing. 
  // NOTE: if the absolute value of the probe offset (x,y directions) is > 10, we have to move probe points inboard on that axis.
  #define LEFT_PROBE_BED_POSITION (bed_level_probe_offset[0] <= 10 ? 10 : bed_level_probe_offset[0])
  #define RIGHT_PROBE_BED_POSITION (bed_level_probe_offset[0] >= -10 ? X_MAX_LENGTH - 10 : X_MAX_LENGTH + bed_level_probe_offset[0])
  #define BACK_PROBE_BED_POSITION (bed_level_probe_offset[1] >= -10 ? Y_MAX_LENGTH - 10 : Y_MAX_LENGTH + bed_level_probe_offset[1]) 
  #define FRONT_PROBE_BED_POSITION (bed_level_probe_offset[1] <= 10 ? 10 : `bed_level_probe_offset[1])

how do I set the above setting? is there a guide somewhere for this? Thanks for any information.

-Steve

thinkyhead, 2016-08-27:
Are you sure you're using Marlin? There are no `X_MAX_POS_DEFAULT` etc. settings. Marlin's limits look like this:
```cpp
// Travel limits after homing (units are in mm)
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 950
#define Y_MAX_POS 950
#define Z_MAX_POS 900
```

zenmetsu, 2016-08-27:
He appears to be running the Printrbot fork of Marlin.  The correct place to report the bug would be [https://github.com/Printrbot/Marlin](url)

argus0209, 2016-08-27:
I am using marlin on a printrboard rev f but doesn't the Marlin RC7 look the same?  Thinkyhead your setting look soooooo simple, maybe I should switch to what version your using? What version are you using?

Zenmetsu your link is dead. :(

Thanks for the superfast reply guys.

~STeve

zenmetsu, 2016-08-27:
I fixed the link.  My apology.  The printrbot fork is similar, but as @thinkyhead said, the configs are different.

thinkyhead, 2016-08-27:
Printrboard is being different just to be different, and their fork is intentionally diverging from the main Marlin. As I understand it they have no intention of bringing their fork into alignment with the main project. However, we will continue to support their hardware going forward because we're magnanimous like that.

argus0209, 2016-08-28:
Well thanks zenmetsu and thinkyhead for all the useful information.

closed 2016-08-28 by argus0209