MarlinFirmware/Marlin #4712: What does  "F" represent specifically in gcode?

lizlovesspurs, 2016-08-27:
As  is  known  to  all ,  "F"  means  feedrate  or  speed in gcode.  In  3d  printing plane motion ,  there  are  three   kinds  of   velocity,  including   x-stepper  velocity,  y-stepper  velocity  and   x-y  resultant  velocity.
So   which  does  "F" represent  in  the  three   kinds  of   velocity ? Does  it  represent  the  magnitude of the x-y  resultant  velocity  in  Marlin  source  code?  I  also want  to  know  the  mathematical relationship  between  F   value   with  the  e-stepper  speed  in  Marlin  source  code.

Blue-Marlin, 2016-08-27:
"F" is the __maximum__ speed for a move in the 4 dimensional room (XYZE).
This speed is cut down to the individual axis max limits (DEFAULT_MAX_FEEDRATE).
If "F" can be reached also depends on, if the move is long enough - the acceleration is high enough, and how big the speed difference to the previous/following move is.

For details see planner.cpp Planner::buffer_line(...)

thinkyhead, 2016-08-27:
Note that it represents feedrate in **millimeters per minute**.

Roxy-3D, 2016-08-27:
> If your extruder has a gear with a diameter of 20mm then the extruder will push 20mm of filament per full revolution. 

I think you meant to say:

If your extruder has a gear with a circumference of 20mm then the extruder will push 20mm of filament per full revolution.

thinkyhead, 2016-08-27:
Yes, sorry, senescence is setting in.

Roxy-3D, 2016-08-28:
I had to look up that word.  I didn't know what it meant.   :)

lizlovesspurs, 2016-08-28:
@Blue-Marlin   Thank   you  for   your  answer!  As  you  mentioned  before,  the  4  dimensional  Bresenham algorithm  is  the  core  of  Marlin. However,  I  only  found  some  introduction  about  2  dimensional  Bresenham  algorithm in  the  Internet .  Could   you  please  give  me   a  chained address  of  introduction  about  4  dimensional  Bresenham algorithm? I  think  it  will  help  me   understand how  Marlin  controls   stepper  movement   better.

thinkyhead, 2016-08-28:
For the Bresenham algorithm 4 dimensions (or axes) is exactly the same as 2. You just use 4 counters, each one adding on its own length on each iteration. And all of them use the same value –the negative largest distance– as their reset.

lizlovesspurs, 2016-08-28:
@thinkyhead   As  you  said,  4  demension  Bresenham  algorithm   seems   like  multiaxis  DDA   interpolation  algorithm.  Can  I   understand  4  demension  Bresenham  algorithm   in  this  way?

thinkyhead, 2016-08-28:
It is best understood in its own terms. Using the class I wrote above, you can experiment with it all day.

Blue-Marlin, 2016-08-28:
Bresenham and DDA are relative similar algorithms (originally) for 2 dimensional line drawing on raster displays. The advantage of Bresanham over DDA is - it's fast - it only needs integer calculations.

closed 2017-10-31 by thinkyhead

amrithmmh, 2019-02-25:
thanks @thinkyhead