MarlinFirmware/Marlin #4713: g28 home location

locutus40, 2016-08-27:
on rc6 home used to be if your lookin at your printer to the right and back of the bed.. i have a pruisa i3.. clear frame

now with rc7 the bed slides forward and the nozel almost to the left of the bed.. 

is there a way to tell it or change it back to right and back??

what code and section do i look in thanks...

brainscan, 2016-08-27:
Most printers are set to home to the min endstops which is normally bottom left of print bed. Sounds like in rc6 you were homing to max which would be top right of print bed. Have you compared your configuration files to check you have set things the same way?

locutus40, 2016-08-27:
i looked over everything its all the same... i have end stops all on the right side and y bed is at the front where i sit... its a folgertech prusia i3 that i use im using the stock locations for everything
 i mean its fine homeing where it does now it just seems every time a rc comes out it changes.. and i cant figure out why

brainscan, 2016-08-27:
The homing location is always set by the user it doesn't change from rc to rc, well at least I've never noticed it change. Can you maybe post both config files as .txt so we can take a look?

thinkyhead, 2016-08-27:
Assuming your front-left corner is 0,0 and your back-right corner is 200,200 the configuration to home to the back and right endstops will simply be:
```cpp
#define X_HOME_DIR 1
#define Y_HOME_DIR 1
#define Z_HOME_DIR -1
. . .
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 200
#define Y_MAX_POS 200
#define Z_MAX_POS 200 // or whatever
```

locutus40, 2016-08-28:
yea mine says everything that does.. cept

heres what mine says

i noticed yours are 1 and 1 not -1 -1
see the rc6 said the same thing -1-1-1 soo why woundt rc7 be the same location makes no sense..

also theres this which is new and i didnt see till now...// The center of the bed is at (X=0, Y=0)
//#define BED_CENTER_AT_0_0

// Manually set the home position. Leave these undefined for automatic settings.
// For DELTA this is the top-center of the Cartesian print volume.
//#define MANUAL_X_HOME_POS 0
//#define MANUAL_Y_HOME_POS 0
//#define MANUAL_Z_HOME_POS 0 // Distance between the nozzle to printbed after homing....

now looking at the define bed center0_0 do i unmark that? or will that say hey thats home?  see thats not on the other rc6

// ENDSTOP SETTINGS:
// Sets direction of endstops when homing; 1=MAX, -1=MIN
// :[-1,1]
#define X_HOME_DIR -1
#define Y_HOME_DIR -1
#define Z_HOME_DIR -1

#define min_software_endstops true // If true, axis won't move to coordinates less than HOME_POS.
#define max_software_endstops true  // If true, axis won't move to coordinates greater than the defined lengths below.

// @section machine

// Travel limits after homing (units are in mm)
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 170
#define Y_MAX_POS 155
#define Z_MAX_POS 170

locutus40, 2016-08-28:
oh and i do have a 200x200 bed i put in those low numbers because i use a inductinve probe outside the nozel about -0.38mm to the left of it and -.017mm in front of it.. so it would be outside the bed.. plus the aluminum bed has a cutout for the heater bed where you solder so its like a hair smaller than 200 mm... my end stops are at 0,0 which is if your looking at the thing your right arm and back.. and y sled is where the bed sticks out towards you and your right arm.. so just like the folgertech prusia i3 instructions state... i have no end stops on the left for going out of bounds although i might do that in the future but not now i use software to tell it to stop and its worked fine.. but i notice every update i do of marlin it puts it in the center of the bed then to the back and it does this back and forth and i change nothing i mean litterally nothing in the code i just use the stock code and input my settings for my machine i leave home direection alone above.. so it has me scratching my head...

so imagine rc7 bare bones stock thats what im using.. nothign major changed only add my name my heater..  the  bed size lcd type..

locutus40, 2016-08-28:
http://www.mediafire.com/view/e5l1ckv4ac2xaav/20160827_181310.jpg

heres a picture of my printer... thats where home is now....  home used to be to the right and back.. notice my end stops thats where it suppose to be ignore the end stop on the z axis i use an inductive probe for my home now hooked to the same input as the old end stop.. and i know thats not my problem ive had it for a while now working perfectly with no mods or telling marlin to use it as a end stop  it just thinks it is and works..

Roxy-3D, 2016-08-28:
> because i use a inductive probe outside the nozzle about -0.38mm to the left of it and -.017mm in front of it.. 

From that picture, I'm wondering if you meant to say the nozzle is offset -38mm on the X Axis and -17mm on the Y Axis?   Probably you have the X_OFFSET_FROM_EXTRUDER & Y_OFFSET_FROM_EXTRUDER numbers correct because if you put a -.38 in for X's number, the compiler would complain.

locutus40, 2016-08-28:
yes it sits in front and to the left of the nozzel  this is what i meant... i meassured using a digital calipar.. 
but its been fine since i had it and works rc6 home position was right and back the normal position..
rc5 was back to the middle.. rc7 back to the middle sort of.. see my delema makes no sense how can it change back and forth like that... and how in software can i tell it where i want it to go??? there has to be a line of code somewhere where i can say hey back and right of bed???
#define X_PROBE_OFFSET_FROM_EXTRUDER -37  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -17  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -0.28   // Z offset: -below +above  [the nozzle]

thinkyhead, 2016-08-28:
`#define X_HOME_DIR 1` means home to the right side
`#define Y_HOME_DIR 1` means home to the back (i.e., the endstop is mounted on the front)

Blue-Marlin, 2016-08-28:
All our descriptions about right/left, front/back rely on the lowest values are on the front, left corner and a right handed coordinate system. Marlin can not handle left handed coordinate systems. If your back, right corner has the lowest coordinate values invert all front/back, right/left statements.

locutus40, 2016-08-28:
ok so ill just change the -1
                                       -1
                                       -1

to                                      1
                                          1
                                          -1

and im grasping at straws because the code every time it changes jumps all around its never in the same spot always something is put in and not explained.. i saw that and was like what the... so im going to try the above and see if that works... if not ill just leave it alone

thinkyhead, 2016-08-28:
> front, left corner and a right handed coordinate system

True. If you prefer 0,0 to be in a different corner, just rotate all this stuff around. Basically, you want to set it up so when you home to your XY endstops, you get the coordinates out that you prefer.

> something is put in and not explained

We're working on documentation in our copious spare time.

locutus40, 2016-08-28:
ill try what i said in a bit and ill inform you if it works.. anyway i can say this for myself.. thanks for all the work you guys put into this stuff...

without you guys we woulndt have these printers so cheap and easy to use..

anyway yea i didnt notice the bed center thing in the code at all in previouse versions maybe it was placed    
in another section...... or im just  blind lol...

thinkyhead, 2016-08-28:
> or im just blind lol

Maybe you're like me. I always tend to find things in the last place I look.

locutus40, 2016-08-28:
yup didnt work.. almost destroyed my printer... moved the y bed full and keep going skipping 50 teeth and the x skipped 50 teeth i was lucky to hit emergency stop.. im gunna leave it -1 -1 -1 and where it homes now.. no need to mess around with it.. pheeewww lol

thinkyhead, 2016-08-28:
Grinding the steppers does no harm, it just sounds scary. Glad to hear you have it configured the way you prefer.

locutus40, 2016-08-28:
actually it does do harm i have a drag chain only set up for the lenth of my bed and when it went to far it ripped one of the links..  also when it makes those noises its moving over the belt ridges and it can cause them to tear off  the little rubber ridges causing missed steps.... and problems...
and me having to go buy a whole new belt..

so yes it does and might cause damage.. im lucky it was just a link..

thinkyhead, 2016-08-29:
When experimenting with configuration, risk is part of the game. Always stand close to the reset.

Roxy-3D, 2016-08-29:
> When experimenting with configuration, risk is part of the game. Always stand close to the reset.

And I admit it is paranoia.  But I don't even trust the 'Kill' button on my LCD Panel.  I've soldered a wire onto the Reset pin of my Mega-2560 board and have it connected to a big Red Reset switch.   The people writing the software understand how bad a 'Nozzle Crash' is !!!!

thinkyhead, 2016-08-29:
Remind me to tell you about the couple of times my i3 nearly went up in flames…

locutus40, 2016-08-29:
in flames? do tell... let me guess runaway heater elements..

this is why i never print more than a 2 hour job and i never leave it infact my pc sits next to it so i can stop it if i need too..

anyway if you guys could look into this home position thing i mean.. it just really weird every other revision that comes out it swaps back and forth with out me touching a thing.. litterally download add a couple things and poof.. next tone comes out ie rc7 and back to the middle lol i sware it has a mind of its own.. 

what i think i should do is add more end stops one on the left one at the back of the bed and z up and down of the head does not matter it cant go very fast upward takes atleast 2 minutes to reach the top for me since i have threaded rod and not that fancy rod.. which brings me to this i know its off topic.. but i did want to add that nice rod i forget what its called i have the files to print out all the stuff to install it i just need to find where in the code to put the steps per turn thingie.. and how to calculate it .. <<<total noob.. but learning im like a dog sit boy sit good boy thats all i know haha

thinkyhead, 2016-08-29:
> if you guys could look into this home position thing

@locutus40
- When you move X in the positive direction does it move to the right? For example, does the `G1 X200` command move the X carriage to the right?

- When you move Y in the positive direction does the bed come forward? In other words, does the `G1 Y200` command move the bed all the way to the front, so the nozzle is sitting at the rear edge?

closed 2017-10-31 by thinkyhead