MarlinFirmware/Marlin #4714: X axis movement on rc6 bugfix vs rc7 at Tevo Tarantula

sarf2k4, 2016-08-27:
I ported the settings from repetier firmware of the tevo tarantula settings to the rc7 config.h, and to my surprise is that the x motor giving me trouble upon moving by producing high pitch noise but not whining noise. It really moves on 0.1mm but not on higher. Lowering the max accel on it works but moving the x axis by 10mm on pronterface, the axis just moved roughly about 150mm across the rail. All other axis are good (not thoroughly checked)

I've found a link to rc6-bugfix for this tevo tarantula model that uses MKS BASE V1.2 mainboard, flashed and the x axis moved the way it supposed to. I copied the settings especially mechanical settings from that rc6 to the rc7 config.h, same problem as previous paragraph occurred. If someone got this exact printer, can you test them on?

Here is the link to the marlin tevo tarantula - https://drive.google.com/file/d/0B8vfd0KYy-XSVm0zVTZRV051OEU/view

And here is the following rc7 config.h for this tevo tarantula.
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/440657/Configuration.txt)

thinkyhead, 2016-08-28:
I'm looking over your `Configuration.h`. Did you make sure to use `M502` followed by `M500` to reset the settings in EEPROM?

sarf2k4, 2016-08-28:
It was from repetier from the first place and transitioning between marlin
versions will reset the eeprom prior to that config.h values upon flashing.
On the rc7 i tried reducing the accel, it just shoots like hell eg 10mm
movement ended up from 0mm to 150mm in just 1 second. What about the board
selection to flash marlin to mks base v1.2? Do we use 1.3 or 1.0, i am
confused since there are 2 types of mks board out there

thinkyhead, 2016-08-28:
Actually, it appears the two MKS BASE options are no different from each other. So just use `BOARD_MKS_BASE`.

sarf2k4, 2016-08-28:
Oh, okay. But i wonder if this applies only on mks base boards. I do use
the base rc7 on my printer with rampa 1.4 does not give me this x axis
movement problem but only the tevo. It is kind of puzzling. I also tried
reducing the x axis vref to 0.6v and it is still the same and tried to
assist the movement by pushing the carriage, sounded like skipping steps. =/

thinkyhead, 2016-08-28:
Have you had a look at this?

https://www.dropbox.com/s/lou3yv4mgpgipb0/tarantula-marlin.community-v.0.1d.zip?dl=0#

sarf2k4, 2016-08-29:
I have not, but i will check it out tomorrow

sarf2k4, 2016-08-30:
I checked and downloaded marlin from that dropbox link, it seems that the marlin in there is somewhere around rc3 or 1.0.1 I think from the looks of the config.h structure. I also planning to use the manual mesh bed leveling (not sure when it got introduced). I currently got to stick with this rc6 bugfix for now till there's a workaround to use stock marlin on this tevo

sarf2k4, 2016-09-07:
i need help in the marlin rc6-bugfix posted on google drive, the z-offset are not accessible in the control>motion tree, this printer of mine would be transported from time to time and i need to have the z-offset capability accessible from lcd, How do I enable this thing?

Roxy-3D, 2016-09-07:
> I also planning to use the manual mesh bed leveling.
     
> i need help in the marlin rc6-bugfix posted on google drive, the z-offset are not accessible in the control>motion tree, this printer of mine would be transported from time to time and i need to have the z-offset capability accessible from lcd, How do I enable this thing?

It sounds like you ought to be using the devel_ubl branch.   It has the stuff you want.

thinkyhead, 2016-09-07:
> the marlin rc6-bugfix posted on google drive

Do you have some means of comparing files to see where they fixed the X axis movement versus where the current Marlin code (or configuration) is different? There are several tools and text editors that can compare files side-by-side. You can even just drop files on top of the Marlin tree and use the Github Desktop app to see the differences. Maybe the specific change you're looking for will be obvious.

closed 2017-10-31 by thinkyhead