MarlinFirmware/Marlin #4715: ABL not working in RCBugFix 2016/08/27?

JimBrown, 2016-08-27:
Auto bed leveling does not appear to be doing anything in RCBugfix from today. G28 and G29 commands run as expected, but no Z compensation is observed when printing. When using the grid, running G29 S0 causes the X and Y axis to ever so slowly move and no report is generated.

Configuration file attached. I'm hoping it's something I've done.

[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/440760/Configuration.h.txt)

Blue-Marlin, 2016-08-27:
Grid based leveling and 3-point leveling
```
  /**
   * G29: Detailed Z probe, probes the bed at 3 or more points.
   *      Will fail if the printer has not been homed with G28.
   *
   * Enhanced G29 Auto Bed Leveling Probe Routine
   *
   * Parameters With AUTO_BED_LEVELING_GRID:
   *
   *  P  Set the size of the grid that will be probed (P x P points).
   *     Not supported by non-linear delta printer bed leveling.
   *     Example: "G29 P4"
   *
   *  S  Set the XY travel speed between probe points (in units/min)
   *
   *  D  Dry-Run mode. Just evaluate the bed Topology - Don't apply
   *     or clean the rotation Matrix. Useful to check the topology
   *     after a first run of G29.
   *
   *  V  Set the verbose level (0-4). Example: "G29 V3"
   *
   *  T  Generate a Bed Topology Report. Example: "G29 P5 T" for a detailed report.
   *     This is useful for manual bed leveling and finding flaws in the bed (to
   *     assist with part placement).
   *     Not supported by non-linear delta printer bed leveling.
   *
   *  F  Set the Front limit of the probing grid
   *  B  Set the Back limit of the probing grid
   *  L  Set the Left limit of the probing grid
   *  R  Set the Right limit of the probing grid
   *
   * Global Parameters:
   *
   * E/e By default G29 will engage the Z probe, test the bed, then disengage.
   *     Include "E" to engage/disengage the Z probe for each sample.
   *     There's no extra effect if you have a fixed Z probe.
   *     Usage: "G29 E" or "G29 e"
   *
   */
```

Mesh bed leveling:
```
  /**
   * G29: Mesh-based Z probe, probes a grid and produces a
   *      mesh to compensate for variable bed height
   *
   * Parameters With MESH_BED_LEVELING:
   *
   *  S0              Produce a mesh report
   *  S1              Start probing mesh points
   *  S2              Probe the next mesh point
   *  S3 Xn Yn Zn.nn  Manually modify a single point
   *  S4 Zn.nn        Set z offset. Positive away from bed, negative closer to bed.
   *  S5              Reset and disable mesh
   *
   * The S0 report the points as below
   *
   *  +----> X-axis  1-n
   *  |
   *  |
   *  v Y-axis  1-n
   *
   */
```

With grid/3point-leveling a veeeeeery low speed for "S0" is expected.

What do you want to achieve with "S0"?

JimBrown, 2016-08-27:
Ah. Misread the info here: http://reprap.org/wiki/G-code#G29:_Detailed_Z-Probe. Off to do some more testing with the above info. Thanks.

thinkyhead, 2016-08-27:
> Or is that a TBD thing?

Very short on help writing documentation.

JimBrown, 2016-08-31:
Ok, ABL working as it should, now that I've read the correct instructions.

closed 2016-08-31 by JimBrown