MarlinFirmware/Marlin #4716: Z offset has no effect

tyssoon, 2016-08-27:
hi guys, 

i have a diy delta and when i launch a print, after a G28 and G29, the nozle is at 2 mm of the print surface. I was at 13.4mm on z offset on marlin. i changed it to 11mm and the nozzle were still at 2 mm above the bed.  I put 5mm on marlin on z offset to be sure, and same thing, 2mm.

tell me what you need to help me guys ! :)

thanks by advance.

brainscan, 2016-08-27:
Do you have EEPROM on?

thinkyhead, 2016-08-28:
What version of Marlin are you using? I'd recommend re-downloading `RCBugFix` from today and re-testing. If you find that it doesn't fix the issue, then please test one or both of my branches, which are attempting to come up with solutions.

This one adjusts Z based on the homing position:
https://github.com/thinkyhead/Marlin/tree/rc_fix_leveling_maths

And this one adjusts Z based on the final probe position:
https://github.com/thinkyhead/Marlin/tree/rc_final_z_correction

tyssoon, 2016-08-28:
@thinkyhead 
Hi, can i download last marlin or one of yours, replace the configuration.h and configurationadv.h by mines without problem ?

@brainscan 
im almost sure that EEPROM is on, if i undertood well what it is.

when i woke up i wonder if it could be my heinght definition which could be wrong.

brainscan, 2016-08-28:
If you have EEPROM enabled then your printer may be using the original offset you entered. If you made the changes in configuration.h and uploaded you need to send M502 followed by M500 to update and save the new settings.

tyssoon, 2016-08-28:
swapped the config.h and obtained that

`Arduino : 1.6.10 (Windows 7), Carte : "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\MarlinConfig.h:39:0,

                 from C:\Users\PCQUIVAVITE\Desktop\Marlin-RCBugFix\Marlin\Marlin.ino:37:

SanityCheck.h:137: error: #error "\"dropsegments\" is replaced with MIN_STEPS_PER_SEGMENT (and increases by 1). Please update Configuration_adv.h."

   #error "\"dropsegments\" is replaced with MIN_STEPS_PER_SEGMENT (and increases by 1). Please update Configuration_adv.h."

    ^

exit status 1
#error "\"dropsegments\" is replaced with MIN_STEPS_PER_SEGMENT (and increases by 1). Please update Configuration_adv.h."

i have found drop sagment, replaced it and add one to 5, no changement.

@brainscan

m500+m502 have changed nothing, so i think this is not an eeprom problem. I change the "#define CUSTOM_MACHINE_NAME" everytime i make a modification, to be sure the firmware is good just looking at the screen, so i think it's ok.

`

universam1, 2016-08-28:
So Bug as mine #4708 
+1

tyssoon, 2016-08-28:
so the rcbugfix is still bugged ? is it worth that i bothered config it ? or do i try another ?

brainscan, 2016-08-28:
Sorry if it hasn't helped but did you do m500 m502 or m502 m500? Both will give different results. When you power up it displays the settings that are being used so you can easily verify if things are changing or not. Changing the machine name would have no affect over it reading from EEPROM or not.

Nocturnal42, 2016-08-28:
Personally, I've noticed that using M581 to change the offset does not actually work with G29, even after M500 and M503 to verify, it still uses the default offset regardless of what you set it to. I've had to resort to repeatedly modify->compile->upload to fine tune my offset. Using RCBugFix from about 12 hours ago.

tyssoon, 2016-08-28:
@thinkyhead 
> All it takes is one typo. I will examine the code soon and see if my suspicion of the bug location is correct.

i'll wait so :)

@brainscan 

> Sorry if it hasn't helped but did you do m500 m502 or m502 m500? Both will give different results. When you power up it displays the settings that are being used so you can easily verify if things [...]

M500 then m502, no results.
M502 M500. neither

anyway thanks to trying to help, even it doesnt :)

@Nocturnal42

> Personally, I've noticed that using M581 to change the offset does not actually work with G29, even after M500 and M503 to verify, it still uses the default offset regardless of what you set it to.

M851 ofc; not 581 :D

> I've had to resort to repeatedly modify->compile->upload to fine tune my offset

di you managed to fix the z offset ? If yes, can you be more specific in the way you did please ?

thinkyhead, 2016-08-28:
All my examination of the code shows that the value being used is the one stored in the variable `zprobe_zoffset` as set by `M851`. If you don't mind, please post a log. First, re-download the latest `RCBugFix` which has lots of logging. Enable `DEBUG_LEVELING_FEATURE` and issue the command `M111 S255` before doing a `G28` and `G29`. Post the log output here so we can see what's going on.

tyssoon, 2016-08-28:
i cant comùpile last rcbug fix:

> Arduino : 1.6.10 (Windows 7), Carte : "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\MarlinConfig.h:39:0,

                 from C:\Users\PCQUIVAVITE\Desktop\Marlin-RCBugFix\Marlin\Marlin.ino:37:

SanityCheck.h:137: error: #error "dropsegments" is replaced with MIN_STEPS_PER_SEGMENT (and increases by 1). Please update Configuration_adv.h.

   #error "dropsegments" is replaced with MIN_STEPS_PER_SEGMENT (and increases by 1). Please update Configuration_adv.h.

    ^

exit status 1
#error "dropsegments" is replaced with MIN_STEPS_PER_SEGMENT (and increases by 1). Please update Configuration_adv.h.

Nocturnal42, 2016-08-28:
Changing `Z_PROBE_OFFSET_FROM_EXTRUDER` is having no effect, and apparently was having no effect previously. Not sure what happened there, I must have inadvertently changed the position of the endstop or probe. 

<pre>echo:DEBUG:ECHO,INFO,ERRORS,DRYRUN,COMMUNICATION,LEVELING
ok
echo:G28
>>> gcode_G28
current_position=(70.00, 90.00, 0.40) : setup_for_endstop_or_probe_move
> endstops.enable(true)
Raise Z (before homing) to 5.00
>>> do_blocking_move_to(70.00, 90.00, 5.00)
>>> homeaxis(X)
current_position=(0.00, 90.00, 5.00) : sync_plan_position
echo:busy: processing
> 1st Home-330.00
current_position=(0.00, 90.00, 5.00) : sync_plan_position
current_position=(0.00, 90.00, 5.00) : sync_plan_position
> 2nd Home-10.00
>>> set_axis_is_at_home(X)
For X axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = 0.00
 soft_endstop_max = 220.00
> home_offset[X] = 0.00
current_position=(0.00, 90.00, 5.00) : 
<<< set_axis_is_at_home(X)
current_position=(0.00, 90.00, 5.00) : sync_plan_position
current_position=(0.00, 90.00, 5.00) : > AFTER set_axis_is_at_home
<<< homeaxis(X)
current_position=(0.00, 90.00, 5.00) : > homeX
>>> homeaxis(Y)
current_position=(0.00, 0.00, 5.00) : sync_plan_position
echo:busy: processing
> 1st Home-352.50
current_position=(0.00, 0.00, 5.00) : sync_plan_position
current_position=(0.00, 0.00, 5.00) : sync_plan_position
> 2nd Home-10.00
>>> set_axis_is_at_home(Y)
For Y axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = 0.00
 soft_endstop_max = 235.00
> home_offset[Y] = 0.00
current_position=(0.00, 0.00, 5.00) : 
<<< set_axis_is_at_home(Y)
current_position=(0.00, 0.00, 5.00) : sync_plan_position
current_position=(0.00, 0.00, 5.00) : > AFTER set_axis_is_at_home
<<< homeaxis(Y)
current_position=(0.00, 0.00, 5.00) : > homeY
>>> homeaxis(Z)
current_position=(0.00, 0.00, 0.00) : sync_plan_position
echo:busy: processing
> 1st Home-270.00
current_position=(0.00, 0.00, 0.00) : sync_plan_position
current_position=(0.00, 0.00, 0.00) : sync_plan_position
echo:busy: processing
> 2nd Home-4.00
>>> set_axis_is_at_home(Z)
For Z axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = 0.00
 soft_endstop_max = 180.00
*** Z HOMED TO ENDSTOP (Z_MIN_PROBE_ENDSTOP) ***
> home_offset[Z] = 0.00
current_position=(0.00, 0.00, 0.00) : 
<<< set_axis_is_at_home(Z)
current_position=(0.00, 0.00, 0.00) : sync_plan_position
current_position=(0.00, 0.00, 0.00) : > AFTER set_axis_is_at_home
[... 208 lines cut ...]
deploy: 0
do_probe_raise(15.00)
>>> do_blocking_move_to(164.00, 230.00, 15.00)
echo:busy: processing
echo:busy: processing
>>> do_blocking_move_to(164.00, 230.00, 15.00)
current_position=(164.00, 230.00, 15.00) : clean_up_after_endstop_or_probe_move
current_position=(164.00, 230.00, 15.00) : > probing complete
Eqn coefficients: a: 0.00097561 b: -0.00019345 d: 1.86576261
uncorrected_position=(164.00, 230.00, 15.00) : >>> set_bed_level_equation_lsq
current_position=(164.00, 230.00, 15.00) : >>> set_bed_level_equation_lsq
corrected_position=(164.01, 230.00, 14.88) : <<< set_bed_level_equation_lsq
current_position=(164.01, 230.00, 14.88) : sync_plan_position

Bed Level Correction Matrix:
+1.000000 +0.000000 +0.000976 
+0.000000 +1.000000 -0.000193 
-0.000976 +0.000193 +1.000000 
> BEFORE apply_rotation_xyz > stepper_z = 15.00 ... z_tmp  = 14.88
> AFTER apply_rotation_xyz > z_tmp  = 15.05
current_position=(164.01, 230.00, 14.94) : sync_plan_position
current_position=(164.01, 230.00, 14.94) : > corrected Z in G29
<<< gcode_G29
X:164.01 Y:230.00 Z:14.94 E:0.00 Count X: 13120 Y:18400 Z:6021
</pre>

tyssoon, 2016-08-28:
is there a way to not  manually reconfigure the configs.h files ?

thinkyhead, 2016-08-28:
> i cant compile last rcbug fix:

It's telling you to replace…
```cpp
const int dropsegments = 5;
```
…with…
```cpp
#define MIN_STEPS_PER_SEGMENT 6
```
…as you can see if you look at the (unaltered) `Configuration_adv.h` file.

thinkyhead, 2016-08-28:
> is there a way to not manually reconfigure the configs.h files ?

No. There are only so many hours in a day, and I haven't gotten around to writing conversion tools.

thinkyhead, 2016-08-29:
@Nocturnal42 I notice that your probe triggers regularly when your nozzle is at least 1.8mm above the bed, implying that the correct probe offset for your setup is closer to -1.8 than -5.00 as shown. There's one outlier, which triggers the probe earlier, at 2.3, a half-millimeter higher than the other probes.

Anyway, it should be noted that actually **the probe offset doesn't matter in bed leveling**, because all it does it determine the tilt of the bed. The only place where the probe offset can mess with the Z position is if you used the probe to do Z homing also.

Roxy-3D, 2016-08-29:
@Nocturnal42   It would be prudent to do an M48 where ever on the bed you home the Z-Axis.   Let's make sure your probe 'repeats' well enough that it isn't causing problems.   Because, if it doesn't have good repeat ability, we will never get this resolved.

thinkyhead, 2016-08-29:
@Roxy-3D Most of those probed points seem fine to me. But do you think that one outlier is throwing anything off?

Roxy-3D, 2016-08-29:
Yes.   Remember, the whole purpose of a Least Squares Fit is to make sure no point is 'too far' off.   The error from any point is squared, so in a sense, it is exponential.    One 'bad' point, especially, if it it is off a bunch, will pull things much more than it 'should' in that direction.

> It would be prudent to do an M48 where ever on the bed you home the Z-Axis. 

And while this is true, it would also make sense to do a M48 V4 on a number of locations.  Especially the locations where there is any 'outlier' point.   Is this an 'inductive' probe?   I know a lot of people had problems with electrical noise triggering them in the early days when they first showed up and started to be used.   I wouldn't be surprised if there are repeat-ability issues when you are probing near the edge of the bed.

mjmeans, 2016-08-29:
Does the current Least Squares Fit use Danish robust estimation to diminish outliers that are more than 2 times standard deviation? If not maybe this should be considered to reduce the effect of signal noise specially when using z probe designs that induce considerable noise into the measurements. [see](http://a-a-r-s.org/aars/proceeding/ACRS2008/Papers/TS%2024.5.pdf).

tyssoon, 2016-08-29:
@thinkyhead 

> @tyssoon It's telling you to replace…

const int dropsegments = 5;
…with…

#define MIN_STEPS_PER_SEGMENT 6
…as you can see if you look at the (unaltered) Configuration_adv.h file.

my attempt was: const int  MIN_STEPS_PER_SEGMENT = 6; not so far ! :D

i managed to correct the rest of the errors alone like a big boy. i will postthe results soon.

thanks !

tyssoon, 2016-08-29:
hi again !

my log of m11/g29/g28 and print then reset.

[m111 - Copie (2).txt](https://github.com/MarlinFirmware/Marlin/files/442006/m111.-.Copie.2.txt)

Tell me if it helped !

thinkyhead, 2016-08-29:
@tyssoon If you do a print with just `G28` and no `G29` is your nozzle still 2mm above the print surface?

I keep forgetting that the probe Z offset has nothing to do with the home Z position.

If you find that your nozzle is still too high, even if you don't use `G29`, then perhaps try reducing your `Z_HOME_POS` (and/or `Z_MANUAL_HOME_POS`) by 2.

tyssoon, 2016-08-29:
i dont understand what the z offset is for so ^^

but u were right, i tuned z home pos to 685 and now it's okay i can print. i will test what happen with a G29 before.

i have to sand a bit my effector to be sure that i can print.

Roxy-3D, 2016-08-29:
> i dont understand what the z offset is for so

z_offset if primarily for the Mesh Bed Leveling schemes to adjust the distance from the Mesh to the nozzle.  It is also used by the Z-Baby-Stepping.    If you are not using Mesh Bed Leveling, just set it to 0.000

```cpp
  class bed_leveling {
  public:
	struct ubl_state {
		bool active;
		float z_offset;
		int  EEPROM_storage_slot = -1;
		int  n_x = MESH_NUM_X_POINTS;
		int  n_y = MESH_NUM_Y_POINTS;
		float mesh_x_min = MESH_MIN_X;
		float mesh_y_min = MESH_MIN_Y;
		float mesh_x_max = MESH_MAX_X;
		float mesh_y_max = MESH_MAX_Y;
		float mesh_x_dist = MESH_X_DIST;
		float mesh_y_dist = MESH_Y_DIST;
		float G29_Correction_Fade_Height = 10.0; 
		unsigned char padding[28];  	// This is just to allow room to add state variables without
						// changing the location of data structures in the EEPROM.   
						// This is for compatability with future versions to keep 
						// people from having to regenerate thier mesh data.
						//
						// If you change the contents of this struct, please adjust
						// the padding[] to keep the size the same!
	} state, pre_initialized;

    float z_values[MESH_NUM_X_POINTS][MESH_NUM_Y_POINTS];
    float mesh_index_to_X_location[MESH_NUM_X_POINTS+1];	// +1 just because of paranoia that we might end up on the
    float mesh_index_to_Y_location[MESH_NUM_Y_POINTS+1];	// the last Mesh Line and that is the start of a whole new cell

    bed_leveling();	// { reset(); }
```

Roxy-3D, 2016-08-29:
> Does the current Least Squares Fit use Danish robust estimation to diminish outliers that are more than 2 times standard deviation? If not maybe this should be considered to reduce the effect of [...]

No it doesn't throw out an outliers.   Especially in the case of Deltas, I think we expect there to be 'outliers' because of the cup and bowl shape of the bed.

Blue-Marlin, 2016-08-29:
@Roxy-3D 
At witch place in the delta levelling is LSF in use?

Roxy-3D, 2016-08-29:
> At which place in the delta levelling is LSF in use?

As you know, it isn't currently used for Delta's.   But it will be very soon in the UBL branch.

mjmeans, 2016-08-29:
@Roxy-3D 
Printers without a cup or bowl potential (or delta printers if they are known to have minimal (less than 0.1mm) cup or bowl issue) can use robust regression to a plane.

Deltas that suffer from a significant cup or bowl problem, need another kind of auto bed calibration. In this case the head is moving in a parabolic shape which will also have the effect of an xy scaling problem near farther away from the center. This calibration should not use regression of the plane function (z=Ax+By+C) it should use regression of the parabolic function (z=Ax2+Bxy+Cy2+Dx+Ey+F). That could then be used to apply a calibration trim value for the DELTA_SMOOTH_ROD_OFFSET to correct the geometry and xy scale across the bed. After that is calculated and applied you do another probe of the entire bed and regression to the plane function to correct any remaining tiny Z offsets.

Roxy-3D, 2016-08-29:
Yes.  I understand what you are saying.   I am not rejecting those thoughts.  But let me help you understand why I'm being slow accepting the ideas.   

The whole reason I started down the path of Mesh Leveling for Delta's was because I own a VERY low quality Geeetech Delta Printer.   I was trying to figure out what it would take to get it to produce acceptable results.    I don't have the code up and running yet on Delta's.  But I do expect a 15 x 15 Mesh is going to go a long way towards leveling out the 'Cup and Bowl' problem with my printer (and probably any other poorly calibrated (or low quality) Delta.   

I don't have numbers yet.  But once I get the Delta's up and running, I'm going to bring up the Grid Based ABL correction on top of the High Resolution Mesh correction.   I some what expect to see 'outliers' when I start probing the the edge of the Print Radius.   I don't want to throw out data.  But by the same token, that may be the right answer to get a good plane defined in the event some dirt or plastic pieces end up under the glass.   I will keep your idea of throwing out points that are too far off the plane defined by the other points in my back pocket.  I hope I won't need to resort to that 'fix'.   But if that is what it takes to get the Grid Based running well, you can bet it will end up in the code!

thinkyhead, 2016-08-30:
> i dont understand what the z offset is for so ^^

I'm beginning to wonder, myself. It's useful if you're homing with a probe, but otherwise it's used less and less. For instance, we changed the `Z_RAISE_*` options into `Z_*_HEIGHT` options, so `Z_OFFSET_FROM_EXTRUDER` offset is no longer used to ensure a minimum height before/between/after probing. (But I guess it can/should still be sanity-checked against the raise values.)

> tuned z home pos

The easiest way to tune the homed height is to simply do homing, then move the nozzle all the way down to the bed, and if it isn't zero, adjust the z home position by the difference.

closed 2016-08-30 by thinkyhead

tyssoon, 2016-08-30:
i have a bowl bed, i dont understand how since it's a borosilicate which cost me an arm but i do. So is the autolevel useful  or not ? do i have to bother ?

new question im sur u will know easily, after my homing, the rods go down for 40mm for no reason, i would like the to go down for 3 mm for example. where is the place in marlin where i can change that please.

Blue-Marlin, 2016-08-30:
Dear @tyssoon 
If you have a delta  its much more likely you calibration is off, then the bed is not flat.

You coul try to remove or replace https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Marlin_main.cpp#L3240-L3243

thinkyhead, 2016-08-30:
@tyssoon If you lay a straight-edge over the bed, I bet it fits right snug against it and shows it to be flat. Tuning a delta is tricky, but when out of tune, the height at center and at the edges will differ. Keep tuning.

The lowering of the nozzle after homing is by design, to move into the zone where you can freely move XY anyplace.

tyssoon, 2016-08-30:
i go on vacation tommrow guys. i'll be around, but not near my delta. so i'll pause the tuning for some time, seeya in a week or two. 

And thanks !