MarlinFirmware/Marlin #4724: Fwretract causes odd pausing during printing rc7

scruffy4386, 2016-08-28:
since changing over to rc7 from rc6, when I activate the fwretract option, the machine retracts fine, but the return happens slowly ( 3-5 seconds). If I leave it turned off, the print continues, but with massive stringing. I have adjusted the un retract speed as high and as low as possible, both with no visible change. I think I've done all the standards to test it out before making an issue of it. I can provide more information if needed, however I didn't see anyone else putting this issue up, so maybe it's just a small configuration issue.

Blue-Marlin, 2016-08-28:
This is already fixed in RCBugFix.

scruffy4386, 2016-08-28:
thank you. my apologies for not reading further

closed 2016-08-28 by scruffy4386