MarlinFirmware/Marlin #4728: Marlin off lcd and refresh lcd

wqchan1, 2016-08-29:
I want to control the LCD, when I want to display and then open the LCD, turn off the LCD when not in use. Another problem, how to refresh the LCD. Because when I did not take before energizing MEGA2560 LCD, the LCD display is not normal, how to refresh the LCD?

thinkyhead, 2016-08-30:
I've been thinking of something just like that. A timeout during printing, after which the LCD would "go to sleep" until you pressed the button to wake it up again. This would help with Delta printing, especially, because we need all the CPU power we can get.

marcio-ao, 2018-03-02:
> This would help with Delta printing, especially, because we need all the CPU power we can get.

@thinkyhead : Once everyone goes 32-bits, we can use all that extra cycles to implement OpenGL screensavers for the LCD! :D

AnHardt, 2018-03-02:
Refresh? Pushing the button twice (go menu and back) should reinitialize the display.
Currently the next display refresh (normally ~1/s) is delayed when there are not enough moves in the buffers to refresh the display twice, until there is enough time. Shutting the display off completely should not be needed.

If you mean switching off the backlight - sorry. I'm not aware of any 3D-printer panel able to do that without additional hardware.

boelle, 2019-02-18:
this will require that we can turn off the power to the backlight as well or else it would be rather pointless

and with a re-arm there is no need to conserve cpu cycles... i would vote down this one or even reject it

seanboothm, 2019-03-24:
I would love to have this feature, but after looking at the schematics for the two most popular displays, it appears the hardware will never allow this feature to happen. 

On the [Reprap Discount Smart Controller](https://reprap.org/mediawiki/images/7/70/Controller_final_reprapdiscount.pdf), the LED backlight is pins 15 and 16 [(Page 8 of the DMC204LCD datasheet)](http://pdf.dzsc.com/DMC/DMC20481NY-LY-AGE_1210738.pdf) which are tied directly to VCC and GND.

On the [Reprap Discount Full Graphic Smart Controller](https://reprap.org/mediawiki/images/5/51/RRD_FULL_GRAPHIC_SMART_CONTROLER_SCHEMATIC.pdf), pins 19 and 20 [(KXM12864M-3 datasheet)](http://shop.aiscube.com/datasheet/Displays/GLCD12864/KXM12864-M3.pdf) control the backlight and are also tied to VCC and GND.

Unless this feature is worth looking into for the various other display controllers (which may or may not even allow this possibility), this FR should probably be closed.

universam1, 2019-03-25:
There are many features in this firmware that work only with specific hardware or even modifications. Otherwise this firmware would be branded, right. 
So for those who are willing to customize by connecting a pin to the lcd this would be a very valuable feature

thisiskeithb, 2022-02-26:
Added in https://github.com/MarlinFirmware/Marlin/pull/23768