MarlinFirmware/Marlin #4730: RAMBo 1.3 - RCBugFix Fan Control and Setup - Fans not working

dvseliteus, 2016-08-29:
Marlin-RCBugFix with RAMBo 1.3 Board none of my fans are powering on during operation.
http://reprap.org/mediawiki/images/b/ba/Rambo1-2-conn-main.jpg
I have my Hotend Fans at FAN0 and FAN1. My cooling fans are at FAN2.
I'm not sure how to set these up?

thinkyhead, 2016-08-30:
Try adding to your configuration:
```cpp
#define EXTRUDER_0_AUTO_FAN FAN_PIN
#define EXTRUDER_1_AUTO_FAN FAN1_PIN
#define CONTROLLER_FAN FAN2_PIN
```

dvseliteus, 2016-08-30:
@thinkyhead Thank you for the info.
I added
#define EXTRUDER_0_AUTO_FAN FAN_PIN
#define EXTRUDER_1_AUTO_FAN FAN1_PIN
#define CONTROLLER_FAN FAN2_PIN
to the configuration file and
#define FAN_PIN            8
#define FAN1_PIN           6
#define FAN2_PIN           2
to the RAMBo.pins file
All fans are working well now :+1:

thinkyhead, 2016-08-30:
I believe those fan pins are already in the RAMBo pins file, and you now have them in there twice!

dvseliteus, 2016-08-30:
Here is what it had in the rambo.pins file
[image]
The circle in red is what I had to add to get it working.
The info was there for the pin assignments but I had to define in the code.
I don't really know what I am doing so I am sure to someone more experience this was a straight forward thing, but now I know what to do.
Thanks!

thinkyhead, 2016-08-30:
You must be using older Marlin.
```cpp
//
// Heaters / Fans
//
#define HEATER_0_PIN        9
#define HEATER_1_PIN        7
#define HEATER_2_PIN        6
#define HEATER_BED_PIN      3

#define FAN_PIN             8
#define FAN1_PIN            6
#define FAN2_PIN            2

//
// Misc. Functions
//
```

dvseliteus, 2016-08-30:
Quite possible. This was the RCBugFix version.
Non the less thank you for your help!

closed 2016-08-30 by dvseliteus

thinkyhead, 2016-08-30:
> This was the RCBugFix version.

Dig it. Changing every day, it is.

dvseliteus, 2016-08-30:
Change is good! Next build will be a delta :-)

dvseliteus, 2017-09-21:
@thinkyhead 
Recently upgraded my RAMbo 1.3 board to Marlin 1.1.4. The Fans are not working correctly for some reason.

I have a single extruder and part cooling fan.
Extruder Fan – Connected to Fan 0
Part cooling Fan – Connected to Fan 2
Based on [Wiring Digram](http://reprap.org/wiki/File:Rambo1-2-conn-main.jpg)

Fan control on LCD is as follows;
Fan 1 – controls extruder fan
Fan 3 – controls part cooling fan
Both fans work when manually set on LCD

During print behavior;
Extruder fan – turns on after first layer
Part cooling fan – does not turn on

I have not modified any board pins or messed with fans in the configuration file. Any idea as to why they are not working correctly?