MarlinFirmware/Marlin #4731: Strange Motor Behaviour

AcrimoniousMirth, 2016-08-29:
Running 1.1.0-RC7 (two versions behind as of posting)
RAMPS 1.4 running CoreXY setup though standard setup doesn't make a difference.
Checked motors for physical reasons and all signs point to code.
Motors will move POSITIVE along ALL axis in reasonable steps. Tell it "+10X" and it moves +10X
Tell it "-10X" and it moves roughly -1X. This happens with all axis, regardless of microstepping level.

ALSO
Minimum endstops only used, tell it to home and it moves positive in all axis by about 50mm then stops. See in RC7 notes this may be a stack overflow and may be fixed.

ALSO
Rotary encoder steps not well scaled. When selecting move axis then through to move 1mm when turning 1 click it moves 4mm. Fiddling with options doesn't seem to make any difference.

ALSO
Option to turn D9 (or any sensible pin) on RAMPS etc into a constant 12V output would be excellent. Need some small microcircuits and don't have the room to splice into the PSU.

Other than that, love Marlin! Running it on two printers and set it up for several others too. Just fantastic! :D

thinkyhead, 2016-08-30:
Some or all of this is probably fixed in `RCBugFix`, soon to be released as RC8. Give it a try and report back.

AcrimoniousMirth, 2016-08-31:
Issue was solved by user, incorrect wiring was the culprit, Marlin not to blame!

closed 2016-08-31 by AcrimoniousMirth