MarlinFirmware/Marlin #4732: FeatureRequest - Change DualZAxis driver to a #-ZAxis Driver

lephino, 2016-08-29:
Several of us use more than 2 stepper motors for the z axis, especially when using larger machines.  It'd be nice to be able to just specify "3" or "4" z axis motors, instead of just simply enabling dual.   Since most of the dual logic was abstracted out, it seems the next step would be to increase this to more than 2.

thinkyhead, 2016-08-30:
@lephino I see more and more electronics with lots of spare stepper drivers, so this is not a bad idea. But it will have to wait for 1.1.1, I think. We'll just change the configuration options so you can specify how many X, Y, or Z steppers you want, and they will be assigned in order to the spare E connectors. Meanwhile, I note that boards like the Megatronics 3 allow you to chain as many external stepper drivers as you want, so that's another nice solution.

lephino, 2016-08-30:
I agree with your assessment, and like the approach.  Look forward to 1.1.1!

thinkyhead, 2018-12-10:
Currently we support up to 3 Z axis steppers in Marlin 2.0.x.

boelle, 2019-02-18:
@lephino with 3 z axis possible is this one covered?

lephino, 2019-02-20:
Yep, works for me!

closed 2019-02-20 by lephino