MarlinFirmware/Marlin #4735: Max Temp Accessible in LCD?

sarf2k4, 2016-08-30:
Hi, I am wondering, was marlin designed to have the following calculation?

```
max lcd temp = maxtemp - 15
```

I one of my printer to 285 for hotend max temp, max temp that I can access from lcd is just 270 (base rc7), same goes to this tevo tarantula that I limited it to 255, max is 240 from lcd menu here (rc6 bugfix)

birkett, 2016-08-30:
Yes, the code is doing exactly that:
https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/ultralcd.cpp#L1590

sarf2k4, 2016-08-30:
Can I request an option/variable introduced in configuration.h for max temp accessible on lcd?

The only variable or lines that I can think of for this one would be as follows
```
//Sets the maximum temperature accessible from lcd menu 
#define max_lcd_temp (max_temp - 15)
```
This reason alone is that I set my hot end max safety temp and printing temp to be very close e.g hotend with teflon tubes I set the max cutoff temp at 260c and my printing probably set from start of gcode to 240c, I wanted to increase the temp through the lcd to probably 255c. But with this limitation, I'm not able to reach the desired temp during printing.

Hope this changes can be considered =D

Blue-Marlin, 2016-08-30:
Without subtracting a (random) value here you will get maxtemp errors every time you have a overshot while heating.
You can reduce 15 to the value of your max overshot.

sarf2k4, 2016-08-30:
What about giving an error
```
if (max_lcd_temp == max_temp)
   echo "lcd max temp must be lower"
```

I was thinking to put it at least -5 from max temp would be ok for me. I rely on PID on controlling the temp. if the temp kept overshoot even though pid autotune were done several time, I would lower down the pwm max power. 

Just like this tevo I'm managing, it kept overshot the temp even with pid autotune roughly about -4 to +9, I reduced the pwm max for the hotend. After that the reading is only about +3c after reaching the target temp

thinkyhead, 2016-08-30:
I believe the settings you're looking for already exist. I will look it up and point out the options when I have a chance.

closed 2017-10-31 by thinkyhead