MarlinFirmware/Marlin #4736: Corexy (or other printer) with moving bed and raise_before_homing

DjDemonD, 2016-08-30:
On corexy, my bed starts at the bottom of the printer, it currently lifts up to home by using an ABL sensor at the print head. Initially I had to set raise_before_homing to 0 otherwise the firmware asked the printer to push the bed downwards before raising it to home, which was not possibly as it was at its physical max limit so the motor skipped. 

There seems to be an assumption in the firmware that nozzles lift when moving z towards max direction, rather than beds descending to move towards max endstops.

What I (and several other SmartRapCore users) are looking to do now is switch to a microswitch/opto z_max endstop and do away with ABL. The bed would home at z max and then lift after heating to begin printing, like a delta but upside down. However I would like to have the firmware effectively raise the bed (move towards z min direction, which in this configuration is perfectly safe I have 170mm of travel betwen z-max and z=0) say 5-10mm before attempting to home to z_max (move downwards). I've tried entering a negative value for raise_before_homing but this doesn't do anything.

thinkyhead, 2016-08-30:
>  I've tried entering a negative value for raise_before_homing but this doesn't do anything.

```diff
-        if (destination[Z_AXIS] > current_position[Z_AXIS]) {
+        if (destination[Z_AXIS] != current_position[Z_AXIS]) {
```

For machines that home Z with a probe, we actually make no assumptions. We cannot, because before homing we have _no clue_ where Z is located. So there's a caveat with `Z_HOMING_HEIGHT`, that if your Z is already at the top-end (or bed at the bottom) you will grind the axis against the frame. And if you set a negative value, you might _bang your bed into the nozzle_, which is often worse.

If you know, _absolutely_, that your nozzle and bed will always start very far apart, then you may use zero (or a negative value) for `Z_HOMING_HEIGHT` and zero for `Z_PROBE_DEPLOY_HEIGHT`.

If you plan to home Z to max then you can simply home to max like on any other machine, and you will get Z homing first before XY.

A more comprehensive solution —when homing with a probe— would be to first check and see if the probe is triggered when it's first deployed/enabled, and if so, move away from the bed by at least the `Z_PROBE_TRAVEL_HEIGHT`. That's about the only way that the distance to the probe can be gleaned.

> …what the idea behind the config files have version numbers is?

The configuration version number will be matching release version numbers. We're not changing it with every RC.

DjDemonD, 2016-08-31:
Thanks Scott.

closed 2016-08-31 by DjDemonD