MarlinFirmware/Marlin #4737: UBL Phase 2 manual probing

OrbitAVI, 2016-08-30:
@Roxy-3D, when I run P2 to probe the remaining parts of my bed the probe always tries to go off the right edge before returning to normal operation. Hopefully I can explain this right. 

I run P1 without issue, however due to my probe offset the left edge of my bed cannot be probed (X home). When I run G29 P2 the print head immediately tries to go to some position that does not exist. It homes y then attempts to go past the X max. Right after that it starts at X 0 Y 0 probing the correct spots that need to be probed manually.

Any idea what could cause it to first run off the right edge causing it to crash?

Roxy-3D, 2016-08-30:
For the Manual probing, if you don't specify a reference point, it tries to start at one end of the unprobed points and work its way around the two edges.    It may be that there is  a bug in how it determines the start location for your setup.

> I run P1 without issue, however due to my probe offset the left edge of my bed cannot be probed (X home). When I run G29 P2 the print head immediately tries to go to some position that does not [...]

If you could post a video, of it moving wrong, that would be very helpful to figure out why the logic is wrong.

But with that said, you have a couple of options.    First, you can manually move the nozzle to one end of the L shaped un-probed area and then do a G29 P2 C    which will tell it to use the Current location as its reference point.  It will start trying to find points close to the start location and work its way around the un-probed area.

Another, quicker option is to just ignore the Manual probing with G29 P2 and do a G29 P3 R C .25    That will fill the entire un-probed area with .5mm rise of the nozzle.    With all of the values of the Mesh filled in, you can start a G29 from the far corner away from the unprobed regions.   Let the G26 run and be prepared to stop it (by pressing and holding the Encoder Wheel for the current Mesh location to finish).   If you see the nozzle getting too close to the glass, stop it, otherwise, let it run.

Once you have the G26 done, you can move the nozzle to any area where things don't look right and edit the Mesh with a G29 P4 R.

In case you didn't understand what I'm saying:  I think it is much quicker to not manually probe the bed.  Instead, let G26 show you how much of a 'correction' from a 'safe value' is necessary.

OrbitAVI, 2016-08-30:
I will post a video as soon as I can. Just to clarify. The un-probed area is not L shaped. It is just along the "west edge" sorry if that makes no sense. It's basically X 0 all the way through the length of the bed. The point that it travels to is all the way off to the other side past X200 (bed is 200 x 300) after attempting to do that it goes to X0 Y0 and starts probing the un-probed area. So I am able to probe the spots. It's just the odd move it does before it starts that.. 

Sorry I can't think of a good way to explain it. I will post a video as soon as I can.

Roxy-3D, 2016-08-30:
Can you tell me the X_OFFSET_FROM_EXTRUDER and Y+OFFSET_FROM_EXTRUDER ?   My guess is with one of those being 0.00 we don't end up with an L shaped area that is unprobed and that maybe where the funny move is coming from.

OrbitAVI, 2016-08-30:
Y0, X32

OrbitAVI, 2016-08-31:
@Roxy-3D This really doesn't relate to this, but I'm wondering where to start. I've been searching forums etc.. to no avail. Can I define my home offsets outside of my bed dimensions? I am thinking of doing this so I can probe the entire bed withe the sensor.  Right now I cannot do this because of the X limit switch. It stops the print head right at the edge of the left side. If I allow it to travel past that to probe it that would be ideal. I'm thinking I can do this using the Home offsets, but I can't find much documentation on this or at at least i am not searching for the right thing.

OrbitAVI, 2016-08-31:
I'm thinking I can do this by setting my X min travel to a negative value (value outside of the print bed. In my case it would be about -30. This would put the sensor right where I want it.

Roxy-3D, 2016-08-31:
For the Mesh Bed Leveling, (or UBL) you are not allowed to have negative coordinates (yet).    That will be allowed soon in the UBL System.     I understand why you want to do this, but I think historically, it has been assumed the user would want the bed size to be the full X & Y travel.   You are kind of saying the opposite,  You want to be able to probe the entire bed, even if that means the bed size has to be smaller.

Setting up the Mesh is a one time deal.   And you can manually fill the un-probed regions.   At that point the G26 does a very good job of 'Validating' the Mesh.  And areas that are not perfect can be quickly edited.     My suggestion is to not worry about getting the entire Mesh constructed with the probe.

Have you had a chance to try the G26 and follow that up with a G29 P4 R yet?

OrbitAVI, 2016-08-31:
@Roxy-3D It was more of a thought to further automate it for myself. The mesh that is automatically created is spot on. From there I can P2 to get the reset very easily (with the exception of that odd first move). 

Unfortunately I cannot get a G26 test print to complete with my volcano and .8 nozzle. I'll play around with it to see if I can get one to print. Retraction and speed is touchy with that nozzle. I can tell you however that I am now printing edge to edge without issue which I never could do that before. My first layers are very nice!

Roxy-3D, 2016-08-31:
The G26 was coded for 'normal' nozzles.  But I did add some retraction settings to let the user have some control.  And you can control how much filament gets extruded by shifting the filament size number up or down.    If you can take a picture of what is happening with the G26, I might be able to add some more control so you can get the Validation print to complete.    

I am convinced that the G26 Validation Print is half of the value of the UBL System.  It lets you see exactly where you have an issue and very quickly edit it.

OrbitAVI, 2016-08-31:
I'll try again. What speed does this print at? Is there a way to change that?

Roxy-3D, 2016-08-31:
The speed varies.  It depends on what you are talking about:
```cpp
  if ( x!=destination[X_AXIS] || y!=destination[Y_AXIS])  {	// Check if X or Y is involved in the movement.
	feed_value =PLANNER_XY_FEEDRATE()/(10.0);		// Yes!  It is a 'normal' movement
  } else  {
	feed_value = planner.max_feedrate[E_AXIS]/(1.5);	// it is just a retract() or un_retract()
  }
```
I have it set for 10% of the max XY feedrate for the printing moves.   And about 65% of the max feed rate of the extruder for Retract operations.   If those need to be made configurable to handle your nozzle, that can be done!

OrbitAVI, 2016-09-01:
I would suggest that you make the speed configurable in mm/s or mm/m for a future release.  I would also make the extrusion multiplier separate rather than an adjustment to the filament size. It just helps newbies like myself understand the settings a little more clearer as its similar to slicer settings. Again this is just a suggestion from a newbie, so I'm sure there are better arguments otherwise.

Roxy-3D, 2016-09-01:
That is not hard to do...    The biggest reason not to do it is it will add a few more parameters and that adds some extra complexity.    But I can set any new parameters to reasonable defaults so they can be ignored by most people most of the time.    Right now, in theory, the speed should not affect anything.  I have it set lower than the printer can handle.   

However, for the extrusion, we may need more control.   Besides filament amount and size, we may need more control of the retraction.   Very possibly, the speed of the retraction, the amount of the retraction, and how much 'extra' un-retraction is done.

Can you play with the filament size and see how much of a change you need to get your Volcano nozzle working?

OrbitAVI, 2016-09-01:
Yes I'll give it a shot tonight if I can or Sat.

Roxy-3D, 2016-09-01:
If you decide you need to play with the amount of retraction, you probably will have to tweak the source code.   Right now, it retracts 1.0mm and un-retracts by 1.2mm.   You can scale those numbers up or down with the 'M' option, but it is very possible that ratio is not going to work for you.   I'm guessing the default layer height of .2mm doesn't work with the Volcano.  But that one you can set with the 'L' option.

```cpp
void retract_filament()
{
  if ( !retracted ) {		// Only retract if we are not already retracted!
	retracted = true;
	move_to( destination[X_AXIS], destination[Y_AXIS], destination[Z_AXIS], -1.0 * Retraction_Multiplier );
  }
}

void un_retract_filament()
{
  if ( retracted ) {		// Only un-retract if we are retracted.
	move_to( destination[X_AXIS], destination[Y_AXIS], destination[Z_AXIS],  1.2 * Retraction_Multiplier );
	retracted = false;
  }
}
```

OrbitAVI, 2016-09-01:
Thank you I'll play around with it as soon as I can spend some time on it. The larger nozzle does have some challenges, but in a slicer its very easy to overcome. A .2 layer height hasn't been an issue so far. I'll report back as best I can. Bare with me I am still pretty new to this.

closed 2017-10-31 by thinkyhead