MarlinFirmware/Marlin #4739: Coding Noob: LCD Controller :"Move Axis" Step Size

AcrimoniousMirth, 2016-08-31:
I've tried playing around with the LCD options in the config file to no avail. I'm noticing that when I select "move 1mm" then turn 1 click on the rotary encoder it moves 4mm. This seems to be the same for all axis and all step sizes.
I'm a bit at loss and honestly asking someone with actual knowledge seems the best way to get this sorted quickly!
**How do I get the rotary encoder scaled so 1 click is 1 step, be it 10, 1 or 0.1mm?**
Thanks for the help!
Sincerely,
                _Lost and Confused Noob_

Roxy-3D, 2016-08-31:
If you turn on Baby Stepping...  I can assure you...   It doesn't move 1mm or 4mm steps.

I don't use the LCD to move nozzle (with the exception of the Z_BABYSTEPS very occasionally).   But these #define's will have an effect on what you are talking about:
~~~~
ENCODER_PULSES_PER_STEP
ENCODER_100X_STEPS_PER_SEC
ENCODER_10X_STEPS_PER_SEC
~~~~

Kaibob2, 2016-09-02:
Uncomment Line 1009
`//#define ENCODER_PULSES_PER_STEP 1`
and Line 1015
`//#define ENCODER_STEPS_PER_MENU_ITEM 5`
in Configuration.h and edit the values to your need.
For example, i use a RepRapDiscount FULL GRAPHIC Smart Controller which requires 
```
#define ENCODER_PULSES_PER_STEP 3
#define ENCODER_STEPS_PER_MENU_ITEM 1
```

AcrimoniousMirth, 2016-09-02:
Hi Kaibob2,
Thanks, I'm using the same controller. So having pulses per step at 3 works for you?
I'll try that. Thanks!

Kaibob2, 2016-09-02:
Yes,
`#define ENCODER_PULSES_PER_STEP 3
#define ENCODER_STEPS_PER_MENU_ITEM 1`
are taken from my installed marlin RC7 BugFix

AcrimoniousMirth, 2016-09-02:
Works perfectly! Thank you :)

closed 2016-09-02 by AcrimoniousMirth