MarlinFirmware/Marlin #4742: Auto Bed Leveling ,RC7 4Point,each detection 1,RC3  each detection 2？

jackshi618, 2016-08-31:
WHY?

my config

//===========================================================================
//============================ Bed Auto Leveling ============================
//===========================================================================

// @section bedlevel

#define AUTO_BED_LEVELING_FEATURE // Delete the comment to enable (remove // at the start of the line)

// Enable this feature to get detailed logging of G28, G29, M48, etc.
// Logging is off by default. Enable this logging feature with 'M111 S32'.
// NOTE: Requires a huge amount of PROGMEM.
//#define DEBUG_LEVELING_FEATURE

#if ENABLED(AUTO_BED_LEVELING_FEATURE)

  // There are 2 different ways to specify probing locations:
  //
  // - "grid" mode
  //   Probe several points in a rectangular grid.
  //   You specify the rectangle and the density of sample points.
  //   This mode is preferred because there are more measurements.
  //
  // - "3-point" mode
  //   Probe 3 arbitrary points on the bed (that aren't collinear)
  //   You specify the XY coordinates of all 3 points.

  // Enable this to sample the bed in a grid (least squares solution).
  // Note: this feature generates 10KB extra code size.
  #define AUTO_BED_LEVELING_GRID

  #if ENABLED(AUTO_BED_LEVELING_GRID)

    #define LEFT_PROBE_BED_POSITION 5
    #define RIGHT_PROBE_BED_POSITION X_MAX_POS-5
    #define FRONT_PROBE_BED_POSITION 20
    #define BACK_PROBE_BED_POSITION Y_MAX_POS-20

    #define MIN_PROBE_EDGE 10 // The Z probe minimum square sides can be no smaller than this.

    // Set the number of grid points per dimension.
    // You probably don't need more than 3 (squared=9).
    #define AUTO_BED_LEVELING_GRID_POINTS 2

  #else  // !AUTO_BED_LEVELING_GRID

    // Arbitrary points to probe.
    // A simple cross-product is used to estimate the plane of the bed.
    #define ABL_PROBE_PT_1_X 15
    #define ABL_PROBE_PT_1_Y 180
    #define ABL_PROBE_PT_2_X 15
    #define ABL_PROBE_PT_2_Y 20
    #define ABL_PROBE_PT_3_X 170
    #define ABL_PROBE_PT_3_Y 20

  #endif // !AUTO_BED_LEVELING_GRID

  #define Z_PROBE_END_SCRIPT "G1 X20 Y0 F9000\nG1 Z0.2 F6000\n" // These commands will be executed in the end of G29 routine.
                                                                             // Useful to retract a deployable Z probe.

  // If you've enabled AUTO_BED_LEVELING_FEATURE and are using the Z Probe for Z Homing,
  // it is highly recommended you also enable Z_SAFE_HOMING below!

#endif // AUTO_BED_LEVELING_FEATURE

Roxy-3D, 2016-08-31:
Why did I close this thread?   Because nobody has a clue what you need help understanding.

closed 2016-08-31 by Roxy-3D

Blue-Marlin, 2016-08-31:
```
// Use double touch for probing
//#define PROBE_DOUBLE_TOUCH
```

jackshi618, 2016-09-01:
Could not fiind this phrase.

thinkyhead, 2016-09-02:
@jackshi618 Please update to a newer version of Marlin. RC3 is an old, dusty, random, known-to-be-full-of-bugs release candidate that only fills us with shame and longing.

thinkyhead, 2016-09-02:
> Could not fiind this phrase.

"Even if you can't find it, you can still add it to your configuration." — Buddha
```cpp
//#define PROBE_DOUBLE_TOUCH
```

Blue-Marlin, 2016-09-02:
"If you can't find a configuration item in your config. - you likely have a too old version." - Confucius.

On the other hand this can't be root of the problem in RC7 and before. 
DELTA?

AnHardt, 2016-09-02:
> On the other hand this can't be root of the problem in RC7 and before. 

No. PROBE_DOUBLE_TOUCH option is in RC7.

"Don't copy old configs in new RCs" --- Me

jackshi618, 2016-09-02:
thank you,above you