MarlinFirmware/Marlin #4744: MAX31855 Breakout Board throws Mintemp error

derletztename, 2016-08-31:
I am using this Breakout Board: [Thermocouple Amplifier MAX31855 breakout board (MAX6675 upgrade) - v2.0](https://www.adafruit.com/index.php?main_page=product_info&products_id=269&zenid=516505f41182f3a344fea2f8955968e0). I am not using any LCD Display or SD Card. And I am using the newest Marlin RC 7.
I connected everything by following [this tutorial](http://www.instructables.com/id/How-to-use-MAX6675-Thermocouple-Amplifier-with-RAM/). 
Nevertheless I only get Mintemp errors or nothing happens at all (0 °C). I bought a second Breakout Board, a new Arduino and a new USB-Cable. It's driving me crazy, and I really need the MAX31855 breakout board for my project. I tried the [External thermocouple board v1.0](https://reprapworld.com/?products_details&products_id/351/cPath/1616_1652) and everything was easy and fine, but for this project I need the MAX31855 breakout board. Maybe I did forget something else to change in the code?
It would be so great if someone could help me out

Grogyan, 2016-08-31:
CS pin is on pin 66.
This ensures that there is no incompatibility with an SD card, if you so desire to use one at a later stage.

The pins_RAMPS.h file reflects this

derletztename, 2016-09-01:
Thank you for the answer. I tried pin 66, pin 53 and pin 49. Neither of those worked. I changed the pins in pins_RAMPS_1.4.h.
Is there some hint how to localise the error, maybe?

Blue-Marlin, 2016-09-01:
Try RCBugFix.
When without SDSUPPORT there was a error in the initialization. (#4564)

derletztename, 2016-09-01:
Awesome. Thank you so much. I'm going to try this tonight and post if it helped

thinkyhead, 2016-10-17:
Does #5033 do the trick then?

derletztename, 2016-10-19:
Yes, up to 1000°C it's working :)
Above that temperature I hadn't had the chance to try yet

closed 2016-10-22 by thinkyhead