MarlinFirmware/Marlin #4745: Problem: "corxy and ENDSTOP " in rc7

genzo76, 2016-08-31:
I was using the RC6 is I reported the configuration RC7. But I have a problem with the z and x axis, while the y-axis working properly.

It seems a problem because endstop autohome the axes do not move.

This is my configuration:

> #define X_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
> #define Y_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
> #define Z_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
> #define X_MAX_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
> #define Y_MAX_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
> #define Z_MAX_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
> #define Z_MIN_PROBE_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.

I tried to change the endsto of the two axes that do not work but the X and Y axis motors do not move in sync and the printer stalls favendo the classic sound of the engines losing steps, as if he had lost the corexy mode.

You have no idea why I maybe I missed something in the new configuration file, or you may be a bug in the new version?

Ps.

Congratulations to all for the work.

kmbecker13, 2016-09-01:
Could you clarify which axis is having issues?  You mentioned Z and X first, but then say that X and Y do not move in sync later on.  I'm assuming you mean the X and Y axis since those are very similar for a CoreXY printer.

Are both motor plugs connected with the same orientation?  Is one of the axis inverted in the firmware?

thinkyhead, 2016-09-02:
@genzo76 Have you already tested the endstops with `M119` to make sure they are triggered properly?

The issue of Endstops and CoreXY has come up before, but we addressed the problem and it seemed to be all well and good. However, it does depend on the motor directions being set in certain ways, so maybe the assumptions in the code are incorrect. For example, if you are using `QUICK_HOME` that might confuse the logic. Try disabling that option if you're using it and see if it helps.

alphapilot380, 2016-09-05:
I have the same problem on a CoreXY setup, I upgraded from RC2(fully working) to the latest RCBugFix.
The M119 command only shows the endstops as "open",  I cannot get them triggered by pressing the switches, only the Z-probe switch works as it should. I only have X_min, Y_min switches and a Z-probe(connected to Z_min) on my printer. Autolevel is disabled. The board is ramps 13 EFB, but this hasn't changed since RC2.

I don't know if this is related but I also cannot home, when I issue the G28 command I get "Home X and Y first" while this was done first automatically in RC2. I can home Z when I issue G28 Z but after this I still cannot move the X and Y axis nor will they home when I issue the G28 command.

alphapilot380, 2016-09-05:
I had a similar issue which was just my own fault. On my printer I'm using the X_max, Y_max and Z_min(=probe) endstops. In RC2 I had to invert the Min_endstop of X and Y in order for this configuration to work, I copied this over to the RC7 but forgot that I was using the max endstops so the USE_XMAX_PLUG was not defined while the USE_XMIN_PLUG was still defined while X_HOME_DIR = 1(max), same for Y axis. My printer didn't want to home and the M119 always showed "open" of course.

Maybe there should be a check that when X(YZ)_HOME_DIR = 1 the USE_X(YZ)MAX_PLUG is also defined?

thinkyhead, 2016-09-05:
> …when `X(YZ)_HOME_DIR` = 1 the `USE_X(YZ)MAX_PLUG` is also defined…

Makes sense. I can't think of any common exceptions.

genzo76, 2016-09-07:
I managed to solve, I re-downloaded the project and reconfigured throughout the printer from scratch, now works.

I continue to have strong problems with the hot bed level car, I think you understand the problem, you open a topic explaining good behavior.

thinkyhead, 2016-09-07:
@genzo76 Yes, bed leveling definitely needs a proper patch. I've been asking people to test this branch and report whether it works any better:

https://github.com/thinkyhead/Marlin/tree/rc_fix_leveling_maths

Roxy-3D, 2016-09-07:
> I've been asking people to test this branch and report whether it works any better:

I tried it just yesterday and ran into some compile time problems.  (I got past them, but with Grid Based auto leveling enabled there is an out of scope error on the bed level correction matrix in Marlin_main.cpp)

thinkyhead, 2016-09-07:
Sorry about the compile issues. I added commits to the wrong branch. I have removed them.

chorca, 2016-09-13:
Having what may be a similar issue here.

I've got a CoreXY machine with Dual Z stepper drivers, and Dual Z endstops. The second endstop is hooked to XMAX as suggested in the configuration_adv.h and I've enabled all the PLUGs in configuration.h. When attempting to home, the Z motors will make a quick noise, and then the machine seems to believe it's homed. and allows me to move the axis. However, any repeat of the z home gives the same very quick noise (almost like a microstep forward and back suddenly) and then it considers itself homed, wherever it is. I've found by triggering one of the endstops, that motor will not make a quick noise when pressing home, so it seems to just be jumping through the home routine instantly and setting itself to 'home' without moving all the way to test the switches properly.

adamfilip, 2016-09-13:
running into a CoreXY Endstop issue aswell.
homing to Y Max

[https://github.com/MarlinFirmware/Marlin/issues/4790](url)

boelle, 2016-11-02:
@genzo76 @Roxy-3D @adamfilip 

is the issue still present?

thinkyhead, 2016-11-03:
I believe this may be fixed, if it was related to a recent issue where triggering the probe also set the Y endstop bit. @genzo76 Please double-check with `RCBugFix` when you have a chance. Thanks!

boelle, 2016-11-11:
@genzo76 still with us?

@adamfilip have you tried the latest RCBugfix?

boelle, 2016-11-19:
@adamfilip did you try the latest RCBugfix or an older version?

boelle, 2016-11-26:
@thinkyhead 

since neither @genzo76 @Roxy-3D @adamfilip have replied over the last 24 days can we safely assume that either the problem is fixed or they have lost interest?

closed 2016-11-26 by thinkyhead