MarlinFirmware/Marlin #4751: Stop print not working after pause print in RC7

condac, 2016-09-02:
RC7 commit d97dc102
When pressing pause print in the LCD menu, the printer pauses. But when trying to press stop print when in pause it doesn't stop and show the print from sd option again. 

Also pressing stop during print is useless because it often runs away with the steppers and move x y outside it limits. I always press pause before stop to avoid this, and it looses where on the x and y it is. ( this runaway and loosing where it is have been issue since 3 years back...)

configuration: include reprapdiscount lcd and include sd support, ramps1.4 and changed steps per mm everything else default.

jackshi618, 2016-09-02:
Pause when you can remove the print head? Or a custom implementation of a custom gcode

thinkyhead, 2016-09-02:
Try `RCBugFix`. This issue should be fixed with #4610

closed 2017-10-31 by thinkyhead