MarlinFirmware/Marlin #4752: Firmware retract not working with mixer setup

sangthip, 2016-09-02:
Firmware retraction problem exists if using mixing setup  with 3 extruders.
problem if only using 2 extruders all 3 extruders retract but the one not being used will not recover the retracted amount where as the other two do.
This leads to the unused extruder being left with no filament and the other two forcing their filaments back up the unused side.

This causes under extrusion and massive clog ups thet need the hot end be disassembled to unblock them if it has been happening for some time.

I hope this explains the problem

thinkyhead, 2016-09-02:
That makes sense. So basically it sounds like I forgot to handle retract/recover properly in the implementation. I will look at this soon and patch it up asap.

sangthip, 2016-09-02:
I am using the latest Bugfix;  Please explain what LIN_ADVANCE does I have seen the option but have no idea how it works of if it will help so it has not been enabled.

thinkyhead, 2016-09-02:
LIN_ADVANCE decouples the E axis so that it is not linear-interpolated along with the XYZ axes, but instead operates to produce the right amount of material pressure at the proper instants. This prevents ooze at the end of lines and allows better extrusion at the start of lines.

sangthip, 2016-09-02:
Sounds usefull Ill enable , recompile test and get back to you

sangthip, 2016-09-02:
Back so soon... not good news
Enabling LIN_ADVANCE throws error on compile 
Arduino: 1.6.9 (Mac OS X), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

/var/folders/y2/jx1h1b7x11vf3nqcq422ds600000gn/T/build38ee351eb3f96c2b2cb8074b479b6407.tmp/sketch/stepper.cpp: In static member function 'static void Stepper::isr()':
stepper.cpp:397: error: 'counter_m' was not declared in this scope
             counter_m[j] += current_block->steps[E_AXIS];
             ^
Using library U8glib at version 1.19.1 in folder: /Volumes/Raptor/Documents/Arduino/libraries/U8glib 
exit status 1
'counter_m' was not declared in this scope

thinkyhead, 2016-09-02:
Hmm. Somehow half of the `counter_m` are called `counter_M`. Patching!

sangthip, 2016-09-06:
Immmm Back
This is weird.  enable LIN_ADVANCE or ADVANCE with FWRETRACT leads to nothing being extruded ??
Turn them off  FWRETRACT still on still works the same as when this was started (mixing extruder using 2 all 3 retract only 2 recover.
Turning off FWRETRACT with LIN_ADVANCE or ADVANCE also off, it prints 1 2 or 3 extrudes as mixing without problem

thinkyhead, 2016-09-07:
Sounds like it will be fun to figure out and fix. I hope I can find the time this week to look into it.

boelle, 2016-11-02:
@sangthip is the issue still present in the latest RCBugfix ?

thinkyhead, 2016-11-03:
I believe this is fixed now, but will await confirmation from @sangthip.

boelle, 2016-11-11:
@sangthip still with us ?

boelle, 2016-11-19:
@sangthip ??

boelle, 2016-11-26:
@thinkyhead 

since  @sangthip  have replied over the last 24 days can we safely assume that either the problem is fixed or he/she have lost interest?

closed 2016-11-26 by thinkyhead