MarlinFirmware/Marlin #4753: lround function used to calculate position might create wooble

Nutz95, 2016-09-02:
Hello, 
I have Z wooble issues and i believe that this beahvior comes from the firmware.

I'm using RC6 and here is my config:
lead screw: 1.5mm lead
stepper: 16 microstep (4988)
layer height : 0.21 or 0.12
Zsteps/mm: 2133.33

I have wooble on my printer when using this leadscrew and i have not when using standard M8 leadscrew.
I've been tweaking my printer to make sure my z axis was moving perfectly straight by placing a laser on my z axis and monitoring the projection of the laser spot on my bed when moving up.
The conclusion is that when i move up the laser spot stay on the same location. => no oscillation or similar behavior.

This said, i can confirm using prusa calculator that when i use a 1.5mm leadscrew i need to use 2133.33mm as Zsteps/mm.
And when i double check the compatible layer heights i find 0.12mm and 0.21mm being compatible, meaning that they need an integer amounts of steps to perform this layer height.

But when i look at the code we are always using  axis_steps_per_unit[Z_AXIS] inside a lround() function.

for example inside planner.cpp : `target[Z_AXIS] = lround(z * axis_steps_per_unit[Z_AXIS])`; (around line 582)

the lround function is not well documented and from the wooble i can notice i understand that it's doing a lower rounding on the value inside the ().

and to me if z is 0.21 and axis_steps_per_unit[Z_AXIS] = 2133.33 i get 
0.21 * 2133.33 = 447.9993
when rounding down i have 447.

Again from calculation (we can confirm using prusa calculator) the proper amount of steps i need to perform is 448 not 447.

200 steps/turn * 16microsteps * 0.21mm(layer height) / 1.5mm (screw lead) = 448

And i believe the function used in planner.cpp needs to be rounded to the closest integer not to the lower one.

Does this makes sens to you people? or Am i wrong somewhere ?

brainscan, 2016-09-02:
The only way I've found to eliminate the z wobble is to use layer heights that divide nicely into the lead of my z screws. For 1.5mm lead I'd use 0.3, 0.15 or 0.075 or for M8 0.25, 0.125, 0.0625 otherwise I always have wobble. Sorry if this is of no help to you.

thinkyhead, 2016-09-02:
`lround` is actually "round to nearest and cast to long integer." So this actually rounds up if the decimal part is over 0.5 or higher.

thinkyhead, 2016-09-02:
> use layer heights that divide nicely into the lead

Also make sure the coupler is of the flexible variety and don't constrain the lead-screw. Let it hang as loose as possible, otherwise it will divert the axis.

emartinez167, 2016-09-02:
Wild thought occurred (out of ignorance). Could this rounding/Z wobbling be
inadvertently causing issues with bed levelling?

Nutz95, 2016-09-02:
Thanks you both, 
- I'll try what you say Brainscan, i've never really tried theses layer heights with my 50mm tower.
- Thanks thinkyhead for clarifyng the lround!
if it's really doing this then i guess it's fine on this part.

On the Coupler and coupler side i've tried everything, and i could'nt notice any difference.
i've even tried
- to let everything free
- to contrain the bottom
- to contrain the top
- to contrain bottom and top
- to push the lead screw all the way down against the motor shaft inside the coupler
- to let some space from the motor shaft and the leadscrew inside the coupler

I spent a few weeks testing everything, mounting/unmounting my printer... making only 1 change at a time.

About bed leveling i haven't tried to disable it to check the difference.
But i guess it could amplify the issue as the Z axis is moving both upward and downward during the print.

thinkyhead, 2016-09-02:
> inadvertently causing issues with bed levelling

No. The cause of that is well known.

thinkyhead, 2016-09-02:
> everytime i get wooble

Please post some pictures of the banding on your prints. This will give some clue as to the nature of your wobble or irregular layer heights.

Nutz95, 2016-09-02:
sure i'll post some this coming week end

Blue-Marlin, 2016-09-02:
> 2133.33mm as Zsteps/mm

The max error is one step -> 1/2133.33mm -> forget it.

Blue-Marlin, 2016-09-02:
If you want to use DISABLE_Z it is advisable to use a laeyer height with a 'even' full step.

Nutz95, 2016-09-02:
> The max error is one step -> 1/2133.33mm -> forget it.

I don't say the total height of my plastic part is not correct.
I says that each of my layers don't have the same height, some are smaller, and some are higher, but the overall height is good.

A 50mm tall rectangular tower is 50mm high

thinkyhead, 2016-09-02:
All kinds of things can affect this. For example, a direct-drive extruder can get pulled upward by the tightly-curled filament near the end of a spool. Periodic thermal expansion of the heated bed can produce uneven layer heights. And as mentioned, if you use layer heights that don't comport with your whole steps, and/or if you allow the Z steppers to sleep between layers…. Many possibilities. Is this a "Prusa i3" type model with moving bed?

Nutz95, 2016-09-02:
Yes it's a prusa i3 like printer.
I print on a cold bed so no expansion issues on this part
I've tried with 0.15 and 0.3 so far and i still get wooble.
But i'm doing a more complete and documented set of tests i hope i can bring you some nice pictures by the end of the week end.

I also want to take some time to thanks all of you for 
- your support
- the great job you are doing, guys you are all amazing.

Nutz95, 2016-09-08:
I couldn't make the photo yet but i could solve my wooble issues.

I changed 1 thing at a time and i could see that what Brainscan proposed had some improvement on the wooble. 
It was not the root cause but it definitely helped.

The root cause was a slight mechanical oscillation of my nut holder on my Z axis.
So i used 0.3 or 0.15mm layer height + increased the rigidity on this holder  => no wooble anymore.

I could also test the impact of using 2133.33 and also 2134 as ZStep/mm and the lrounding seems to work properly as it had no impact on the final print.

I'll close this git issue and thanks again for your great advices.

closed 2016-09-08 by Nutz95

thinkyhead, 2016-09-08:
On Prusa style printers, one end of the z rods must be let free. Constraining the rods only transfers wobble in the rods to the carriage.