MarlinFirmware/Marlin #4755: Hotend gets cold and bed not on function lcd_sdcard_stop

bikkesbakker, 2016-09-02:
Previous version did not start to cooldown the hotend on lcd_sdcard_stop.

I try a lot of things to prevent cooldown without success

    static void lcd_sdcard_stop() {
      card.stopSDPrint();
      //clear_command_queue();
      quickstop_stepper();
      print_job_timer.stop();
      //thermalManager.autotempShutdown();
      //wait_for_heatup = false;
      lcd_setstatus(MSG_PRINT_ABORTED, true);
    }

need some help-

thinkyhead, 2016-09-03:
> Previous version did not start to cooldown the hotend on lcd_sdcard_stop.

Apparently it did. Here's the code from 1.0.2-1…
```cpp
static void lcd_sdcard_stop() {
  card.sdprinting = false;
  card.closefile();
  quickStop();
  if (SD_FINISHED_STEPPERRELEASE)
    enquecommand_P(PSTR(SD_FINISHED_RELEASECOMMAND));
  autotempShutdown();
  cancel_heatup = true;
  lcd_setstatus(MSG_PRINT_ABORTED);
}
```
> I try a lot of things to prevent cooldown

I can't see anyplace else where the heaters are shut off after `lcd_sdcard_stop` in the current code.

thinkyhead, 2016-09-03:
I can't find any code that would shut off the heaters once they're at temperature. The `wait_for_heatup` flag would only stop waiting for a heating in-progress.

bikkesbakker, 2016-09-03:
maybe adding a code for selection?, so anyone can change if the bed/hotend gets cold or not after stoping a print progress .
In my case it would be great if hotend and bed stay the same after i cancel a job.

thinkyhead, 2016-09-04:
Yes, it would have to be optional, because canceling an SD print, in my opinion, should definitely turn off the heaters. However, I have not been able to find the code that specifically shuts off the heaters in response to `lcd_sdcard_stop`. Maybe you will have better luck than me.

I did find one case where the hotend heater would be turned off:

When `AUTOTEMP` is enabled and turned on (check your GCode for `M109 F...`) and the current temperature is higher than the minimum temperature setting (look for `M109 S... F...`), then stopping the print will shut off the hot end with `thermalManager.autotempShutdown()`. That doesn't shut off the bed, however.

Q: Are you using `RCBugFix`? That's the code I'm looking at.

closed 2017-10-31 by thinkyhead