MarlinFirmware/Marlin #4759: UBL Config

OrbitAVI, 2016-09-04:
@Roxy-3D Is it possible for you to post the default config again? The one included now defaults to your config which I cannot use and the probe section is a little different than the default. I cannot find were to disable the servo probe that you use.

OrbitAVI, 2016-09-04:
Never mind I forgot you moved it to the end, but it would be nice to have the "un-configured" config file.

Roxy-3D, 2016-09-04:
The Servo stuff can be where ever is most clear.   Where do you want that positioned in the Configuration.h file?   Historically, the Servo stuff has always been at the end of the Configuration.h file.  But because the Servo stuff is only used if the system has a Z-Probe, I think it feels more logical to move it close to the other Z-Probe stuff.

If you have an opinion where items should be placed, now is a good time to make a suggestion!

OrbitAVI, 2016-09-07:
I have no pref, but would suggest that all z probe options stay together. I was only suggesting that you keep the default config "default" its easier, at least for me, to set up from default. The servo stuff in your config is moved to the end where as in the default config it is with the z probe options.

Roxy-3D, 2016-09-07:
Yes.  I understand!  And in fact, there is some extra clutter you didn't mention.   Some of the #define's for the old Mesh leveling and Grid based stuff are still in the Configuration.h file.   That needs to be cleaned up also.

But I agree with the thinking that all of the Z-Probe stuff should be kept together.   I will try to migrate everything to one spot in the Configuration.h file.

OrbitAVI, 2016-09-30:
@Roxy-3D I just downloaded the most recent as I'm finally now going to have some time to try it out the previous release you had has been working so well once I get the z offset figured out that I haven't wanted to mess with it too much. 

I downloaded your latest which again defaults to your configuration, so I would ask again that you leave the default configuration "vanilla". After loading it I realized you had reversed the encoder direction in yours and also you added some values to your safe z homing. These are minor things, but when I run into issues I'm not sure if its a Roxy config thing that I may have missed or my own. Does this make sense?

Again this is just my opinion and maybe others testing this prefer to see your config as the default, but my opinion is that it may cause some confusion. With that said I am pretty new to all of this. 

Thanks again for all your hard work. The auto mesh is amazing!

Roxy-3D, 2016-09-30:
I do understand your point.   The reason I've been against that is I believe the closer a person is to my configuration, the more likely they won't have any problems.   So, by starting with my configuration.h file and making necessary changes, it is more likely the files will be as similar as possible.

But I guess we can start considering doing that change.   As we get more people using the system, and less of them are reporting issues, it makes sense to shift to what you are suggestion.

closed 2017-10-31 by thinkyhead