MarlinFirmware/Marlin #4760: Mesh bed leveling pushes my Z endstop down

Tyler-Darby, 2016-09-04:
I have a switch based standard Z endstop. I set the Z endstop to home at about ~1mm above what appears to be the highest corner of my bed.

 From there I send a G29 S1 command to start the procedure. It moves to the first point, I begin lowering the hot end to the proper position.

Here's where the problem is: Before I can lower the hotend enough, my Z endstop actually gets pushed down by my x carriage, which I would assume throws the whole system out of whack since my Z endstop is now actually lower than it was before. 

How do I resolve this? I can't lower the bed anymore because it moves the endstop, and I can't lower the endstop beforehand because the hotend would drag on my bed at the highest point.

epatel, 2016-09-04:
It sounds like your endstop does not have an arm, i.e. like this [image]

The method of setup and the docs I wrote was for either using an arm-based switch as above or an opto endstops (like these http://www.ebay.com/itm/Opto-Endstop-Switch-Kit-for-CNC-3D-Printer-RepRap-Makerbot-Prusa-Mendel-RAMPS-/221977263509).

My concern to have the setup like that was to minimise the risk of hitting the hotend on the bed unintentionally.

But, recently changes have been added so the setup will lift the hotend and can maybe accommodate endstops without and arm. Well, I just now tried to find those options but not sure where they are anymore (I did not write those parts and I am not using them myself). Will look in another place.

epatel, 2016-09-04:
Oh, the *new* thing was called `Z_RAISE_BEFORE_HOMING `. But that changed to `Z_HOMING_HEIGHT` ...you can try using those. Good luck! I'd suggest changing to opto endstops.

closed 2017-10-31 by thinkyhead