MarlinFirmware/Marlin #4761: UBL only scans 6 points on the bed as opposed to 16

printerguykw, 2016-09-05:
So looking at the main readme, I sent G29 P1 M, however, only 6 points on the bed gets scanned.

In my firmware I have set the following:

    #define MESH_MIN_X X_MIN_POS
    #define MESH_MAX_X X_MAX_POS
    #define MESH_MIN_Y Y_MIN_POS
    #define MESH_MAX_Y Y_MAX_POS
    #define MESH_NUM_X_POINTS 4  // Don't use more than 15 points per axis
    #define MESH_NUM_Y_POINTS 4
    #define MESH_HOME_SEARCH_Z 4

As per the screenshot: http://imgur.com/MUdoY3T It is clearly visible that there are 16 points that need to be populated, however only 6 are set. 

Anybody have a clue?

Thanks

apballard, 2016-09-05:
I also experienced the same.  I believe what you are seeing is related to other settings that describe the position of the probe in relation to the nozzle.

With those settings set, it's possible that the probe cannot reach those points.  You would need to go through the different phases of the G29 to populate those values.

In my case I had to manipulate the settings, mesh_min/max_x/y values.  However if end stops are enabled, you also won't be able to reach those areas.

Just some thoughts, based on my experience. Even after tinkering, I cannot get the first two rows across my y access measured because my ir sensor sits behind my nozzle and I have a physical limit switch preventing my sensor reaching the front of my build plate.

Can't remember if the settings further down (LEFT_PROBE_BED_POSITION....) impacts this as well. I also had to play around with those numbers. From the screenshot I see that what seems to be your y axis is limited to over 20 so not sure if that could also be a problem.

Roxy-3D, 2016-09-05:
If you do a G29 M you will see the Map of the Mesh Points.   The .'s are the un-reachable areas of the bed.  Those are Mesh Points that can not be probed because of the offset the probe has to the nozzle.

Those points can either be manually probed with a G29 P2 R or they can just be filled in with a G29 P3 R C 0.0      You can also do a combination and just probe a few of the points by moving the nozzle to a location and doing a G29 P2     (with no R).    Then when you do a G29 M you will know the right value to fill each area.

The G26 C P O2.0 will tell you how much to raise or lower a Mesh Point when you do a G29 P4 R

closed 2017-10-31 by thinkyhead