MarlinFirmware/Marlin #4762: not an issue dont know where to put this type of ?

locutus40, 2016-09-05:
i was wondering if it would be possible to add a pause feature.. meaning.. if we want to pause it then resume like say i want to change out filament to a different color lift the nozzle and then purge the nozzle 
then hit resume and it goes right back to printing...and when i say pause i mean from the lcd.. add something that says pause and resume.. in the menu

or how bout a feature that...you can add custom logos and or animations on the lcd or music it has a small speaker something like i wanna ad Mario or something or something stupid like that..

its sooo hard to do now... i cant even figure it out...
so it would be nice to see it simple like drag and drop files etc make it simple to do...

we have these cool big lcds that show hardly anything cool im sure they can animate stuff etc...

or an app to interface the printer and print from a phone or your network..

i have more ideas lol but i bet theres a better place to put them

Blue-Marlin, 2016-09-05:
Please have a look at FILAMENT_CHANGE_FEATURE (M600) in Configuration_adv.h

What we do not have is memory and time. Forget about animations, music, ethernet, brain interface, printing from 2 fotos, ... - as long we are using ATmega processors.

thinkyhead, 2016-09-05:
Custom boot screens have recently gained support…
```cpp
#if ENABLED(SHOW_BOOTSCREEN) && ENABLED(SHOW_CUSTOM_BOOTSCREEN)
  #include "_Bootscreen.h"
#endif
```

locutus40, 2016-09-06:
wow wasnt expecting this much response but thanks guys...  where do i add that buzz sound ? is it mario lol?  

yea custom boot screens are enabled but its so darn hard to figure out how to do it that you hardly see anyone do it its like rare.... at least im a noob at this stuff ive had my prusa i3 for 2 years now learned alot off you guys thanks for all the input and years and effort into this stuff.. without you wed have no printers cheaply....
anyway thinkyhead ok say i want just a transformers logo as a boot screen.... how do i actually do that?

theres no tutorials or anything i can find... tried google theres one but its so freaking complicated...that i feel like a 1 year old trying to learn secrets of the universe..

locutus40, 2016-09-06:
oh and the fan animation i didnt know it was that horrible on purpose..  but then again it is a small processor on their.. well if it can animate 1 frame at a time.. you still can do like a character say walking one frame at a time do a stick figure thing.. i dunno im just thinking outside the box here ... you know those old shorts they used to do back before tv where oyu looked into a device and paper spun around and it animated frame by frame.. we used to do it in school draw on paper note books flip the book and you got animation... maybe a pacman eating white thingies thats dont take much 1 frame at a time i dunno lol

Roxy-3D, 2016-09-06:
> oh and the fan animation i didnt know it was that horrible on purpose.. but then again it is a small processor on the(re).. well if it can animate 1 frame at a time.. you still can do like a character say walking one frame at a time do a stick figure thing...

I know it is stupid...  But I would like the Marlin Screen at boot up to have a Marlin fish jumping out of the water.     Can you make that happen?

thinkyhead, 2017-10-31:
> oh and the fan animation i didnt know it was that horrible on purpose

All it needs is a nice 3-frame version that looks more spinny.

thinkyhead, 2017-10-31:
Side note: I fixed the broken image converter on the Marlin website and made it a little better as well. Give it a try at http://marlinfw.org/tools/u8glib/converter.html

closed 2017-10-31 by thinkyhead

forum999, 2019-01-30:
so i notice there is no real answer here? is ther ea way to make an animated boot screen?