MarlinFirmware/Marlin #4763: M48 Z probe repeatability no longer works since RC7

zborschuk, 2016-09-05:
Maybe I'm doing something wrong, but, since RC7, (currently on bugfix branch)  M48, the z probe repeatability command no longer works. I've tried prefacing it with G28 and by itself to no avail.  Can anyone else confirm? I'm running a modified Prusa i3 with dual extrusion and can provide my config file if necessary, but I feel like that isn't at issue here.  Thanks!

Roxy-3D, 2016-09-05:
I've been using M48 within RC-7 for the UBL work.   It seems to work fine there.   Can you tell us why you don't think it is working?

zborschuk, 2016-09-05:
I've tired running it from the command line numerous times in different variations and it never does anything.  just says command received and sits there.   I've tried just plain M48 X100 Y100, M48 P10 X100 Y100 E, M48 P10 X100 Y100 V2 E L2 and numerous other combos within all with the same result, no action by the printer.

Roxy-3D, 2016-09-05:
That sounds like you did not turn it on in the Configuration.h file.

But you can prove that by giving it the command:   M48 V99 X9999 Y9999 P999

One of those parameter will trigger an error message if it is built into the firmware.

zborschuk, 2016-09-05:
That didn't trigger an error. And I have DEBUG_LEVELING_FEATURE uncommented.  Also ran M111 S32 before trying G28 followed by an M48 command to no avail.  Am I looking in the wrong place?

Roxy-3D, 2016-09-05:
> That didn't trigger an error.

The start of the M48 code looks like this:
```cpp
 inline void gcode_M48() {

    if (axis_unhomed_error(true, true, true)) return;

    int8_t verbose_level = code_seen('V') ? code_value_byte() : 1;
    if (verbose_level < 0 || verbose_level > 4) {
      SERIAL_PROTOCOLLNPGM("?Verbose Level not plausible (0-4).");
      return;
    }

    if (verbose_level > 0)
      SERIAL_PROTOCOLLNPGM("M48 Z-Probe Repeatability test");

    int8_t n_samples = code_seen('P') ? code_value_byte() : 10;
    if (n_samples < 4 || n_samples > 50) {
      SERIAL_PROTOCOLLNPGM("?Sample size not plausible (4-50).");
      return;
    }
```
It would have said:  ?Verbose Level not plausible (0-4).   if it was built into the firmware.   I think we can concentrate on the problem of getting it built into the firmware.   You need to have a Z-Probe configured (properly) and this line of code enabled:
```cppp
#define Z_MIN_PROBE_REPEATABILITY_TEST
```

zborschuk, 2016-09-05:
Yeah. That did it. I need to learn to read more thoroughly.  Thanks!

closed 2016-09-05 by zborschuk