MarlinFirmware/Marlin #4764: Question: Why UNIFIED_BED_LEVELING_GRID when using UBL ?

lrpirlet, 2016-09-05:
@Roxy-3D I am (slowly, when I get time) setting your implementation ofthe UBL in order to replace my auto mesh leveling...

In my view, the UBL will set the Z axis perpendicular (more or less) to the XY plane... Then the mesh leveling will take care of the variable bed to hot tip distance function of the location over the plane...

Why do we need a grid? 
Mathematicaly I can unserstand the need of a grid to integrate several slightly out of plane points... 

But as soon as we follow this probing with a probe for a mesh, I totaly miss the need. The mesh is able to follow the bed, including when the bed is nearly perpendicular to the Z axis (but not quite)...

To use a 3 dot probing seems to me quite enough  ( and makes the first tilt calculation a lost faster and mush shorter in memory... ( I may be wrong)

Roxy-3D, 2016-09-05:
The Grid feature isn't implemented yet.  But the main reason would be to have a more accurate way to 'Tilt' the Mesh, especially when being used on Delta printers.  (This hasn't been moved to Delta's yet either).

lrpirlet, 2016-09-08:
@Roxy-3D thanks... I have a issue compiling your cloned branch... I will close this and open a new issue

closed 2016-09-08 by lrpirlet