MarlinFirmware/Marlin #4766: QUESTION: Inconsistent extrusion (not really marlin associated)

Kaibob2, 2016-09-06:
I'm seeing an issue when printing more complicated objects like in the pictures below. I tried almost every combination of retract, startpoint, coast, wipe, speed, temperature and so on. The printer is really well calibrated and sqaures or circles come out perfect. Also objects like [Moai](http://www.thingiverse.com/thing:149271) or [Gecko](http://www.thingiverse.com/thing:1363148) come out with a very smooth surface. 
Like you can see in pictures the diagonal front is perfect too. On the backside there are stripes which align with the holes. The object is printed with fixed startpoints on the left back. The area between the holes is fine, but when the layerheight reaches the first hole the path goes "inside" the object and when it comes back to the surface the extrusion is a litlle lower than at heights where there is no hole. It is a little hard to describe, but maybe somenone can give me a hint how to get rid of this. This is the last thing i just can't get fixed, i'm fidlling around with this problem for months now and asking here seems like my last resort.
 
Sliced with S3D
Printed with PLA at (calibrated) 210°C
Bed 50°C
60mm/s inner perimeter
24mm/s outer perimeter
2 Outline shells
0,2mm layer height
0,4mm nozzle

[image]
[image]
[image]
[image]
[image]
[image]

boelle, 2016-09-07:
I have this issue too...

i use cura for slicing and to print over usb

I'm going to rebuild my printer at some point... allready printed new parts for my mendel90.

i "think" the problem is the Z screws... i use the cheap threaded rods and they do take some wear... thinking of replaceing them with trapez rods instead

Kaibob2, 2016-09-07:
@boelle I can confirm that this issue has nothing to do with the Z-screws. I had wobbly surfaces from the beginning, but i eliminated it completely finally. First i had TR8x1.5 trapez rods, then M10x1.5 threaded rods, now TR10x2 trapez rods. The wobble stayed the same according to the rod pitch.I think it's needless to say that i added anti wobble Nut Traps etc. quite early. To avoid the wear i connected two trapez nuts via 3 screws with a spring on each of them. This makes an adjustable nut and the wear is =0. Both rods had a bearing on the top end and an unflexible aluminium coupling on the stepper side. The problem in the end was, me, coming from mechanical engineering, that everything was fitted too tight. Long story short. I removed the top bearings, the adjustable nut and fixed the z-steppers loose with rubber washers. The connection to the X-slide is done like this:
[image]
Btw., the printer is completely own designed, but prusa inspired.
[image]
So, no more wobble on the surface.

The issue i'm describing here is something different. The problem is that the extrusion width gets thinner and thinner the further it is travelled from the startpoint. If the startpoints are fixed to X0Y0 for example it happens too, but every layer is the same so you don't see it.
If the startpoints are randomized for example, the first layer startpoint is at X0Y0 and the second layer startpoint is at X30Y0 then the "thick" part of the layer is first on the left front and then for the second layer on the right front. Assuming we're printing an 30x30 cube.
For cubes etc. this is easy to workaround with fixed startpoints, but not for complex models.

Just a thought: I guess the acceleration of the printhead is not taken into account for the extruder speed. So, if i set a final speed of 60mm/s the extrusion rate is calculated for this speed, correct?
If the acceleration would be very low, let's say 50mm/s², wouldn't it overextrude until it reaches top speed?

Sorry for the long post :)

thinkyhead, 2016-09-07:
Although it requires (yet more) tuning, the advance extrusion options are intended to prevent issues like these. The `LIN_ADVANCE` option is newer and known to work in most circumstances. The older `ADVANCE` option has been patched up, but has not been well-tested.

killpop, 2016-09-20:
I have also run into this issue and have found two ways to solve it. Depending on the slicer that you use, you can set the tool path to start on "the inside" or "the outside" of the loop. Some slicers have this but call it something else and some don't have it all (I use Simplify3D). My other method is to design the part so you can fillet the hole. This does two things for you, it widens the arc path the tool travels so it doesn't have to slow down as much and it helps to blend it with the rest of the part. In general, I fillet every hard edge on any model that gets printed. Your printer will thank you if you do the same.

Kaibob2, 2016-09-20:
@killpop: I use S3D too. Do you mean "Outline Direction" Setting?  What exactly do you mean with filleting the hole? Can you perhaps send a picture?

killpop, 2016-09-21:
A fillet is simply the rounding of an edge. I printed the model shown as a demonstration and it clearly shows where the problem with the hard edges but the rounded edges are perfect. Unfortunately I can't find my good camera and my phone wont take a good enough picture to show you the difference. I'll update this when I find my camera. 

[image]

Kaibob2, 2016-10-07:
@killpop Starting the loop on the outside makes it difficult when the objects surface is not straight up as the outer loop will print in the air. I tried your suggestion with filleting the holes but this changed nothing. This issue is not really a problem for me, because design parts usually don't have these kinds of holes and for constructional items it is not that important how they look like. I think LIN_ADVANCE will be a good start, but the E Stepper in my bowden setup has problems keeping up...

closed 2016-10-21 by Kaibob2

macst34, 2016-12-08:
Hi @Kaibob2 

That's a pretty nice design for a printer. Have you opened up the design for sharing with other people?

Impressive prints by the way....The elongated holes on your  X axis ends...do they still appear if the piece is printed rotated 90 or 45 degrees? What it looks like to me is that the line of hot filament that the nozzle puts down gets pulled into the piece at the corner of the hole and then bunches up slightly when coming back out the other corner.

for example in this picture...
[image]

For the bottom and top left holes it seems the nozzle approached from the left, pulled a little of the hot, just laid filament into the piece and then pulled some out when it rounded the corner on the other side.  The hole on the right seems like the nozzle was coming from the right. If you load your gcode up, does it confirm what I am guessing? If so...does it just mean it is rounding the 90 degree corner too fast? 
Will tweaking jerk make this less noticeable? Or maybe a fan at 100% just at the corners help?

Anyways....your custom printer is really interesting. Are the weights in the X-axis end body to increase heft and minimize resonance? Is the offset hole in the body for tightening the x axis tension pulley?

take care,

Michael

Kaibob2, 2016-12-09:
@macst34 

> That's a pretty nice design for a printer. 

Thanks, meanwhile it's in Version 10 but produces quite good results
[image]

> Have you opened up the design for sharing with other people?

Not really, the problem is not the opening, it's the support. With the drawing on it's own no one will be able to build the printer. There are quite a lot of parts which require additional tools (lathe, mill, etc.)
But if you want i can upload it somewhere for you.

> The elongated holes on your X axis ends...do they still appear if the piece is printed rotated 90 or 45 degrees?

Yes, the elongation itself is not really a problem. The parts are printed with a layerheight of 0,2mm. With a layerheight of 0,05mm holes look more round than eliptical. The problem i adressed here i this phenomenon:
[image]

>  The hole on the right seems like the nozzle was coming from the right. If you load your gcode up, does it confirm what I am guessing? If so...does it just mean it is rounding the 90 degree corner too fast?
Will tweaking jerk make this less noticeable? Or maybe a fan at 100% just at the corners help?

The toolhead travel counter clockwise EVERY outer perimeter. Fan is at 100% all the time (except 1st layer) with PLA.

> Are the weights in the X-axis end body to increase heft and minimize resonance? Is the offset hole in the body for tightening the x axis tension pulley?

Yes, the left X-axis body is quite heavy due to the stepper (480g). The right X-axis body is pretty lightweight (150g). I added the weights for having good contact to the ballscrews which decouple the trapezoid rod against Z-Wobble. 
The offset hole indeed is for tensioning the x-axis belts.
[image]