MarlinFirmware/Marlin #4767: UBL on Icon on home screen

adamfilip, 2016-09-06:
@Roxy-3D Can we add a UBL on icon or "UBL ON" or "UBL OFF" text to the idle screen so you can tell its enabled or not. currrently i cant tell without going into command line

Roxy-3D, 2016-09-06:
That is probably a good idea to do.   I'll add it to the 'To Do' list.   The biggest problem with doing this, is I don't know how to do that.   Probably, I can look at how the thermal stuff does it and copy it.    I wonder what kind of icon would be appropriate.

Incidentally, I'm close to putting out a version that fixes the z_offset issues.   But I'm probably still a few days away because I want to do some testing on it.

coreymorreale, 2016-09-06:
awesome. (the z_offset part)

closed 2017-10-31 by thinkyhead